clash-lang/clash-compiler #1933: Specializing a function on a non work-free cast

gergoerdi, 2021-09-26:
https://github.com/gergoerdi/clash-chip8/tree/fea87ec8fbcb964e507a8f7b0a5d5d86a2c2f126

To reproduce, run

```
stack build --flag clash-chip8:sim --flag clash-chip8:verilator
```

The warning I get is:

```
Clash: Compiling CHIP8.VideoTestBench.topEntity
WARNING: Clash.Normalize.Transformations(1069):  specializing a function on a non work-free cast. Generated HDL implementation might contain duplicate work. Please report this as a bug. 

Expression where this occured: 

CHIP8.CPU.CPUState[8214565720323822319]
  ((c$nt[167819][LocalId] ▷
    Clash.Sized.Internal.Unsigned.Unsigned[8214565720323786301]
      12
    ~
    GHC.Natural.Natural[3674937295934324786]) ▷
    GHC.Natural.Natural[3674937295934324786]
    ~
    Clash.Sized.Internal.Unsigned.Unsigned[8214565720323786301]
      12)
  ((CHIP8.CPU.initState4[8214565720323822678][GlobalId]
  ▷
    Clash.Sized.Internal.Unsigned.Unsigned[8214565720323786301]
      12
    ~
    GHC.Natural.Natural[3674937295934324786]) ▷
    GHC.Natural.Natural[3674937295934324786]
    ~
    Clash.Sized.Internal.Unsigned.Unsigned[8214565720323786301]
      12)
Clash: Normalization took 4.301s
Clash: Netlist generation took 0.151s
Clash: Total compilation took 12.201s
```

gergoerdi, 2021-09-26:
The triggering change is https://github.com/gergoerdi/clash-chip8/commit/25c5adcc6b93ae3f8426624f936182ddb5c46124 which moves most of the definitions of the circuit into a library.

christiaanb, 2021-09-27:
Thanks for the report.

I think: https://github.com/clash-lang/clash-compiler/blob/b4ddd620ad6faa6673e06a0091b695082ea55378/clash-lib/src/Clash/Normalize/Transformations/Cast.hs#L58-L59

is wrong. It should check whether we're dealing with a global variable, as opposed to checking whether it's not a primitive.

commit 742d0562 (2021-10-15): Only cast-specialize global binders

Fixes #1933
commit 8a704cfb (2021-10-15): Only cast-specialize global binders

Fixes #1933

(cherry picked from commit 742d0562977ad4f23d68f2afb9d70f3c844e040a)