clash-lang/clash-compiler #1934: Clash 1.4.3 generates Verilator-unfriendly code

gergoerdi, 2021-09-26:
The clashilated version of https://github.com/gergoerdi/clash-compucolor2 compiles just fine with Clash 1.4.2, but on 1.4.3 I get the following error from Verilator:

```
%Error:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:3814:39:
Value too wide for 32-bits expected in this context 64'sh100000003
: ... In instance topEntity
3814 | assign c$f5_case_alt = result_22[6-1-(64'sd0)*3 -: 3];
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:3814:35:
Bit extraction of var[5:0] requires 3 bit index, not 64 bits.
: ... In instance topEntity
3814 | assign c$f5_case_alt = result_22[6-1-(64'sd0)*3 -: 3];
| ^
... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable
this message.
%Error:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:3818:41:
Value too wide for 32-bits expected in this context 64'sh100000000
: ... In instance topEntity
3818 | assign c$f5_case_alt_0 = result_22[6-1-(64'sd1)*3 -: 3];
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:3818:37:
Bit extraction of var[5:0] requires 3 bit index, not 64 bits.
: ... In instance topEntity
3818 | assign c$f5_case_alt_0 = result_22[6-1-(64'sd1)*3 -: 3];
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:3887:14:
Bit extraction of array[1023:0] requires 10 bit index, not 64 bits.
: ... In instance topEntity
3887 | f5 <= ROM[(wild_0)];
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:3959:16:
Bit extraction of array[63:0] requires 6 bit index, not 64 bits.
: ... In instance topEntity
3959 | b2 <= ROM_0[(wild_1)];
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:3982:18:
Bit extraction of array[63:0] requires 6 bit index, not 64 bits.
: ... In instance topEntity
3982 | b1_2 <= ROM_1[(wild_2)];
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:4231:20:
Bit extraction of array[2047:0] requires 11 bit index, not 64 bits.
: ... In instance topEntity
4231 | result_31_RAM[(wild_3)] <= w0;
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:4233:31:
Bit extraction of array[2047:0] requires 11 bit index, not 64 bits.
: ... In instance topEntity
4233 | result_31 <= result_31_RAM[(wild_4)];
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:4410:20:
Bit extraction of array[2047:0] requires 11 bit index, not 64 bits.
: ... In instance topEntity
4410 | result_33_RAM[(wild_5)] <= w0_0;
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:4412:31:
Bit extraction of array[2047:0] requires 11 bit index, not 64 bits.
: ... In instance topEntity
4412 | result_33 <= result_33_RAM[(wild_6)];
[... 766 lines cut ...]
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:13483:39:
Bit extraction of array[7:0] requires 3 bit index, not 64 bits.
: ... In instance topEntity
13483 | assign c$ds2_app_arg_29 = vecArray_4[(v_38)];
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:13510:38:
Bit extraction of array[16:0] requires 5 bit index, not 64 bits.
: ... In instance topEntity
13510 | assign c$ds_case_scrut = vecArray_5[(wild_22)];
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:17928:33:
Bit extraction of array[255:0] requires 8 bit index, not 64 bits.
: ... In instance topEntity
17928 | assign c$ds_app_arg_22 = ROM_3[(wild1)];
| ^
%Warning-WIDTH:
/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:19236:44:
Bit extraction of array[255:0] requires 8 bit index, not 64 bits.
: ... In instance topEntity
19236 | assign c$addressingOut_case_scrut = ROM_4[(wild14)];
| ^
%Error: Exiting due to 2 error(s)
```

gergoerdi, 2021-09-26:
I managed to bisect it to the following commit:

```
600aa7e2bf20eefe7f7b2dc739a6ecf3e09254a8 is the first bad commit
commit 600aa7e2bf20eefe7f7b2dc739a6ecf3e09254a8
Author: Leon Schoorl <leon@qbaylogic.com>
Date:   Mon Jan 4 17:02:14 2021 +0100

    Fix index_int verilog prim
    
    ... for non-var vecs
    
    (cherry picked from commit b48575b22b2ed37b96f31a004c5d707a1788f585)

 clash-lib/src/Clash/Primitives/Sized/Vector.hs | 27 ++++++++++++++++----------
 tests/shouldwork/Vector/IndexInt2.hs           | 10 +++++-----
 2 files changed, 22 insertions(+), 15 deletions(-)
```

gergoerdi, 2021-09-26:
To reproduce it easily, check out [the `clash-issue-1934` branch](https://github.com/gergoerdi/clash-compucolor2/tree/clash-issue-1934) and build with:

```
stack build --resolver=clash-good.yaml
```

and 

```
stack build --resolver=clash-bad.yaml
```

gergoerdi, 2021-09-26:
Not quite minimal yet, but I managed to shrink my input to a mere 135 lines:

```
{-# LANGUAGE NumericUnderscores, RecordWildCards, ViewPatterns #-}
{-# LANGUAGE ApplicativeDo #-}
module Hardware.Compucolor2
    ( topEntity
    ) where

import Clash.Prelude
import Clash.Annotations.TH

import qualified Clash.Signal.Delayed.Bundle as D
import RetroClash.Utils
import RetroClash.VGA
import RetroClash.Video
import RetroClash.Delayed hiding (delayedBlockRam1)

import Control.Monad
import Data.Maybe (isJust, isNothing)

type TextWidth = 64
type TextHeight = 32
type FontWidth = 6
type FontHeight = 8

attributes :: Unsigned 8 -> (Bool, Bool, (Bit, Bit, Bit), (Bit, Bit, Bit))
attributes = bitCoerce

fromBGR :: (Bounded r, Bounded g, Bounded b) => (Bit, Bit, Bit) -> (r, g, b)
fromBGR (b, g, r) = (stretch r, stretch g, stretch b)
  where
    stretch 0 = minBound
    stretch 1 = maxBound

type TextSize = TextWidth * TextHeight
type VidSize = TextSize * 2
type VidAddr = Index VidSize

-- | 40 MHz clock, needed for the VGA mode we use.
createDomain vSystem{vName="Dom40", vPeriod = hzToPeriod 40_000_000}

video
    :: (HiddenClockResetEnable Dom40)
    => Signal Dom40 (Maybe (Bool, VidAddr))
    -> Signal Dom40 (Maybe (Unsigned 8))
    -> ( VGAOut Dom40 8 8 8
       , Signal Dom40 (Maybe (Unsigned 8))
       )
video (unsafeFromSignal -> extAddr) (unsafeFromSignal -> extWrite) =
    ( delayVGA vgaSync rgb
    , toSignal extRead
    )
  where
    VGADriver{..} = vgaDriver vga800x600at60
    (fromSignal -> x1, fromSignal -> x0) = scale (SNat @FontWidth) . fst . scale (SNat @2) . center $ vgaX
    (fromSignal -> y1, fromSignal -> y0) = scale (SNat @FontHeight) . fst . scale (SNat @2) . center $ vgaY

    hblank = isNothing <$> x1

    newCol = liftD (changed Nothing) x0
    newChar = liftD (changed Nothing) x1
    visible = isJust <$> x1 .&&. isJust <$> y1

    extAddr' = schedule <$> hblank <*> extAddr
      where
        schedule hblank extAddr = do
            (urgent, addr) <- extAddr
            guard $ urgent || hblank
            return addr
    extAddr1 :> extAddr2 :> Nil = D.unbundle $ unbraid <$> extAddr'

    intAddr = guardA newChar $ liftA2 toAddr <$> x1 <*> y1
      where
        toAddr :: Index TextWidth -> Index TextHeight -> Index TextSize
        toAddr x1 y1 = bitCoerce (y1, x1)

    videoMem extAddr = sharedDelayedRW ram $
        extAddr `withWrite` extWrite :>
        noWrite intAddr :>
        Nil
      where
        ram = singlePort $ delayedRam (blockRam1 NoClearOnReset (SNat @TextSize) 0)

    extRead1 :> charRead :> Nil = videoMem extAddr1
    extRead2 :> attrRead :> Nil = videoMem extAddr2
    extRead = extRead1 .<|>. extRead2

    plotAddr = charRead .<| 0

    attr = delayedRegister 0 (.|>. attrRead)
    (_, _, back, fore) = D.unbundle $ attributes <$> attr

    block = enable (delayI False newChar) $
            (plotRom plotAddr (delayI Nothing y0 .<| 0))
    pixel = liftD2 shifterL block (delayI False newCol)

    border = pure (0x30, 0x30, 0x30)

    (_, isCursor) = D.unbundle $ do
        x1 <- fromIntegral <$> (delayI Nothing x1 .<| 0)
        y1 <- fromIntegral <$> (delayI Nothing y1 .<| 0)
        y0 <- delayI Nothing y0 .<| 0
        pure (False, False)

    white = (1, 1, 1)

    palette = mux isCursor (pure $ repeat white) $ D.bundle $
        back :>
        fore :>
        Nil

    rgb = mux (not <$> delayI False visible) border $
        fromBGR <$> palette .!!. pixel

plotRom
    :: (HiddenClockResetEnable dom)
    => DSignal dom n (Unsigned 8)
    -> DSignal dom n (Index 8)
    -> DSignal dom (n + 1) (BitVector 8)
plotRom char row = pure 0

topEntity
    :: "CLK_40MHZ" ::: Clock Dom40
    -> "RESET"     ::: Reset Dom40
    -> "SWITCHES"  ::: Signal Dom40 (BitVector 8)
    -> ( "VGA"     ::: VGAOut Dom40 8 8 8
       )
topEntity = withEnableGen board
  where
    board switches = vga
      where
        vidAddr = pure Nothing
        vidWrite = pure Nothing
        (vga, vidRead) = video vidAddr vidWrite

makeTopEntity 'topEntity
```

gergoerdi, 2021-09-27:
With no external dependencies:

```
module Hardware.Compucolor2
    ( topEntity
    ) where

import Clash.Prelude
import Clash.Annotations.TH
import qualified Clash.Signal.Delayed.Bundle as D

delayedRegister
    :: (NFDataX a, HiddenClockResetEnable dom)
    => a
    -> (DSignal dom d a -> DSignal dom d a)
    -> DSignal dom (d + 1) a
delayedRegister initial feedback = r
  where
    r = unsafeFromSignal $ register initial $ toSignal new
    old = antiDelay (SNat @1) r
    new = feedback old

video
    :: (HiddenClockResetEnable dom)
    => Signal dom (Unsigned 8)
video = toSignal $ (!!) <$> palette <*> 0
  where
    attr = delayedRegister 0 id
    palette = mux (pure False) (pure $ repeat 0) $ D.bundle $
        attr :>
        Nil

topEntity
    :: "CLK"   ::: Clock System
    -> "RESET" ::: Reset System
    -> "VGA"   ::: Signal System (Unsigned 8)
topEntity clk rst = withClockResetEnable clk rst enableGen video

makeTopEntity 'topEntity
```

gergoerdi, 2021-09-27:
Without all that `DSignal` business. Now it's getting to the point where it is very hard to remove anything and still trigger the problem:

```
module Hardware.Compucolor2
    ( topEntity
    ) where

import Clash.Prelude
import Clash.Annotations.TH

topEntity
    :: "CLK"   ::: Clock System
    -> "RESET" ::: Reset System
    -> "VGA"   ::: Signal System (Unsigned 8)
topEntity clk rst = withClockResetEnable clk rst enableGen video
  where
    video :: (HiddenClockResetEnable dom) => Signal dom (Unsigned 8)
    video = (!!) <$> palette <*> 0
      where
        attr = register 0 attr
        palette = mux (pure False) (pure $ repeat 0) $ bundle $
            attr :>
            Nil

makeTopEntity 'topEntity
```

gergoerdi, 2021-09-27:
Generated Verilog:

from "good" (1.4.2-equivalent):

```
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.4.2. DO NOT MODIFY.
*/
`timescale 100fs/100fs
module topEntity
    ( // Inputs
      input  CLK // clock
    , input  RESET // reset

      // Outputs
    , output wire [7:0] VGA
    );
  wire [7:0] f2;
  reg [7:0] \$s!!_attr  = 8'd0;

  assign VGA = f2[7:0];

  assign f2 = \$s!!_attr ;

  // register begin
  always @(posedge CLK or  posedge  RESET) begin : _$s_attr_register
    if ( RESET) begin
      \$s!!_attr  <= 8'd0;
    end else begin
      \$s!!_attr  <= \$s!!_attr ;
    end
  end
  // register end

endmodule
```

From "bad" (1.4.3-equivalent):

```
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.4.2. DO NOT MODIFY.
*/
`timescale 100fs/100fs
module topEntity
    ( // Inputs
      input  CLK // clock
    , input  RESET // reset

      // Outputs
    , output wire [7:0] VGA
    );
  wire [7:0] f2;
  reg [7:0] \$s!!_attr  = 8'd0;

  // index lit begin
  assign VGA = f2[8-1-(64'sd0)*8 -: 8];
  // index lit end

  assign f2 = \$s!!_attr ;

  // register begin
  always @(posedge CLK or  posedge  RESET) begin : _$s_attr_register
    if ( RESET) begin
      \$s!!_attr  <= 8'd0;
    end else begin
      \$s!!_attr  <= \$s!!_attr ;
    end
  end
  // register end

endmodule
```

gergoerdi, 2021-09-27:
The difference is that "good" has:

```
assign VGA = f2[7:0];
```

whereas "bad" has:

```
  // index lit begin
  assign VGA = f2[8-1-(64'sd0)*8 -: 8];
  // index lit end
```

Unfortunately, this is where my expertise runs out...

gergoerdi, 2021-09-27:
I've moved all code related to this to https://github.com/gergoerdi/clash-issue-1934 please use that repo (or, really, just copy the code from https://github.com/clash-lang/clash-compiler/issues/1934#issuecomment-927450612, it is fully standalone now)

christiaanb, 2021-09-27:
I guess this is an unintended consequence of https://github.com/clash-lang/clash-compiler/commit/10aca06ea0058a0ec4ef3369b454c447feb0d380

I'll have to look at the debug logs to see what changes exactly.

leonschoorl, 2021-09-27:
I couldn't reproduce this with v1.4.3 and the code from https://github.com/clash-lang/clash-compiler/issues/1934#issuecomment-927450612
Here is just generates:
```verilog
  assign VGA = f2[7:0];
```

Tried it with clash-1.4.3 from snapcraft and local builds of the `v1.4.3` tag build both with ghc-9.0.1 and 8.10.2.

I also tried you cloning your repo and doing `stack run --resolver=clash-bad.yaml compucolor2-verilator` there.
After a lot of compiling that just outputs a single `0`, which I assume is the correct behavior.

Could you confirm that issue still happens with the current example, and you haven't accidentally minimized it too much or  something like that.

gergoerdi, 2021-09-28:
Here is what I see with https://github.com/gergoerdi/clash-issue-1934/tree/f47158f752cd13dfb5f15b57e8515f9ad42aadcd:

```
$ rm -rf .stack-work/ && stack build --resolver=clash-bad.yaml
Building all executables for `clash-compucolor2' once. After a successful build of all of them, only specified executables will be rebuilt.
clash-compucolor2> configure (lib + exe)
[1 of 2] Compiling Main             ( /home/cactus/prog/clash/bugs/clash-issue-1934/Setup.hs, /home/cactus/prog/clash/bugs/clash-issue-1934/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/setup/Main.o )
[2 of 2] Compiling StackSetupShim   ( /home/cactus/sdk/stack/setup-exe-src/setup-shim-mPHDZzAJ.hs, /home/cactus/prog/clash/bugs/clash-issue-1934/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/setup/StackSetupShim.o )
Linking /home/cactus/prog/clash/bugs/clash-issue-1934/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/setup/setup ...
Configuring clash-compucolor2-0.1.0...
clash-compucolor2> build (lib + exe)
Preprocessing library for clash-compucolor2-0.1.0..
Building library for clash-compucolor2-0.1.0..
[1 of 2] Compiling Hardware.Compucolor2
[2 of 2] Compiling Paths_clash_compucolor2
GHC: Parsing and optimising modules took: 1.516s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.002s
Clash: Parsing and compiling primitives took 0.161s
GHC+Clash: Loading modules cumulatively took 1.946s
Clash: Compiling Hardware.Compucolor2.topEntity
Clash: Normalization took 0.003s
Clash: Netlist generation took 0.001s
Clash: Total compilation took 2.019s
setup: 'make' exited with an error:
%Error:
/home/cactus/prog/clash/bugs/clash-issue-1934/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:17:22:
Value too wide for 32-bits expected in this context 64'sh100000000
: ... In instance topEntity
17 | assign VGA = f2[8-1-(64'sd0)*8 -: 8];
| ^
%Warning-WIDTH:
/home/cactus/prog/clash/bugs/clash-issue-1934/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v:17:18:
Bit extraction of var[7:0] requires 3 bit index, not 64 bits.
: ... In instance topEntity
17 | assign VGA = f2[8-1-(64'sd0)*8 -: 8];
| ^
... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable
this message.
%Error: Exiting due to 1 error(s)
make: ***
[.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/verilator/Makefile:25:
.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/verilator/obj/VSim.mk]
Error 1

--  While building package clash-compucolor2-0.1.0 (scroll up to its section to see the error) using:
      /home/cactus/prog/clash/bugs/clash-issue-1934/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/setup/setup --builddir=.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0 build lib:clash-compucolor2 exe:compucolor2-verilator --ghc-options " -fdiagnostics-color=always"
    Process exited with code: ExitFailure 1
```

leonschoorl, 2021-09-28:
I get:
```
$ git log
commit f47158f752cd13dfb5f15b57e8515f9ad42aadcd (HEAD -> master, origin/master, origin/HEAD)
Author: Dr. ERDI Gergo
Date:   Mon Sep 27 10:21:02 2021 +0800

    Reproducer for https://github.com/clash-lang/clash-compiler/issues/1934
$ stack --version
Version 2.7.3, Git revision 7927a3aec32e2b2e5e4fb5be76d0d50eddcc197f x86_64 hpack-0.34.4
$ verilator --version
Verilator 3.916 2017-11-25 rev verilator_3_914-65-g0478dbd
$ rm -rf .stack-work/ && stack build --resolver=clash-bad.yaml
Building all executables for `clash-compucolor2' once. After a successful build of all of them, only specified executables will be rebuilt.
clash-compucolor2> configure (lib + exe)
[1 of 2] Compiling Main             ( /home/leon/qbaylogic/clash-issue-1934/Setup.hs, /home/leon/qbaylogic/clash-issue-1934/.stack-work/dist/x86_64-linux/Cabal-3.2.1.0/setup/Main.o )
[2 of 2] Compiling StackSetupShim   ( /home/leon/.stack/setup-exe-src/setup-shim-mPHDZzAJ.hs, /home/leon/qbaylogic/clash-issue-1934/.stack-work/dist/x86_64-linux/Cabal-3.2.1.0/setup/StackSetupShim.o )
Linking /home/leon/qbaylogic/clash-issue-1934/.stack-work/dist/x86_64-linux/Cabal-3.2.1.0/setup/setup ...
Configuring clash-compucolor2-0.1.0...
clash-compucolor2> build (lib + exe)
Preprocessing library for clash-compucolor2-0.1.0..
Building library for clash-compucolor2-0.1.0..
[1 of 2] Compiling Hardware.Compucolor2
[2 of 2] Compiling Paths_clash_compucolor2
GHC: Parsing and optimising modules took: 1.124s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.001s
Clash: Parsing and compiling primitives took 0.159s
GHC+Clash: Loading modules cumulatively took 1.463s
Clash: Compiling Hardware.Compucolor2.topEntity
Clash: Normalization took 0.003s
Clash: Netlist generation took 0.001s
Clash: Total compilation took 1.476s
Preprocessing library for clash-compucolor2-0.1.0..
Building library for clash-compucolor2-0.1.0..
Preprocessing executable 'compucolor2-verilator' for clash-compucolor2-0.1.0..
Building executable 'compucolor2-verilator' for clash-compucolor2-0.1.0..
[1 of 3] Compiling Clash.Clashilator.FFI
[2 of 3] Compiling Main
[3 of 3] Compiling Paths_clash_compucolor2
Linking .stack-work/dist/x86_64-linux/Cabal-3.2.1.0/build/compucolor2-verilator/compucolor2-verilator ...
clash-compucolor2> copy/register
Installing library in /home/leon/qbaylogic/clash-issue-1934/.stack-work/install/x86_64-linux/522b5a433bb0354cf332655aa74cf797289920f806b7ff504ccd0e4ad91d3df1/8.10.4/lib/x86_64-linux-ghc-8.10.4/clash-compucolor2-0.1.0-8RVr44BMq8GzInuGxGxlr
Installing executable compucolor2-verilator in /home/leon/qbaylogic/clash-issue-1934/.stack-work/install/x86_64-linux/522b5a433bb0354cf332655aa74cf797289920f806b7ff504ccd0e4ad91d3df1/8.10.4/bin
Registering library for clash-compucolor2-0.1.0..
```

Maybe my (old) verilator behaves differently then yours.
But I have now found the bad verilog code:
```
$ cat ./.stack-work/dist/x86_64-linux/Cabal-3.2.1.0/build/compucolor2-verilator/_clashilator/clash-syn/Hardware.Compucolor2.topEntity/topEntity.v
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.4.2. DO NOT MODIFY.
*/
`timescale 100fs/100fs
module topEntity
    ( // Inputs
      input  CLK // clock
    , input  RESET // reset

      // Outputs
    , output wire [7:0] VGA
    );
  wire [7:0] f2;
  reg [7:0] \$s!!_attr  = 8'd0;

  // index lit begin
  assign VGA = f2[8-1-(64'sd0)*8 -: 8];
  // index lit end

  assign f2 = \$s!!_attr ;

  // register begin
  always @(posedge CLK or  posedge  RESET) begin : _$s_attr_register
    if ( RESET) begin
      \$s!!_attr  <= 8'd0;
    end else begin
      \$s!!_attr  <= \$s!!_attr ;
    end
  end
  // register end

endmodule
```

gergoerdi, 2021-09-28:
I have *Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad* installed.

gergoerdi, 2021-09-28:
I have now pushed a version that you can just `stack run` without any Verilator involvement and then just look at `_build/Hardware.Compucolor2.topEntity/topEntity.v`

gergoerdi, 2021-09-28:
Please let me know if you can reproduce this now with https://github.com/gergoerdi/clash-issue-1934/tree/aeaf219c620edafb9bd7271e8fdd67328d2d89b3 and if you need anything more from me to look into this.

leonschoorl, 2021-09-29:
Thanks

I've since updated to`Verilator 4.212 2021-09-01 rev v4.212` and now I seen the same errors as you did.

Although mostly a distraction for this bug, it is interesting that building through clashilator seems to result in different HDL then building with a standalone clash binary.

The offending verilog template is just in there: https://github.com/clash-lang/clash-compiler/blob/b4ddd620ad6faa6673e06a0091b695082ea55378/clash-lib/src/Clash/Primitives/Sized/Vector.hs#L275-L277

And can be triggered more reliably with something like:
```haskell
f :: Unsigned 8 -> Unsigned 8
f x = (3 :> x :> Nil) !! 1
{-# NOINLINE f #-}

topEntity = f
```

I'll try to find a better way to write that template tomorrow.

christiaanb, 2021-09-30:
As a work-around you can use `-fclash-intwidth=32`, which is "unsafe" if your code depends on `Int` and `Word` being 64 bit.

christiaanb, 2021-09-30:
So the fix is to extend `mkLit` https://github.com/clash-lang/clash-compiler/blob/0415bcf1421f8d6c3f910536beb4c8b26e23493a/clash-lib/src/Clash/Netlist/BlackBox/Util.hs#L603-L607
To also match on `BlackBoxE "GHC.Types.I#" ...` and `BlackBoxE "GHC.Types.W#" ...`

commit 9cc11f68 (2021-10-14): Fix ~LIT rendering for more types of literals

It now works for:
  KnownNat, SNat,
  Integer, Natural,
  Int, Word and their 8,16,32,64 bits variants
  Unsigned,Signed,Index
  Int#, Word#

And better document what ~LIT and ~CONST are supposed to do.

Fixes #1934
commit d99523ec (2021-10-19): Fix ~LIT rendering for more types of literals

It now works for:
  KnownNat, SNat,
  Integer, Natural,
  Int, Word and their 8,16,32,64 bits variants
  Unsigned,Signed,Index
  Int#, Word#

And better document what ~LIT and ~CONST are supposed to do.

Fixes #1934
closed 2021-10-19 by leonschoorl