clash-lang/clash-compiler #1987: `Reducer` test case has become 7-50 times slower in `master`

martijnbastiaan, 2021-11-07:
`v1.4.0`: 5.16s
`v1.4.6`: 38.82s
`master`: 262.34s

christiaanb, 2021-11-07:
I cannot reproduce the 1.4 slowdown, neither on the v1.4.6 tag nor 4ce46d58fb1d4d016cf40c0b501e7bae9dfdb0b3 (current tip of the 1.4 branch)

```
christiaan@DESKTOP-O5QKI8I:~/clash-compiler$ time cabal run clash -- --vhdl -fclash-no-cache -fclash-clear examples/Reducer.hs 
Up to date
Loaded package environment from /home/christiaan/clash-compiler/.ghc.environment.x86_64-linux-8.10.7
GHC: Parsing and optimising modules took: 3.106s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.002s
Clash: Parsing and compiling primitives took 0.151s
GHC+Clash: Loading modules cumulatively took 3.639s
Clash: Compiling Reducer.topEntity
Clash: Ignoring previously made caches
Clash: Normalization took 0.706s
Clash: Netlist generation took 0.420s
Clash: Total compilation took 4.820s

real    0m5.463s
user    0m4.992s
sys     0m0.451s
```

christiaanb, 2021-11-07:
Also head of master doesn't seem slower:
```
christiaan@DESKTOP-O5QKI8I:~/clash-compiler$ time cabal run clash -- --vhdl -fclash-no-cache -fclash-clear examples/Reducer.hs 
Up to date
Loaded package environment from /home/christiaan/clash-compiler/.ghc.environment.x86_64-linux-8.10.7
GHC: Parsing and optimising modules took: 3.204s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.002s
Clash: Parsing and compiling primitives took 0.160s
GHC+Clash: Loading modules cumulatively took 3.752s
Clash: Compiling Reducer.topEntity
Clash: Ignoring previously made caches
Clash: Normalization took 0.836s
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.integerToInt: GHC.Integer.Type.integerToInt: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
Clash: Netlist generation took 0.508s
Clash: Total compilation took 5.142s

real    0m5.814s
user    0m5.369s
sys     0m0.401s
```

christiaanb, 2021-11-07:
Ah, there does seem to be a difference with `DebugSilent`:

`v1.4.2`:
```
christiaan@DESKTOP-O5QKI8I:~/clash-compiler$ time cabal run clash -- --verilog -fclash-no-cache -fclash-clear -fclash-debug DebugSilent examples/Reducer.hs 
Resolving dependencies...
Up to date
Loaded package environment from /home/christiaan/clash-compiler/.ghc.environment.x86_64-linux-8.10.7
GHC: Parsing and optimising modules took: 3.403s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.002s
Clash: Parsing and compiling primitives took 0.172s
GHC+Clash: Loading modules cumulatively took 3.986s
Clash: Compiling Reducer.topEntity
Clash: Ignoring previously made caches
Clash: Normalization took 1.162s
Clash: Netlist generation took 0.442s
Clash: Total compilation took 5.632s

real    0m9.388s
user    0m8.716s
sys     0m0.662s
```
`v1.4.3`:
```
christiaan@DESKTOP-O5QKI8I:~/clash-compiler$ time cabal run clash -- --verilog -fclash-no-cache -fclash-clear -fclash-debug DebugSilent examples/Reducer.hs 
Resolving dependencies...
Up to date
Loaded package environment from /home/christiaan/clash-compiler/.ghc.environment.x86_64-linux-8.10.7
GHC: Parsing and optimising modules took: 3.651s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.002s
Clash: Parsing and compiling primitives took 0.176s
GHC+Clash: Loading modules cumulatively took 4.224s
Clash: Compiling Reducer.topEntity
Clash: Ignoring previously made caches
Clash: Normalization took 38.996s
Clash: Netlist generation took 0.180s
Clash: Total compilation took 43.445s

real    0m47.718s
user    0m46.815s
sys     0m0.893s
```

martijnbastiaan, 2021-11-07:
Ah right, I'm running it using `cabal run clash-testsuite -- -p Reducer.VHDL`. So perhaps it's just increased logging then :sweat: .

christiaanb, 2021-11-07:
No, the testsuite runs everything with `-fclash-debug Silent`, which just performs extra checks: it doesn't increase I/O actions.

christiaanb, 2021-11-07:
So it seems going from 99f0e181665258d2e79888c0e5eec3392525ee68
```
christiaan@DESKTOP-O5QKI8I:~/clash-compiler$ time cabal run clash -- --verilog -fclash-no-cache -fclash-clear -fclash-debug DebugSilent examples/Reducer.hs 
Up to date
Loaded package environment from /home/christiaan/clash-compiler/.ghc.environment.x86_64-linux-8.10.7
GHC: Parsing and optimising modules took: 2.839s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.002s
Clash: Parsing and compiling primitives took 0.430s
GHC+Clash: Loading modules cumulatively took 3.676s
Clash: Compiling Reducer.topEntity
Clash: Ignoring previously made caches
Clash: Normalization took 1.414s
Clash: Netlist generation took 0.163s
Clash: Total compilation took 5.292s

real    0m5.934s
user    0m5.497s
sys     0m0.416s
```
to 10b929c49571edcec2205b4b5388e7e28009832a
```
christiaan@DESKTOP-O5QKI8I:~/clash-compiler$ time cabal run clash -- --verilog -fclash-no-cache -fclash-clear -fclash-debug DebugSilent examples/Reducer.hs 
Up to date
Loaded package environment from /home/christiaan/clash-compiler/.ghc.environment.x86_64-linux-8.10.7
GHC: Parsing and optimising modules took: 2.837s
GHC: Loading external modules from interface files took: 0.001s
GHC: Parsing annotations took: 0.002s
Clash: Parsing and compiling primitives took 0.424s
GHC+Clash: Loading modules cumulatively took 3.670s
Clash: Compiling Reducer.topEntity
Clash: Ignoring previously made caches
Clash: Normalization took 38.265s
Clash: Netlist generation took 0.181s
Clash: Total compilation took 42.161s

real    0m42.793s
user    0m42.213s
sys     0m0.571s
```
increases the normalization time significantly.

christiaanb, 2021-11-07:
Which makes sense, because 10b929c49571edcec2205b4b5388e7e28009832a added https://github.com/clash-lang/clash-compiler/commit/10b929c49571edcec2205b4b5388e7e28009832a#diff-ef03bca01fe87dd3c262b523212da81f7626bdc2a0f9af328fe0376bcc3f65b1L272
```diff
- Monad.when (lvl >= DebugApplied && not hasChanged && not (exprOld `aeqTerm` exprNew)) $
+ Monad.when (lvl >= DebugSilent && not hasChanged && not (exprOld `aeqTerm` exprNew)) $
```

christiaanb, 2021-11-07:
When I revert that particular line `v1.4.3` run as normal again:
```
christiaan@DESKTOP-O5QKI8I:~/clash-compiler$ time cabal run clash -- --verilog -fclash-no-cache -fclash-clear -fclash-debug DebugSilent examples/Reducer.hs 
Up to date
Loaded package environment from /home/christiaan/clash-compiler/.ghc.environment.x86_64-linux-8.10.7
GHC: Parsing and optimising modules took: 2.808s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.002s
Clash: Parsing and compiling primitives took 0.165s
GHC+Clash: Loading modules cumulatively took 3.653s
Clash: Compiling Reducer.topEntity
Clash: Ignoring previously made caches
Clash: Normalization took 1.438s
Clash: Netlist generation took 0.167s
Clash: Total compilation took 5.299s

real    0m5.923s
user    0m5.529s
sys     0m0.383s
```

martijnbastiaan, 2021-11-07:
Hmm, not sure I agree with the `aeqTerm` there. Seems to me that _any_ change should be reported, which would also be a lot quicker to check.

christiaanb, 2021-11-07:
Well, all we have is alpha-equality for terms.

christiaanb, 2021-11-07:
Also, the renaming isn’t that expensive. So even if we were to use structural equality, we wouldn’t go back to our 5.5s compile.

christiaanb, 2021-11-07:
But you can always write a structural equivalence check and see if it does improve things significantly: it’d be better than PR #1988

martijnbastiaan, 2021-11-07:
Given that in 95%+ of the time the exact input is returned, we could try and exploit [`StableName`](https://hackage.haskell.org/package/base-4.16.0.0/docs/System-Mem-StableName.html) or perhaps even [`reallyUnsafePtrEquality#`](https://hackage.haskell.org/package/base-4.16.0.0/docs/GHC-Exts.html#v:reallyUnsafePtrEquality-35-).

christiaanb, 2021-11-07:
@martijnbastiaan turns out renaming is expensive. I'm working on a patch that introduces structural equivalence. While we don't get 5s, we do get 13s with a structural equivalence check.

martijnbastiaan, 2021-11-07:
Okay, thanks for looking into it. For your interest, I've got a branch [here](https://github.com/clash-lang/clash-compiler/commit/64148b880e68fccd394ed61066f59d4060de4ade) which I was working on as an alternative for https://github.com/clash-lang/clash-compiler/pull/1986. It makes the default structural.

christiaanb, 2021-11-07:
Yeah, I think we should get rid of the `Eq Term` and `Eq Type` instances. And introduce:
```haskell
newtype Structural a = Structural a
newtype Alpha a = Alpha a

instance (Structural Term) where
  (==) = coerce eqTerm

instance (Alpha Term) where
  (==) = coerce aeqTerm
  
instance (Structural Type) where
  (==) = coerce eqType

instance (Alpha Type) where
  (==) = coerce aeqType
```

martijnbastiaan, 2021-11-07:
@christiaanb Is that 13s on `master` or head of `1.4`?

christiaanb, 2021-11-07:
13s on `master`

martijnbastiaan, 2021-11-07:
> Yeah, I think we should get rid of the Eq Term and Eq Type instances. And introduce:

Please feel free to cherry-pick from my branch. Given renovations at home, I don't think I'll be working on it for at least two weeks.

commit 2797c037 (2021-11-07): Use structural equality in "No notice change check"

This brings the `DebugSilent` run for Reducer down from 4m30s to
13s. A very welcome improvement for CI.

Fixes #1987
commit 47aa0d0a (2021-11-08): Use structural equality in "No notice change" check

This brings the `DebugSilent` run for Reducer down from 4m30s to
13s. A very welcome improvement for CI.

Fixes #1987
PR #1990 (2021-11-08): Use structural equality in "No notice change" check
This brings the `DebugSilent` run for Reducer down from 4m30s to 13s. A very welcome improvement for CI.

Fixes #1987

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files
commit 0f1c5ca1 (2021-11-08): Use structural equality in "No notice change" check (#1990)

This brings the `DebugSilent` run for Reducer down from 4m30s to
13s. A very welcome improvement for CI.

Fixes #1987
closed 2021-11-08 by christiaanb