clash-lang/clash-compiler #1996: Invalid namespace reference in generated VHDL (for Synplify)

basile-henry, 2021-11-11:
Clash generates VHDL that is not considered valid by Synopsis' Synplify Pro tool.

The following clash code:

```haskell
import Clash.Prelude

topEntity
  :: Clock System
  -> Reset System
  -> Signal System (Int, Int)
  -> Signal System (Int, Int)
topEntity clk rst =
  withClockResetEnable clk rst enableGen register (0, 0)
```

Generates the following VHDL for the types file (the `topentity.vhdl` is straight forward and not relevant for this issue):

`vhdl/Main/main_types.vhdl`:
```vhdl
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

package main_types is
  subtype rst_system is std_logic;

  type tup2 is record
    tup2_sel0_signed_0 : signed(63 downto 0);
    tup2_sel1_signed_1 : signed(63 downto 0);
  end record;
  subtype clk_system is std_logic;
  function toSLV (sl : in std_logic) return std_logic_vector;
  function fromSLV (slv : in std_logic_vector) return std_logic;
  function toSLV (s : in signed) return std_logic_vector;
  function fromSLV (slv : in std_logic_vector) return signed;
  function toSLV (p : main_types.tup2) return std_logic_vector;
  function fromSLV (slv : in std_logic_vector) return main_types.tup2;
end;

package body main_types is
  function toSLV (sl : in std_logic) return std_logic_vector is
  begin
    return std_logic_vector'(0 => sl);
  end;
  function fromSLV (slv : in std_logic_vector) return std_logic is
    alias islv : std_logic_vector (0 to slv'length - 1) is slv;
  begin
    return islv(0);
  end;
  function toSLV (s : in signed) return std_logic_vector is
  begin
    return std_logic_vector(s);
  end;
  function fromSLV (slv : in std_logic_vector) return signed is
    alias islv : std_logic_vector(0 to slv'length - 1) is slv;
  begin
    return signed(islv);
  end;
  function toSLV (p : main_types.tup2) return std_logic_vector is
  begin
    return (toSLV(p.tup2_sel0_signed_0) & toSLV(p.tup2_sel1_signed_1));
  end;
  function fromSLV (slv : in std_logic_vector) return main_types.tup2 is
  alias islv : std_logic_vector(0 to slv'length - 1) is slv;
  begin
    return (fromSLV(islv(0 to 63)),fromSLV(islv(64 to 127)));
  end;
end;
```

Synplify Pro errors out with:

```
error[CD157]: Expecting return type name
  --> /nix/store/ps5r3nr97z7x4qkxgpb8b1bx76axsyg3-virtual-simple-src/rtl/vhdl/Main/main_types.vhdl:18:54
   |
18 |   function fromSLV (slv : in std_logic_vector) return main_types.tup2;
   |                                                       ^^^^^^^^^^

error[CD157]: Expecting return type name
  --> /nix/store/ps5r3nr97z7x4qkxgpb8b1bx76axsyg3-virtual-simple-src/rtl/vhdl/Main/main_types.vhdl:44:54
   |
44 |   function fromSLV (slv : in std_logic_vector) return main_types.tup2 is
   |                                                       ^^^^^^^^^^
 ```
 
The issue as far as I understand it is that Synplify Pro doesn't like qualified/namespaced names used inside of the namespace definition itself. Maybe it's not in scope yet?
It could very well just be an issue with that tool, since Vivado and Quartus are happy with such VHDL. But it would be nice for Clash to generate code without these qualified names. Do these even need to be qualified anyway? Is it ever ambiguous what they refer to? If so, maybe Clash could have a flag to opt in to generate unqualified names.
The current workaround we have is to apply `sed -i '/^use work/b; s/main_types\.//g'` to this type file, after which Synplify Pro works just fine.

basile-henry, 2021-11-11:
I should have specified, this is Clash 1.2, I have not test on a more recent version.

alex-mckenna, 2021-11-15:
@basile-henry The fix for this has been backported to 1.4 and 1.2 now. I don't think we automate builds from the 1.2 branch anymore though, but it gives you a way to get the fix without forcing you to upgrade to 1.4 immediately

basile-henry, 2021-11-15:
Great, thank you very much @alex-mckenna ! I can confirm this fix on the 1.2 branches fixes our issue. :tada: :smile:

commit 3056cbb6 (2021-11-11): Don't prefix package name in `_types` package in VHDL

Similar to ece7f26 for SystemVerilog, types should not appear
qualified in the package body for the types package in VHDL.

Fixes #1996.
commit 3bb6522a (2021-11-11): Don't prefix package name in `_types` package in VHDL

Similar to ece7f26 for SystemVerilog, types should not appear
qualified in the package body for the types package in VHDL.

Fixes #1996.
commit 5c6bf8b0 (2021-11-11): Don't prefix package name in `_types` package in VHDL

Similar to ece7f26 for SystemVerilog, types should not appear
qualified in the package body for the types package in VHDL.

Fixes #1996.
PR #1997 (2021-11-11): Don't prefix package name in `_types` package in VHDL
Similar to ece7f26 for SystemVerilog, types should not appear
qualified in the package body for the types package in VHDL.

Fixes #1996.

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files

[comment]: # (Depending on the change, some of these points may not be necessary. If so, leave the boxes unchecked so it is easier for a reviewer to see what has been omitted.)