clash-lang/clash-compiler #2008: Manifest changes cause "target directory not empty" errors

martijnbastiaan, 2021-11-21:
Upgrading Clash can cause slightly annoying error messages as mentioned in this error:

https://github.com/clash-lang/clash-compiler/blob/88ced730c34214f3020fd01e2e68addb36473216/clash-lib/src/Clash/Driver.hs#L859-L873

We could improve this by defining a version of `Manifest` that just reads in files and their hashes:

https://github.com/clash-lang/clash-compiler/blob/88ced730c34214f3020fd01e2e68addb36473216/clash-lib/src/Clash/Driver/Manifest.hs#L173-L179

as this is the only information used to determine whether a target directory can be cleaned out or not.

leonschoorl, 2021-11-22:
In general a version in the manifest might be a good addition.

But for this specific case it's somewhat problematic.
If we want to be to validate old manifests, we'd have to be able to calculate hashes in the old way.
So we can't drop the problematic `Hashable` instance, nor can we update it to compute hashes in the "right" way to match the `Eq` instance.

martijnbastiaan, 2021-11-22:
Ah wait, I'm proposing defining a version _of_ `Manifest` that only includes the files subset. Something like:

```haskell
data ManifestFiles = ManifestFiles
  { fileNames :: [(FilePath, ByteString)]
    -- ^ Names and hashes of all the generated files for the @TopEntity@. Hashes
    -- are SHA256.
  }
```

..as this is the only thing used to decide whether we can remove a target directory. I.e., Clash will refuse to compile if the target directory contains any files not generated by Clash. Failures to parse the full manifest file will continue to invalidate the cache (and a proper version would help here).

commit 7e4e38c4 (2022-01-30): Detect file changes even if Manifest changes

Fixes #2008
commit 131b943a (2022-01-30): Detect file changes even if Manifest changes

Fixes #2008
commit 8b23abd3 (2022-01-30): Detect file changes even if Manifest changes

Fixes #2008
commit 98c5960e (2022-01-30): Detect file changes even if Manifest changes

Fixes #2008
closed 2022-01-30 by martijnbastiaan