clash-lang/clash-compiler #2014: Render synthesis attributes in more places

alex-mckenna, 2021-12-03:
From the FP slack:

> Is there any way to generate module annotations from Clash? I see there's the experimental [Annotate](https://hackage.haskell.org/package/clash-prelude-1.4.6/docs/Clash-Annotations-SynthesisAttributes.html) but that seems to only cover IO ports of a module rather than the module itself

Both verilog and VHDL allow user-defined attributes to be placed on different things. To quote the VHDL spec (IEEE 1076-1993)

> An attribute may be associated with an entity declaration, an architecture, a configuration,a procedure, a function, a package, a type, a subtype, a constant, a signal, a variable, a component, a label, a literal, a unit, a group, or a file.

And the Verilog spec (IEEE 1364-2001)

> An attribute_instance can appear in the Verilog description as a prefix attached to a declaration, a module item, a statement, or a port connection.

Supporting this in Clash consists of two main changes:

  * Changing the HDL backends to emit annotations on more types of thing, such as functions, entities/modules and types
  * Providing a means to annotate more things in Haskell with synthesis annotations, such as top level definitions. Ideally such a change will not create unnecessary syntactic weirdness where there are multiple syntax for specifying attributes

martijnbastiaan, 2021-12-03:
I think this calls for something like:

```haskell
{-# ANN f (x :: Attr) #-}
```

alex-mckenna, 2021-12-03:
Yes most likely. On the point of avoiding syntactic weirdness I think it would be a good idea to allow port attributes in the annotations applied to functions though. This would also be more... reliable than the current synthesis annotations which can disappear when you eta-reduce definitions

leonschoorl, 2021-12-03:
Instead of just annotating binders, we could add a `t_attributes :: [Attr]` field to the [Synthesize constructor](https://github.com/clash-lang/clash-compiler/blob/a076fa9e6fdcd0d07f547a1177e5bff28981809d/clash-prelude/src/Clash/Annotations/TopEntity.hs#L254).
That would make it clear it's only supported on synthesis boundaries.
And we wouldn't have to worry what would happen with `x` when `f` gets inlined.
Or that such an annotation would have to imply that `f` shouldn't be inlined.
It would be less flexible, most notably it couldn't work for NOINLINEd polymorphic functions.

To add `Attr`ibutes to directly ports we might add a `PortAttributes [Attr] PortName` constructor to the [PortName type](https://github.com/clash-lang/clash-compiler/blob/a076fa9e6fdcd0d07f547a1177e5bff28981809d/clash-prelude/src/Clash/Annotations/TopEntity.hs#L370).

alex-mckenna, 2021-12-03:
Extending the `Synthesize` constructor is fine if we only care about addressing the current request of synthesis attributes on modules / entities. I'm inclined to say if we're supporting more things anyway we may as well support attributes on data types too, so it's easier to write things like `enum_encoding` attributes without having to hardcode them in blackboxes etc.