clash-lang/clash-compiler #2020: warnNonSynthesizable triggers error even when code is unused

acairncross, 2021-12-08:
I'm using `warnNonSynthesizable` expecting it to error only when actually using (synthesizing) the code labelled as non-synthesizable, e.g. if it's possible/easy to statically determine that the code shouldn't be synthesized then it won't be, but instead it triggers an error. Is this expected behaviour, and if it is, is there another way to annotate non-synthesizable code that would work for this use case?

```haskell
module Test where

import Clash.Prelude
import Clash.Annotations.Primitive

unsupported :: String -> a
unsupported msg =
  error $ msg <> ": unsupported"
{-# ANN unsupported (warnNonSynthesizable "unsupported") #-}

topEntity
  | True = 42
  | otherwise = unsupported "hello"
```

With Clash 1.4:
```
GHC: Parsing and optimising modules took: 0.367s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.000s

<no location info>: error:
    Clash error call:
    No BlackBox definition for 'Test.unsupported' even though this value was annotated with 'HasBlackBox'.
    CallStack (from HasCallStack):
      error, called at src/Clash/Primitives/Util.hs:142:11 in clash-lib-1.4.6-EBw2mxbesS05MW7zTaSOaM:Clash.Primitives.Util
```

With Clash 1.2:
```
GHC: Parsing and optimising modules took: 0.370s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.000s

<no location info>: error:
    Clash error call:
    No BlackBox definition for 'Test.unsupported' even though this value was annotated with 'WarnNonSynthesizable', implying it has a BlackBox.
    CallStack (from HasCallStack):
      error, called at src/Clash/Primitives/Util.hs:137:11 in clash-lib-1.2.5-H3gfsjD1Tn39TYN6Isgnmk:Clash.Primitives.Util
```

alex-mckenna, 2021-12-09:
This error is expected: when you use `warnNonSynthesizable` it implicitly enables `hasBlackBox` which makes Clash throw an error for any black boxes without definitions provided. You can provide this definition by creating a `.primitives` file like those found [here](https://github.com/clash-lang/clash-compiler/tree/master/clash-lib/prims), or by using something like [`InlinePrimitive`](https://hackage.haskell.org/package/clash-prelude-1.4.6/docs/Clash-Annotations-Primitive.html#t:Primitive).

As for why the error is thrown when the black box is trivially removed in normalization: black box definitions are loaded earlier in compiler start-up, so this error is triggered by the definition of `unsupported`, not by it's use in `topEntity` AFAIK.

basile-henry, 2021-12-09:
> black box definitions are loaded earlier in compiler start-up, so this error is triggered by the definition of unsupported, not by it's use in topEntity AFAIK.

I understand it might be easier to implement it this way, but I think it would be much more useful if it could check if it's used in the final generated HDL.

We have the following pattern in our codebase (more or less):

```haskell
somePrimitiveWrapper backend =
  case backend of
    Xilinx
      | SAsynchronous <- resetKind @domain -> ...
    Intel -> ...
    _ -> unsupported ("somePrimitiveWrapper: Backend " <> show backend)
```

With `unsupported` as a simple `error`, it works fine for simulation (in Haskell), and crashes as expected when the conditions aren't met.
Unfortunately, when synthesising if the unsupported branch is chosen (statically), Clash still generates HDL without throwing any errors! And then we can even go all the way to generating a bitstream for the circuit (many hours in some cases), to then find out that it doesn't work as expected (or at all).

It would be really useful to have a way to prevent this type of error. Maybe using `warnNonSynthesizable` for this isn't the right approach. Do you know if there's some other feature we could use? Maybe we could create a custom primitive and error out if it ends up in the generated HDL? Though maybe that's not reliable either.

martijnbastiaan, 2021-12-09:
Yeah, this is expected from an implementation point of view, but I agree this isn't how we should handle it.

@basile-henry As for a workaround right now, how about linking it to a blackbox haskell that always throws an error? 

@acairncross Concerning your code snippet: make sure to mark the function as NOINLINE too, or else Clash might not see it due to GHC optimizations.

acairncross, 2021-12-09:
Ah, I see - part of my confusion was because `unsupported` actually does seem to behave the way I expected (at least some of the time) if it's defined in another package.

rowanG077, 2021-12-09:
@basile-henry In your case I would rewrite it to use custom a `TypeError`. But this is not always easy.

DigitalBrains1, 2022-01-19:
The use case for `warnNonSynthesizable` is for generating HDL for HDL simulation. So if you use such a primitive in your design, it will warn, but if you use it in a test bench, the warning is quelled. I think you're looking for `dontTranslate`. It seems to do what you want. I think the documentation should be more explicit in this regard.

basile-henry, 2022-01-19:
@DigitalBrains1 Thanks, you're right `dontTranslate` does work in this case. I will try in a more complicated environment, but this does look very promising! :smile:  :crossed_fingers:

basile-henry, 2022-01-19:
Could Clash add a `dontTranslate` annotation to `error` itself?

When we need undefined data/initialization, we usually use `errorX`, not `error`. We typically only use `error` for branches where we don't want HDL to be generated.

martijnbastiaan, 2022-01-19:
My guess is that that would break quite a lot of existing designs. For example, pattern matches like `Just x <- foo` or things like `fromJust` use `error` for their "impossible" case.

DigitalBrains1, 2022-01-19:
Custom `TypeError`s are even better, though, your code would just not type check if you tried to instantiate `somePrimitiveWrapper` in an unsupported way. However, you might not always want all the complexity that comes with it.

acairncross, 2022-01-19:
`dontTranslate` is what I was looking for, thanks!

closed 2022-01-19 by acairncross

DigitalBrains1, 2023-01-06:
Hi @basile-henry!

I've incorporated your `clashCompileError` almost literally into our Prelude (well, the PR is still open). Thank you for the good idea! Attribution to you is only in the commit message and PR cover letter; I didn't include copyright attribution. Is this enough for you or would you like more?

basile-henry, 2023-01-06:
@DigitalBrains1 Nice, looks great! :smile:

For the copyright attribution, since it was developed for work, use my employer `Myrtle.ai`

I've seen it in other files in `clash-compiler`, for example: https://github.com/clash-lang/clash-compiler/blob/ba4765139ea0728546bf934005d2d9b77e48d8c7/clash-lib/src/Clash/Primitives/DSL.hs#L2-L4