clash-lang/clash-compiler #2021: `clashi` segfaults on macos

pavel-demin, 2021-12-11:
The following simple test with `clashi` segfaults on macos 12:
```
% cat test.hs 
{-# LANGUAGE DataKinds, NoImplicitPrelude #-}
import Clash.Prelude
topEntity :: Signal System Bit -> Signal System Bit
topEntity = id

% clashi test.hs 
Clashi, version 1.4.6 (using clash-lib, version 1.4.6):
https://clash-lang.org/  :? for help
Loaded package environment from /Users/user/.ghc/x86_64-darwin-8.10.7/environments/default
[1 of 1] Compiling Main             ( test.hs, interpreted )
Ok, one module loaded.
*Main> signal = fromList [0, 1, 0, 1]
*Main> sampleN 4 $ topEntity signal
zsh: segmentation fault  clashi test.hs
```
It runs without problem with `ghci`:
```
% ghci test.hs
GHCi, version 8.10.7: https://www.haskell.org/ghc/  :? for help
Loaded package environment from /Users/user/.ghc/x86_64-darwin-8.10.7/environments/default
[1 of 1] Compiling Main             ( test.hs, interpreted )
Ok, one module loaded.
*Main> signal = fromList [0, 1, 0, 1]
*Main> sampleN 4 $ topEntity signal
[0,1,0,1]
```

martijnbastiaan, 2021-12-12:
How did you install Clash? Did you follow the steps at https://clash-lang.org/install/macos/?

pavel-demin, 2021-12-12:
I initially tried to follow the instructions and install Clash using stack, but I did not like the way stack handles versions of GHC and installs multiple copies of the compiler and libraries.

Currently, I am using ghcup to install GHC and cabal to install Clash. Here are the commands: 
```
curl -L https://get-ghcup.haskell.org -o install-ghcup.sh
sh install-ghcup.sh
cabal install clash-ghc
cabal install --lib ghc-typelits-natnormalise ghc-typelits-extra ghc-typelits-knownnat clash-prelude
```
I use these commands on Debian 11 and macOS 12. On Debian 11 everything works, but on macOS 12 `clashi` segfaults.

martijnbastiaan, 2021-12-12:
Right, we stopped recommending `cabal install` a while ago due to a host of issues associated with it. It still shouldn't segfault of course, but as a quick workaround; perhaps you could try a [starter project](https://github.com/clash-lang/clash-starters) and see if that works? Given that you've already installed stack, you can still use it to initialize the project but use `cabal` from there.

pavel-demin, 2021-12-12:
Since I am only starting to experiment with macOS and Clash, I can remove my entire current installation and reinstall Clash following the instructions. I am now curious to see if this fixes the segfault problem.

martijnbastiaan, 2021-12-12:
Good luck!

pavel-demin, 2021-12-12:
Unfortunately, reinstalling Clash using stack does not fix the segfault problem.

I have reinstalled Clash with the following commands:
```
brew install haskell-stack
stack exec --package clash-ghc -- clash test.hs --vhdl
```
`clashi` still segfaults:
```
% cat test.hs 
{-# LANGUAGE DataKinds, NoImplicitPrelude #-}
import Clash.Prelude
topEntity :: Signal System Bit -> Signal System Bit
topEntity = id

% stack exec --package clash-ghc -- clashi test.hs 
Clashi, version 1.4.6 (using clash-lib, version 1.4.6):
https://clash-lang.org/  :? for help
[1 of 1] Compiling Main             ( test.hs, interpreted )
Ok, one module loaded.
*Main> signal = fromList [0, 1, 0, 1]
*Main> sampleN 4 $ topEntity signal
zsh: segmentation fault  stack exec --package clash-ghc -- clashi test.hs
```

martijnbastiaan, 2021-12-12:
Perhaps you're affected by https://github.com/commercialhaskell/stack/issues/5607?

pavel-demin, 2021-12-12:
Thank you for the link. However, it does not look like exactly the same problem (different OS version, different CPU arch, different errors).

If I build clash-ghc via `git clone` and `cabal build clash-ghc` everything works. So, I cannot reproduce and debug the problem this way.

Looks like only `clashi` built via `stack` or `cabal install` has this problem. `clashi` built via `cabal build` works.

pavel-demin, 2021-12-12:
Looks like my problem is similar to the one discussed in https://github.com/haskell/haskell-language-server/issues/469 and [ghc/ghc#19021](https://gitlab.haskell.org/ghc/ghc/-/issues/19021).

The recommended workaround (enable `executable-dynamic`) works for me.

Here are the commands that I now use to install Clash:
```
curl -L https://get-ghcup.haskell.org -o install-ghcup.sh
sh install-ghcup.sh
sed -i.bak '/executable-dynamic/s/.*/executable-dynamic: True/' ~/.cabal/config
cabal install clash-ghc
cabal install --lib ghc-typelits-natnormalise ghc-typelits-extra ghc-typelits-knownnat clash-prelude
```

martijnbastiaan, 2021-12-12:
Ah, this is something we set in `cabal.project`. This explains why you're seeing it for the Hackage distribution, but not when cloning the repository:

https://github.com/clash-lang/clash-compiler/blob/9e7b9d3d335ae9d1115a6990c59c7cefdac70839/cabal.project#L25-L27

V0XNIHILI, 2022-01-05:
I had the same problem. Removed everything that I installed via HomeBrew and followed the steps from @pavel-demin.

Uninstalled previous installations via:

`brew uninstall ghc haskell-stack cabal-install`
` rm -rf ~/.cabal`
`rm -rf ~/.ghc`
`rm -rf ~/.ghcup`