clash-lang/clash-compiler #2033: Add quiet flag to `clash`

alex-mckenna, 2022-01-13:
I find that sometimes Clash is too chatty with it's messages like:

```
Clash: Compiling ManyEntitiesEqual.top2
Clash: Ignoring previously made caches
Clash: Normalization took 4.008s
Clash: Netlist generation took 0.178s
Clash: Total compilation took 4.313s
```

Ideally there would be an `opt_quiet` field in `ClashOpts` which would prevent such messages when set on the command line.

christiaanb, 2022-01-13:
Clash is a synthesis tool, it's relatively quiet by industry standards ;-)

alex-mckenna, 2022-01-13:
> by industry standards

I thought we were trying to do better than that :wink:

alex-mckenna, 2022-01-13:
> I feel like it shouldn't chat this much by default anyway

My feeling is that if you're running it interactively, it's still nice to see these "marker" type messages that tell you where you are by default. Otherwise the user experience could easily become "is this actually doing anything" which is a bit less friendly to beginners. If we have more information to share than such simple markers I agree we should consider introducing verbosity flags

DigitalBrains1, 2022-01-13:
I agree that it's nice to have a sort of poor-man's progress indicator with these lines that are emitted while it is doing its stuff. I'm inclined towards the `opt_quiet` variant over the verbosity increase variant.

alex-mckenna, 2022-01-13:
For long normalization runs we should just ring the terminal bell at random intervals :bell: 

(please no)

martijnbastiaan, 2022-01-13:
Got you fam:

https://downloads.haskell.org/~ghc/9.0.1/docs/html/users_guide/runtime_control.html#rts-flag--B

DigitalBrains1, 2022-01-13:
> And we only have one message for that...

Well, one per top entity, it's more useful if you have multiple of those.