clash-lang/clash-compiler #2040: Expected values in `assert` become `undefined` when using `-fclash-compile-ultra`

alex-mckenna, 2022-01-20:
In the following testbench, the result of normalization replaces the expected value given to `assert` with an `undefined`. This is not correct, as it changes the behaviour of the testbench under simulation.

```haskell
testBench :: Signal System Bool
testBench = done
  where
    testInput = register clk rst en 0 (testInput + 1)
    expectedOutput =
      outputVerifier' clk rst $(listToVecTH $ [0 :: Unsigned 8 .. 8])
    done = expectedOutput $ ignoreFor clk rst en d1 0 $ topEntity clk testInput
    clk = tbSystemClockGen (not <$> done)
    rst = systemResetGen
    en = enableGen
```

DigitalBrains1, 2022-01-20:
Here's a full MWE:
```haskell
module TBBasic where

import Clash.Explicit.Prelude
import Clash.Explicit.Testbench

topEntity
  :: Signal System (Unsigned 8)
  -> Signal System (Unsigned 8)
topEntity = id
{-# NOINLINE topEntity #-}

testBench :: Signal System Bool
testBench = done
 where
  testInput = register clk rst en 0 (testInput + 1)
  expectedOutput =
    outputVerifier' clk rst $(listToVecTH [0 :: Unsigned 8 .. 3])
  done = expectedOutput $ topEntity testInput
  clk = tbSystemClockGen (not <$> done)
  rst = systemResetGen
  en = enableGen
```

christiaanb, 2022-01-20:
The issue is `caseFlat`:
```
caseFlat {67}
Changes when applying rewrite to:
case GHC.Classes.eqInt
       (GHC.Types.I#[3891110078048108562]
          wild[6989586621679056645][LocalId])
       (GHC.Types.I#[3891110078048108562] 0) of
  _ ->
    case GHC.Classes.eqInt
           (GHC.Types.I#[3891110078048108562]
              wild[6989586621679056645][LocalId])
           (GHC.Types.I#[3891110078048108562] 1) of
      _ ->
        case GHC.Classes.eqInt
               (GHC.Types.I#[3891110078048108562]
                  wild[6989586621679056645][LocalId])
               (GHC.Types.I#[3891110078048108562] 2) of
          _ ->
            case GHC.Classes.eqInt
                   (GHC.Types.I#[3891110078048108562]
                      wild[6989586621679056645][LocalId])
                   (GHC.Types.I#[3891110078048108562] 3) of
              _ ->
                Clash.Normalize.Primitives.undefined
                  @(Clash.Sized.Internal.Unsigned.Unsigned[8214565720323788059]
                      8)
              GHC.Types.True[3891110078048108589]  ->
                Clash.Sized.Internal.Unsigned.fromInteger# @8 8
                  3
          GHC.Types.True[3891110078048108589]  ->
            Clash.Sized.Internal.Unsigned.fromInteger# @8 8
              2
      GHC.Types.True[3891110078048108589]  ->
        Clash.Sized.Internal.Unsigned.fromInteger# @8 8
          1
  GHC.Types.True[3891110078048108589]  ->
    Clash.Sized.Internal.Unsigned.fromInteger# @8 8
      0
Result:
case GHC.Types.I#[3891110078048108562]
       wild[6989586621679056645][LocalId] of
  _ ->
    Clash.Normalize.Primitives.undefined
      @(Clash.Sized.Internal.Unsigned.Unsigned[8214565720323788059]
          8)
  0 ->
    Clash.Sized.Internal.Unsigned.fromInteger# @8 8
      0
  1 ->
    Clash.Sized.Internal.Unsigned.fromInteger# @8 8
      1
  2 ->
    Clash.Sized.Internal.Unsigned.fromInteger# @8 8
      2
  3 ->
    Clash.Sized.Internal.Unsigned.fromInteger# @8 8
      3
```
that will be caseCon'd to:
```
Clash.Normalize.Primitives.undefined
      @(Clash.Sized.Internal.Unsigned.Unsigned[8214565720323788059]
          8)
```
what caseFlat should be doing is create:
```
case GHC.Types.I#[3891110078048108562]
       wild[6989586621679056645][LocalId] of
  GHC.Types.I#[3891110078048108562] x[6989586621679056646][LocalId] -> 
  case x of
  _ ->
    Clash.Normalize.Primitives.undefined
      @(Clash.Sized.Internal.Unsigned.Unsigned[8214565720323788059]
          8)
  0 ->
    Clash.Sized.Internal.Unsigned.fromInteger# @8 8
      0
  1 ->
    Clash.Sized.Internal.Unsigned.fromInteger# @8 8
      1
  2 ->
    Clash.Sized.Internal.Unsigned.fromInteger# @8 8
      2
  3 ->
    Clash.Sized.Internal.Unsigned.fromInteger# @8 8
      3

commit 31789320 (2022-01-21): Extract `Int#` out of `Int` in `caseFlat`

Fixes #2040
PR #2045 (2022-01-21): Extract `Int#` out of `Int` in `caseFlat`
Fixes #2040
commit d47b634a (2022-01-21): Extract `Int#` out of `Int` in `caseFlat` (#2045)

Fixes #2040
closed 2022-01-21 by alex-mckenna