clash-lang/clash-compiler #2052: Deprecate "JSON" blackboxes

martijnbastiaan, 2022-01-25:
After merging https://github.com/clash-lang/clash-compiler/pull/2037 Clash will support reading from YAML blackboxes, in addition to it's (standard breaking) JSON blackboxes. For 1.6 this means that from a user POV no changes are required. A number of things need to happen before we can remove JSON support entirely:

* We need to add deprecation messages to Clash when we detect JSON blackboxes.
* Ideally we'd find a way for the `v16-upgrade-primitives` utility to output a "pretty" ordering of fields (i.e., name, type, template) instead of alphabetical.
* We'd need to remove the last remnants where JSON blackboxes are still forced: `InlinePrimitive`. I don't think we should aim to replace it with a Haskell datatype, instead of simply converting it to a YAML string.

christiaanb, 2022-01-25:
I don’t see why we should force migration. I’d rather we just promote YAML, but keep the JSON option alive (yet unmaintained). We shouldn’t deprecate old user code for reasons like this.

alex-mckenna, 2022-01-26:
I agree, dropping the not-quite-JSON format seems like the kind of big thing which should be held off until some mythical "Clash 2.0" release anyway (i.e. the _very_ distant future). We could possibly "guess" if `InlinePrimitive` is old or new format by whether it starts with an opening `{`, or always try to parse as YAML and fallback to the old parser if it fails

martijnbastiaan, 2022-01-27:
I'm fine with keeping JSON alive, however I believe that without deprecation warnings there's no such thing as:

>  but keep the JSON option alive (yet unmaintained)

We should make it really clear that this is an unmaintained/untested feature and people shouldn't rely on it to keep working - even if we don't intentionally break it.

> seems like the kind of big thing which should be held off until some mythical "Clash 2.0" release anyway

Perhaps overly combative, but we shouldn't do releases like this. We should continuously push to make things better, without confronting users with giant upgrade walls. It almost killed Python.

alex-mckenna, 2022-01-27:
This isn't really the same as python 2 to 3 though. Users will be able to use the new style immediately, but we put a hard cutoff to when they _can't_ use the old style. Python 3 was more of a _now you have to do this big upgrade that you couldn't do in advance_.

Given we have no intention to bump major version number any time soon I don't think it's unreasonable to support both for now but warn that they need to be upgraded next major release. I do agree we should make it clear the old style is deprecated though

alex-mckenna, 2022-01-27:
I think miscommunication, what we normally do as a release, i.e. 1.4 to 1.6 I call a minor release in my head. Dropping in Clash 2 means effectively it's around indefinitely (as we have no real plans for a Clash 2 right now), but I think @martijnbastiaan wants to set some more fixed amount of time before removal (e.g. if it was 10 releases it would be gone by Clash 1.24).

Working on the assumption that we still have users on pre-1.4 releases who define their own primitives, I'm cautious about a complete removal particularly soon even if we do have an upgrade tool. Forcing too quick a change feels somewhat disruptive to me in a bad way

martijnbastiaan, 2022-01-27:
> @martijnbastiaan wanted to keep JSON supported for 1.6, deprecated, and drop support by 1.8.

Not necessarily 1.8, I like having grace periods. I think we should account for people writing libraries and wanting to be compatible with a large range of compiler versions. So I'd propose fully dropping it say 1.14 or so.

> I personally feel it's fine to drop by 1.8, especially when we have an automated tool to convert people's primitives for them. 

Yeah, that's what makes it very simple for me too. I don't like having codepaths we don't test in the compiler. 

> Hmmm, it needs to be done manually for InlinePrimitive still... ah well, I think it's still fine.

I think we should do this automatically too, with a tool like [retrie](https://hackage.haskell.org/package/retrie).

alex-mckenna, 2022-01-27:
I can be very quick to give opinions, I'm just happy to also be in the right ballpark this time

christiaanb, 2022-01-27:
@gergoerdi Does the retroclash book make any references to the JSON format for blackboxes? Deprecating anything in the book is something I want to avoid at great cost.

martijnbastiaan, 2022-01-27:
But Christiaan, the proposed grace period (at least the one Alex proposed) would be ~10 years. Surely we've already dropped the newest GHC we currently support by then so the book would be outdated anyway..

martijnbastiaan, 2022-01-27:
Also _not_ deprecating stuff cuts into the business model of @gergoerdi :smirk: