clash-lang/clash-compiler #2069: trueDualPortBlockRam of size 1 generates __VOID__ address in Verilog

gergoerdi, 2022-02-02:
With something like

```
    (readA, readB) = trueDualPortBlockRam @1 @domA @domB ramA ramB
```

the generated Verilog looks like this:

```
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.5.0. DO NOT MODIFY.
*/
`timescale 100fs/100fs
module Hardware_SpaceInvaders_topEntity_trueDualPortBlockRamWrapper_0
    ( // Inputs
      input  c$pTS0 // clock
    , input  c$pTS1
    , input [8:0] c$pTS3
    , input  c$pTS4 // clock

      // Outputs
    , output wire [17:0] result
    );

  // trueDualPortBlockRam begin
  // Shared memory
  reg [9-1:0] mem [1-1:0];

  reg [8:0] data_slow;
  reg [8:0] data_fast;

  // Port A
  always @(posedge c$pTS0) begin
      data_slow <= mem[__VOID__];
      if(c$pTS1) begin
          data_slow <= c$pTS3;
          mem[__VOID__] <= c$pTS3;
      end
  end

  // Port B
  always @(posedge c$pTS4) begin
      data_fast <= mem[__VOID__];
      if(1'b0) begin
          data_fast <= ({9 {1'bx}});
          mem[__VOID__] <= ({9 {1'bx}});
      end
  end

  assign result = {data_slow, data_fast};
  // end trueDualPortBlockRam

endmodule

```

Using a larger size, where there is more than one possible address, the generated Verilog is fine.

DigitalBrains1, 2022-02-02:
For `Index 1`, the following argument:
```
Signal domA (Index nAddrs) ->   ~ARG[8]
```
for instance renders as `__VOID__` if done as `~ARG[8]`. A fix in the black box could be to use `~IF~SIZE[~TYP[8]]` to special-case this case. I'm not saying that's the best way; it's a way.

Lemme look at other memories for inspiration as well.

alex-mckenna, 2022-02-02:
FWIW, this checking the size to special case is done in other black boxes like `fromInteger`. I don't think it's too unseemly

DigitalBrains1, 2022-02-02:
I'll do it.

closed 2022-02-02 by DigitalBrains1