clash-lang/clash-compiler #2075: CI caching broken

martijnbastiaan, 2022-02-03:
For some reason we end up with empty caches on our CI:

[image]

We should investigate:

 - [ ] Whether this originates from a single runner
 - [ ] Whether there's something wrong with our with our cache server (`diepenheim`)

alex-mckenna, 2022-02-03:
If Diepenheim is low on disk space again, that may explain things

martijnbastiaan, 2022-02-03:
```
/dev/mapper/diepenheim--vg-root  915G  578G  291G  67% /
```

doesn't look like it..

martijnbastiaan, 2022-02-03:
The are the last 5 jobs that have an empty cache:

* https://gitlab.com/clash-lang/clash-compiler/-/jobs/2048388879
* https://gitlab.com/clash-lang/clash-compiler/-/jobs/2048388904
* https://gitlab.com/clash-lang/clash-compiler/-/jobs/2050127584
* https://gitlab.com/clash-lang/clash-compiler/-/jobs/2050127731
* https://gitlab.com/clash-lang/clash-compiler/-/jobs/2050127583

I think I see a pattern..

alex-mckenna, 2022-02-03:
So `gammelke` is polluting the cache? For the time being can we disable it as a runner and confirm that is the problem?

martijnbastiaan, 2022-02-03:
I'm fairly sure this is due to some experimenting Leon and I did trying to reduce SSD wear and tear. I've put in clean config - if this doesn't fix it I'll completely disable it.

martijnbastiaan, 2022-02-04:
I've seen the new config create proper caches!

closed 2022-02-04 by martijnbastiaan