clash-lang/clash-compiler #2107: `cabal install clash-ghc` doesn't work as expected

martijnbastiaan, 2022-02-24:
`cabal install clash-ghc` doesn't work as expected, for example:

```
martijn@gammelke:~$ cabal install clash-ghc
...
martijn@gammelke:~$ clashi
Clashi, version 1.6.1 (using clash-lib, version 1.6.1):
https://clash-lang.org/  :? for help
<command line>: Could not find module ‘GHC.TypeLits.Normalise’
Use -v (or `:set -v` in ghci) to see a list of the files searched for.
```

Although we currently don't advertise it anywhere, users might still expect this to work. After giving it some thought, I'm not sure whether we can really solve this as I don't think `clash-ghc` can know what exact package-ids it depends on. (?)

In any case, we've got a few options:

 * Detect this situation and redirect people to our install page
 * Add a section to the website saying how *not* to install Clash (I'm not convinced this works)
 * ...

christiaanb, 2022-02-24:
Perhaps we can mention:
https://cabal.readthedocs.io/en/3.4/cabal-commands.html#cabal-v2-install

Something along the lines of:
```
$ cabal install —lib clash-ghc —package-env clashenv
$ cabal install clash-ghc
$ clashi -package-env clashenv
```

martijnbastiaan, 2022-02-25:
Interesting, I didn't know Cabal supported environments like this. Still.. environments are a UX disaster. People are going to abuse this to build projects, which leads to all sorts of pain as any Pythonist will (hopefully) tell you.

But! I do think we could leverage this to cheaply build a `clashup` tool. I'll see if I can hack something together soon ish.

martijnbastiaan, 2022-03-24:
I looked into this a few days ago; I think we should consider using a [`cabal-doctest`](https://hackage.haskell.org/package/cabal-doctest)-like strategy to hardcode package ids and package dbs into the binary we generate. I tried looking into it, but it immediately got derailed by [a Cabal bug](https://github.com/haskell/cabal/issues/8043) though, which discouraged me from pushing through :sweat_smile:

martijnbastiaan, 2023-05-21:
Cabal 3.8 has a new feature called [`code-generators`](https://github.com/haskell/cabal/pull/7688). This would allow compilation to access package databases without custom setups. At the moment only `testsuite`s are supported, but I'm assuming this could relatively easily be implemented for `executable`s too. I'll try it out on `doctest-parallel` too see if it works as advertised.