clash-lang/clash-compiler #2124: clash.cabal install from snap fails due to missing libraries

swt2c, 2022-03-16:
I attempted to use the clash snap (version 1.6.2) on Ubuntu 20.04 to install a cabal package, but it fails due to missing libraries:

```
$ clash.cabal install strict
Resolving dependencies...
Build profile: -w ghc-8.6.5 -O1
In order, the following will be built (use -v for more details):
 - base-orphans-0.8.6 (lib) (requires download & build)
 - cabal-doctest-1.0.9 (lib) (requires download & build)
 - indexed-traversable-0.1.2 (lib) (requires download & build)
 - th-abstraction-0.4.3.0 (lib) (requires download & build)
 - distributive-0.6.2.1 (lib) (requires download & build)
 - comonad-5.0.7 (lib:comonad) (requires download & build)
 - bifunctors-5.5.11 (lib) (requires download & build)
 - assoc-1.0.2 (lib) (requires download & build)
 - these-1.1.1.1 (lib) (requires download & build)
 - strict-0.4.0.1 (lib) (requires download & build)
Downloading  base-orphans-0.8.6
Downloaded   base-orphans-0.8.6
Downloading  th-abstraction-0.4.3.0
Starting     base-orphans-0.8.6 (lib)
Downloaded   th-abstraction-0.4.3.0
Downloading  distributive-0.6.2.1
Starting     th-abstraction-0.4.3.0 (lib)
Downloaded   distributive-0.6.2.1
Downloading  indexed-traversable-0.1.2
Downloaded   indexed-traversable-0.1.2
Downloading  cabal-doctest-1.0.9
Starting     indexed-traversable-0.1.2 (lib)
Downloaded   cabal-doctest-1.0.9
Downloading  comonad-5.0.7
Starting     cabal-doctest-1.0.9 (lib)
Downloaded   comonad-5.0.7
Downloading  bifunctors-5.5.11
Downloaded   bifunctors-5.5.11
Downloading  assoc-1.0.2
Downloaded   assoc-1.0.2
Downloading  these-1.1.1.1
Downloaded   these-1.1.1.1
Downloading  strict-0.4.0.1
Downloaded   strict-0.4.0.1
Building     cabal-doctest-1.0.9 (lib)
Building     indexed-traversable-0.1.2 (lib)
Building     base-orphans-0.8.6 (lib)
Building     th-abstraction-0.4.3.0 (lib)

Failed to build base-orphans-0.8.6.
Build log (
/home/stalbert/snap/clash/1215/.cabal/logs/ghc-8.6.5/base-orphans-0.8.6-6810165c18561db86da07dec9947c217410c7d33ff034e9aa94440cf7aa1ea8e.log
):
Configuring library for base-orphans-0.8.6..
Preprocessing library for base-orphans-0.8.6..
Building library for base-orphans-0.8.6..
[1 of 2] Compiling Data.Orphans.Prelude ( src/Data/Orphans/Prelude.hs, dist/build/Data/Orphans/Prelude.o )
[2 of 2] Compiling Data.Orphans     ( src/Data/Orphans.hs, dist/build/Data/Orphans.o )
/snap/clash/1215/usr/bin/ld.gold: error: cannot open /snap/clash/1215/usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libgmp.so: No such file or directory
collect2: error: ld returned 1 exit status
`x86_64-linux-gnu-gcc' failed in phase `Linker'. (Exit code: 1)

Failed to build cabal-doctest-1.0.9.
Build log (
/home/stalbert/snap/clash/1215/.cabal/logs/ghc-8.6.5/cabal-doctest-1.0.9-9fcf36e3351ed87109c1a834ad0b5d492d0dab4a93c52a23b7654b5a6f22bce1.log
):
Configuring library for cabal-doctest-1.0.9..
Preprocessing library for cabal-doctest-1.0.9..
Building library for cabal-doctest-1.0.9..
[1 of 1] Compiling Distribution.Extra.Doctest ( src/Distribution/Extra/Doctest.hs, dist/build/Distribution/Extra/Doctest.o )
/snap/clash/1215/usr/bin/ld.gold: error: cannot open /snap/clash/1215/usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libgmp.so: No such file or directory
collect2: error: ld returned 1 exit status
`x86_64-linux-gnu-gcc' failed in phase `Linker'. (Exit code: 1)

Failed to build indexed-traversable-0.1.2.
Build log (
/home/stalbert/snap/clash/1215/.cabal/logs/ghc-8.6.5/indexed-traversable-0.1.2-eace64dd74f538ba2bdc445d289d87e28ec3e56f64fb75515102e6a6ce6833b6.log
):
Configuring library for indexed-traversable-0.1.2..
Preprocessing library for indexed-traversable-0.1.2..
Building library for indexed-traversable-0.1.2..
[1 of 5] Compiling GhcExts          ( src/GhcExts.hs, dist/build/GhcExts.o )
[2 of 5] Compiling WithIndex        ( src/WithIndex.hs, dist/build/WithIndex.o )
[3 of 5] Compiling Data.Traversable.WithIndex ( src/Data/Traversable/WithIndex.hs, dist/build/Data/Traversable/WithIndex.o )
[4 of 5] Compiling Data.Functor.WithIndex ( src/Data/Functor/WithIndex.hs, dist/build/Data/Functor/WithIndex.o )
[5 of 5] Compiling Data.Foldable.WithIndex ( src/Data/Foldable/WithIndex.hs, dist/build/Data/Foldable/WithIndex.o )
/snap/clash/1215/usr/bin/ld.gold: error: cannot open /snap/clash/1215/usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libgmp.so: No such file or directory
collect2: error: ld returned 1 exit status
`x86_64-linux-gnu-gcc' failed in phase `Linker'. (Exit code: 1)

Failed to build th-abstraction-0.4.3.0.
Build log (
/home/stalbert/snap/clash/1215/.cabal/logs/ghc-8.6.5/th-abstraction-0.4.3.0-eec533fe824406df782d108cbff2047ea026e5017fbdeba684dfd1da1d0dea47.log
):
Configuring library for th-abstraction-0.4.3.0..
Preprocessing library for th-abstraction-0.4.3.0..
Building library for th-abstraction-0.4.3.0..
[1 of 3] Compiling Language.Haskell.TH.Datatype.Internal ( src/Language/Haskell/TH/Datatype/Internal.hs, dist/build/Language/Haskell/TH/Datatype/Internal.o )
[2 of 3] Compiling Language.Haskell.TH.Datatype.TyVarBndr ( src/Language/Haskell/TH/Datatype/TyVarBndr.hs, dist/build/Language/Haskell/TH/Datatype/TyVarBndr.o )
[3 of 3] Compiling Language.Haskell.TH.Datatype ( src/Language/Haskell/TH/Datatype.hs, dist/build/Language/Haskell/TH/Datatype.o )
/snap/clash/1215/usr/bin/ld.gold: error: cannot open /snap/clash/1215/usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libgmp.so: No such file or directory
collect2: error: ld returned 1 exit status
`x86_64-linux-gnu-gcc' failed in phase `Linker'. (Exit code: 1)
cabal-3.2: Failed to build base-orphans-0.8.6 (which is required by
strict-0.4.0.1). See the build log above for details.
Failed to build cabal-doctest-1.0.9 (which is required by strict-0.4.0.1). See
the build log above for details.
Failed to build indexed-traversable-0.1.2 (which is required by
strict-0.4.0.1). See the build log above for details.
Failed to build th-abstraction-0.4.3.0 (which is required by strict-0.4.0.1).
See the build log above for details.
```

Sure enough, there are several broken symlinks in /snap/clash/1215/usr/lib/x86_64-linux-gnu:
```
$ find /snap/clash/1215/usr/lib/x86_64-linux-gnu -xtype l
/snap/clash/1215/usr/lib/x86_64-linux-gnu/libbsd.so
/snap/clash/1215/usr/lib/x86_64-linux-gnu/libffi.so
/snap/clash/1215/usr/lib/x86_64-linux-gnu/libform.so
/snap/clash/1215/usr/lib/x86_64-linux-gnu/libformw.so
/snap/clash/1215/usr/lib/x86_64-linux-gnu/libgmp.so
/snap/clash/1215/usr/lib/x86_64-linux-gnu/libmenu.so
/snap/clash/1215/usr/lib/x86_64-linux-gnu/libmenuw.so
/snap/clash/1215/usr/lib/x86_64-linux-gnu/libpanel.so
/snap/clash/1215/usr/lib/x86_64-linux-gnu/libpanelw.so
/snap/clash/1215/usr/lib/x86_64-linux-gnu/libtic.so
/snap/clash/1215/usr/lib/x86_64-linux-gnu/libyaml.so
```

swt2c, 2022-03-16:
I *think* I understand why this is happening.  According to the [Snapcraft documentation](https://snapcraft.io/docs/build-and-staging-dependencies#heading--filtering), snapcraft is filtering out several library packages (e.g., `libgmp10`) because they are part of the core20 snap.  However, the -dev variants of these packages (e.g., `libgmp-dev`) are NOT part of core20.  Thus, there ends up being a dangling symlink from the `libgmp.so` which refers to the non-dev package.  You would think snapcraft would try to handle fixing dangling symlinks, but it doesn't seem to?  In any event, I *think* this could be worked around by adding the packages to staging so they are not filtered out.

christiaanb, 2022-03-16:
Heh... so the snap release is based on the debian bindist that we create, and we do actually test every bindist that we create in CI. But of course, we then went on assuming that we don't need to test the snap release since it's such a minimal repackaging of the bindist...

Thanks for the report, we'll look into it.

DigitalBrains1, 2022-03-17:
I know next to nothing about snap, but your explanation of what goes wrong sounds likely. As far as I know, only the `-dev` packages install symlinks that end in `.so`. Runtime linking is always against `.so.X` or `.so.X.Y` with X the major version and Y the minor version. So if snapcraft is filtering out the actual libraries and the versioned symlinks, that would leave a dangling unversioned symlink that needs to be taken care of.

Thanks for digging!

PR #2125 (2022-03-31): Fix missing libraries with snap package
When building the snap, snapcraft is filtering out several shared library
packages (e.g., libgmp10) because they are included in the core20 snap.  This
causes a problem when the -dev versions of these packages are used (e.g.,
libgmp-dev) where the clash snap will have broken symbolic links to the
shared library packages.  Work around this issue by staging all the relevant
library packages so they are included in the snap as well.

Fixes #2124.
commit e24bf33a (2022-04-01): Fix missing libraries with snap package (#2125) (#2155)

When building the snap, snapcraft is filtering out several shared library
packages (e.g., libgmp10) because they are included in the core20 snap.  This
causes a problem when the -dev versions of these packages are used (e.g.,
libgmp-dev) where the clash snap will have broken symbolic links to the
shared library packages.  Work around this issue by staging all the relevant
library packages so they are included in the snap as well.

Fixes #2124.

(cherry picked from commit 5454b93ebdf1bdb4205f961c4a0a864260717a7c)
commit 5454b93e (2022-04-07): Fix missing libraries with snap package (#2125)

When building the snap, snapcraft is filtering out several shared library
packages (e.g., libgmp10) because they are included in the core20 snap.  This
causes a problem when the -dev versions of these packages are used (e.g.,
libgmp-dev) where the clash snap will have broken symbolic links to the
shared library packages.  Work around this issue by staging all the relevant
library packages so they are included in the snap as well.

Fixes #2124.
closed 2022-03-31 by martijnbastiaan