clash-lang/clash-compiler #2126: Clash 1.6.2 reports internal bug when generating VHDL

pbreuer, 2022-03-21:
> <no location info>: error:
>     Clash error call:
>     Clash.Normalize.Transformations.Case(442): Report as bug: caseCon
> error: case 1# of
>       GHC.Types.I#[3891110078048108562]
>         (c$sel[170698662] :: GHC.Prim.Int#[3674937295934324764]) ->
>         c$sel[170698662][LocalId]
>     CallStack (from HasCallStack):
>       error, called at src/Clash/Normalize/Transformations/Case.hs:442:3 in
> clash-lib-1.6.2-DM9JJDZux0KEcYQd3mJFPd:Clash.Normalize.Transformations.Case
>

alex-mckenna, 2022-03-21:
Hmm, this seems related to https://github.com/clash-lang/clash-compiler/issues/2097

alex-mckenna, 2022-03-21:
Although `Int` has no special `unpackInt#` primitive, so the fix won't be the same. Regardless, something dodgy must be happening to make a pattern match on a non-bignum literal that expects a datacon... Do you have a reproducer you can share @pbreuer?

christiaanb, 2022-03-21:
https://github.com/clash-lang/clash-compiler/commit/0613f218de8422b276cfb13e1e887ec838b392a1 seems a likely candidate for introducing this bug. Things like https://github.com/clash-lang/clash-compiler/commit/0613f218de8422b276cfb13e1e887ec838b392a1#diff-74a81bc4c6c8bf533d9e18184e8d53a1ee8cd6253ac5578e17e7df8003737bdfR2332 return something of type `Int#` where it should be something of type `Int`.

pbreuer, 2022-03-21:
I'm afraid I have no reproducer (50,000+ lines of code). I might be able to do a bit of binary searching among releases, but there doesn't seem to be any 1.5.* about so the choice is limited!
If the above is a patch, I can try it (I don't use git so this interface is all a bit strange to me).

Thanks for the quick response.

PTB

pbreuer, 2022-03-21:
OK, glad to be of (dis)service. 
If you can isolate a patch and post it or signal it to me somehow I'd rather apply that by hand to what I already have and go on from there to 1.6.3 when that emerges. I have no clue about this git thing.
Otherwise I'll go round changing Int# to Int pretty randomly in the primitives directory!

Thanks again

PTB

alex-mckenna, 2022-03-21:
You can get the patch at https://patch-diff.githubusercontent.com/raw/clash-lang/clash-compiler/pull/2127.patch

pbreuer, 2022-03-21:
On Mon, 21 Mar 2022 07:10:32 -0700
Alex McKenna ***@***.***> wrote:

> You can get the patch at https://patch-diff.githubusercontent.com/raw/clash-lang/clash-compiler/pull/2127.patch

I've bene running for 13m cpu time at 10% cpu with no issues so far.
At 30m cpu I'll call it working. That will be in 3h time.

Peter

pbreuer, 2022-03-21:
1h into the 5.5h (cpu time) synthesis and no problems so far. The patch looks good for 1.6.2. I'll know for sure tomorrow.

Thanks again.

PTB

pbreuer, 2022-03-22:
[](https://github.com/alex-mckenna)No, not solved completely. Other bug exposed or consequent bug created.

This is 1.6.2 patched with Christiaan's solve above. I wasn't tracking overnight but it seems to have got past normalization phase, at the start of which (I think) it was crashing before.

...
Clash: Normalization took 17h37m10s
[WARNING] Dubious primitive instantiation for Clash.Signal.Internal.clockGen: Clash.Signal.Internal.clockGen is not synthesizable! (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.integerToInt: GHC.Integer.Type.integerToInt: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.eqInteger#: GHC.Integer.Type.eqInteger#: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.integerToWord: GHC.Integer.Type.integerToWord: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.wordToInteger: GHC.Integer.Type.wordToInteger: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.smallInteger: GHC.Integer.Type.smallInteger: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
Clash: Netlist generation took 34.982s

: error:
Other error:
src/Clash/Primitives/Sized/Signed.hs:36:7-90: Non-exhaustive patterns in [(Literal _
(NumLit sz),
_, _),
(i@(Identifier iV m), Signed szI, _)]

Timer data says 11404.736u 4169.306s 19:50:39.44 21.8% 0+0k 5490947008+306192io 244895156pf+0w

which puts it at 11400s of cpu time, and it should at that point be half way to finishing. It took 5h30m of cpu time to synthesize under 1.5pre, which is 330m, which is 20000s.

It had produced a zero length KPU_Test_types.vhdl file. Nothing else.

Anyway, I'm sure will be a 10s of looking-at-it fix for you :). I can't even locate what it's pointing at :(.

PTB
@[alex-mckenna](https://github.com/alex-mckenna)
Member
alex-mckenna commented [39 minutes ago](https://github.com/clash-lang/clash-compiler/issues/2128#issuecomment-1075120136)

Hmm, this is an unrelated issue to the one reported yesterday. I'll investigate

christiaanb, 2022-03-22:
@pbreuer it's a separate issue. I've created https://github.com/clash-lang/clash-compiler/tree/valid_bb_context to track down the root cause. It would be helpful if you could compile your code with that version of Clash and then open a new issue with the reported error.

pbreuer, 2022-03-22:
I'll see what I can do. That is 1.7.0pre as master plus one patch, right?

pbreuer, 2022-03-22:
My colleague @swt using 1.7.0pre from master with the extra info on error patch as well as the int/int# corrrection, reports an error not from that but apparently IN netlist generation, rather than "after" as apparently happened for me with 1.6.2.   think this is some normalization failure unconnected with the other (explanation below), but the error is as follows:

Clash: Normalization took 1h36m4s
[WARNING] Dubious primitive instantiation for
Clash.Signal.Internal.clockGen: Clash.Signal.Internal.clockGen is not
synthesizable! (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for
GHC.Integer.Type.integerToInt: GHC.Integer.Type.integerToInt: Integers are
dynamically sized in simulation, but fixed-length after synthesis. Use
carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.eqInteger#:
GHC.Integer.Type.eqInteger#: Integers are dynamically sized in simulation,
but fixed-length after synthesis. Use carefully. (disable with
-fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for
GHC.Integer.Type.integerToWord: GHC.Integer.Type.integerToWord: Integers
are dynamically sized in simulation, but fixed-length after synthesis. Use
carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for
GHC.Integer.Type.wordToInteger: GHC.Integer.Type.wordToInteger: Integers
are dynamically sized in simulation, but fixed-length after synthesis. Use
carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for
GHC.Integer.Type.smallInteger: GHC.Integer.Type.smallInteger: Integers are
dynamically sized in simulation, but fixed-length after synthesis. Use
carefully. (disable with -fclash-no-prim-warn)

KPU/RFC7693/Blake2b.hs:500:1: error:

    Clash.Netlist.BlackBox(680): Can't translate non-tycon type:
Clash.Sized.Internal.Unsigned.Unsigned[8214565720323789649]
      64
    -> Clash.Sized.Internal.Unsigned.Unsigned[8214565720323789649]  
         64

    The source location of the error is not exact, only indicative, as it
is acquired
    after optimizations. The actual location of the error can be in a
function that is
    inlined. To prevent inlining of those functions, annotate them with a
NOINLINE pragma.
    |
500 | blake2b1_unit _ins =
    | ^^^^^^^^^^^^^

That is a function type being translated to vhdl, which obviously can't be done. But the code itself is innoccous and about as heavily instantiated as I can imagine, with no genericity that I can think of, and has synthesised fine under my 1.5pre, so I'm kind of dubious about that whole error. It seems to me it is reporting a function used as a parameter or an outcome, as in partial evaluation, and I wouldn't consciously have written that so I would wait for confirmation as to what error I get shown by the extra patch on 1.6.2

The blake2b1_unit referenced in the error above is my implementation of the
blake2b hash function, which standardly has 10 rounds of two half
rounds each. Instead I implemented it as 2 half-rounds, pseudo-randomly chosen
one from the 10 first-halves and one from the 10 second-halves.

It is surely complaining that though the choice is actually determined
by the data that it is applied to, it hasn't pushed the data through into the choice of
which round, leaving the function to be applied late rather than
early. That would appear to me to be a normalization failure in 1.7.0pre

If my guess is correct on that, that is.

I will advise to replace the hash with something really simple like adding or xoring, plus delay 1, and try again, and see if this time the intended error shows up.

I'm due to know tomorrow what 1.6.2 says.

Regards

PTB

pbreuer, 2022-03-25:
Sorry for the delay. I have the error report with the expanded messaging patch in now. Was that for more data from argument validation failure (I don't grok the template stuff) reporting what exactly it is there isn't a treatment case for? This is 1.6.2:

% clash -XCPP -fconstraint-solver-iterations=0 -package silently -fclash-spec-limit=80 -fclash-inline-limit=80 --vhdl Test/Trace.hs &
[1] 19000
GHC: Setting up GHC took: 0.780s
GHC: Compiling and loading modules took: 25m46s
Clash: Parsing and compiling primitives took 2.185s
GHC+Clash: Loading modules cumulatively took 2h13m39s
Clash: Compiling Test.Trace.kpu_test32
Clash: Normalization took 1d3h23m26s
[WARNING] Dubious primitive instantiation for Clash.Signal.Internal.clockGen: Clash.Signal.Internal.clockGen is not synthesizable! (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.integerToInt: GHC.Integer.Type.integerToInt: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.eqInteger#: GHC.Integer.Type.eqInteger#: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.integerToWord: GHC.Integer.Type.integerToWord: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.wordToInteger: GHC.Integer.Type.wordToInteger: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.smallInteger: GHC.Integer.Type.smallInteger: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)

<no location info>: error:
    Clash error call:
    [(Literal (Just (Unsigned 64,64)) (NumLit 32),Unsigned 64,True),(BlackBoxE "GHC.Integer.Type.wordToInteger" [] [] [] BBTemplate [Text "signed(std_logic_vector(",Arg 0,Text "))"] (Context {bbName = "GHC.Integer.Type.wordToInteger", bbResults = [(Identifier (RawIdentifier "result_6" Nothing []) Nothing,Signed 64)], bbInputs = [(Identifier (RawIdentifier "\\x#\\" Nothing []) Nothing,Unsigned 64,False)], bbFunctions = fromList [], bbQsysIncName = [], bbLevel = 4, bbCompName = UniqueIdentifier {i_baseName = "Test_Trace_kpu_test32_crypto_decryptA", i_baseNameCaseFold = "test_trace_kpu_test32_crypto_decrypta", i_extensionsRev = [], i_idType = Basic, i_hdl = VHDL, i_provenance = []}, bbCtxName = Just "$s$fFiniteBitsSigned_$s$fBitsSigned_$s$fBitsSigned6_result_6"}) True,Signed 64,False)]
    CallStack (from HasCallStack):
      error, called at src/Clash/Primitives/Sized/Signed.hs:29:13 in clash-lib-1.6.2-9DlpiycIEEQav411QKO6Y:Clash.Primitives.Sized.Signed

[1]    Exit 1                        clash -XCPP -fconstraint-solver-iterations=0 -package silently -fclash-spec-limit=80 -fclash-inline-limit=80 --vhdl Test/Trace.hs

Regards

PTB