clash-lang/clash-compiler #2128: Compilation fails on master (but not 1.6.2)

swt2c, 2022-03-21:
I'm seeing this compilation error on master branch (using the binaries built for PR #2126).  This same code compiles fine with 1.6.2.  Is this a bug, or some intended change required for 1.7.x?  Thanks.

```
[44 of 91] Compiling KPU.IMM          ( KPU/IMM.hs, interpreted )

KPU/IMM.hs:65:4: error:
    • Could not determine whether the following type contained a non-ambiguous domain:
      
        (Signal dom (IM_R' (2 ^ l)), Signal domS Pddr,
         Signal domS (Maybe (Pddr, Vec (2 ^ l) Word32)))
      
      In the full type:
      
        Signal dom IM_R
        -> Clock domS
        -> Reset domS
        -> Enable domS
        -> Signal domS (Pddr, Vec (2 ^ l) Word32)
        -> (Signal dom (IM_R' (2 ^ l)), Signal domS Pddr,
            Signal domS (Maybe (Pddr, Vec (2 ^ l) Word32)))
      
      Does it contain one?
      
      ------
      
      This might happen if an instance for TryDomain is missing. Try to determine
      which of the types miss an instance, and add them. Example implementations:
      
       * type instance TryDomain t (MyVector n a)    = TryDomain t a
       * type instance TryDomain t (MyCircuit dom a) = Found dom
       * type instance TryDomain t Terminal          = NotFound
      
      Alternatively, use one of the withSpecific* functions.
      
      ------
      
      You tried to apply an explicitly routed clock, reset, or enable line
      to a construct with, possibly, an implicitly routed one. Clash failed to
      unambigously determine a single domain and could therefore not route it.
      You possibly used one of these sets of functions:
      
       * with{ClockResetEnable,Clock,Reset,Enable}
       * expose{ClockResetEnable,Clock,Reset,Enable}
      
      These functions are suitable for components defined over a single domain
      only. If you want to use multiple domains, use the following instead:
      
       * withSpecific{ClockResetEnable,Clock,Reset,Enable}
       * exposeSpecific{ClockResetEnable,Clock,Reset,Enable}
      
    • In the expression: withReset @dom rst (imm_unit_sys_reset' d l n)
      In an equation for ‘imm_unit’:
          imm_unit rst d l n = withReset @dom rst (imm_unit_sys_reset' d l n)
   |
65 |    withReset @dom rst (imm_unit_sys_reset' d l n)
   |    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

rowanG077, 2022-03-21:
The master branch has enabled the `multiple-hidden` flag which allows clash to work with multiple hidden clocks. But as a trade off typing is worse and you have to implement a few type classes for types. This is indicated in the error message.

If you don't want to use multiple hidden clocks I suggest just setting that flag to False.

See [here](https://github.com/clash-lang/clash-compiler/blob/master/clash-prelude/clash-prelude.cabal#L93) for the flag.

pbreuer, 2022-03-22:
No, not solved completely. Other bug exposed or consequent bug created.

This is 1.6.2 patched with Christiaan's solve above. I wasn't tracking overnight but it seems to have got past normalization phase, at the start of which (I think) it was crashing before.

...
Clash: Normalization took 17h37m10s
[WARNING] Dubious primitive instantiation for Clash.Signal.Internal.clockGen: Clash.Signal.Internal.clockGen is not synthesizable! (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.integerToInt: GHC.Integer.Type.integerToInt: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.eqInteger#: GHC.Integer.Type.eqInteger#: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.integerToWord: GHC.Integer.Type.integerToWord: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.wordToInteger: GHC.Integer.Type.wordToInteger: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.smallInteger: GHC.Integer.Type.smallInteger: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
Clash: Netlist generation took 34.982s

<no location info>: error:
    Other error:
    src/Clash/Primitives/Sized/Signed.hs:36:7-90: Non-exhaustive patterns in [(Literal _
                                                           (NumLit sz),
                                                   _, _),
                                                  (i@(Identifier iV m), Signed szI, _)]

Timer data says 11404.736u 4169.306s 19:50:39.44 21.8%  0+0k 5490947008+306192io 244895156pf+0w

which puts it at 11400s of cpu time, and it should at that point be half way to finishing.  It took 5h30m of cpu time to synthesize under 1.5pre, which is 330m, which is 20000s.

It had  produced a zero length KPU_Test_types.vhdl file. Nothing else.

Anyway, I'm sure will be a 10s of looking-at-it fix for you :). I can't even locate what it's pointing at :(.

PTB

alex-mckenna, 2022-03-22:
Hmm, this is an unrelated issue to the one reported yesterday. I'll investigate and see if I can break it

pbreuer, 2022-03-22:
Sorry - too many open windows. Deleted the post here and moved it  to the 1.6.2 vhdl synthesis issue thread https://github.com/clash-lang/clash-compiler/issues/2126 where it should have been, since I am on 1.6.2 not 1.7.0. My fault. Morning confusion.
So you see it as a different issue? OK.

martijnbastiaan, 2022-03-22:
It's a compile-time only flag.

rowanG077, 2022-03-22:
You can add `constraints: clash-prelude -multiple-hidden` to your `cabal.project` file. There are probably other ways to do it but that works for me.

swt2c, 2022-03-22:
OK, ended up just using the `withSpecific*` functions for now.  Thanks.

closed 2022-03-22 by swt2c

DigitalBrains1, 2022-03-23:
FYI, I think it's easiest to add a file `cabal.project.local` in your checked out repo like this:
```
package clash-prelude
  flags: -multiple-hidden
package clash-testsuite
  flags: -multiple-hidden
```

The file is in `.gitignore`, it should all work fluently until you do a too aggresive `git clean` ;-)

alex-mckenna, 2022-03-23:
Yes, if you use `withSpecificX` then you will have problems when you use Clash compiled with `-multiple-hidden` (i.e. turned off) because I think those definitions are conditionally compiled into `clash-prelude`

swt2c, 2022-03-23:
Will the `cabal.project.local` file work if I am compiling just with `clash` or `clashi` commands?

alex-mckenna, 2022-03-23:
No