clash-lang/clash-compiler #2138: Misleading functions in `clash-prelude-hedgehog`

martijnbastiaan, 2022-03-24:
E.g., `genSigned` can generate results outside of its given range:

https://github.com/clash-lang/clash-compiler/blob/690fbea81758444d752e750964141d3f790677c8/clash-prelude-hedgehog/src/Clash/Hedgehog/Sized/Signed.hs#L28-L34

I think this is misleading at best. We probably want to inspect the given range and generate _its_ extreme values.

alex-mckenna, 2022-03-24:
Yes, that seems reasonable. Do you want to make a PR or should I, it's just a case of changing

```haskell
minBound ~~> Range.lowerBound range
maxBound ~~> Range.upperBound range
```

martijnbastiaan, 2022-03-24:
Go for it

closed 2022-03-28 by leonschoorl

martijnbastiaan, 2022-04-08:
We've released [v1.6.3](https://github.com/clash-lang/clash-compiler/releases/tag/v1.6.3), which includes a fix for this issue.