clash-lang/clash-compiler #2149: Internal error after normalization when synthesising vhdl in 1.6.2

pbreuer, 2022-03-26:
This follows on from issue https://github.com/clash-lang/clash-compiler/issues/2126 . Christiaan's patch cured that bug, but a different issue arises later on in the vhdl syntehsis with 1.6.2 (works fine in some 1.5.0pre) that is also deadly. I presume it happens during netlist generation, but it may be just prior ... or after. Christiaan asked me to re-run with a patch that produces more error reporting, and this is the output:

    % clash -XCPP -fconstraint-solver-iterations=0 -package silently -fclash-spec-limit=80 -fclash-inline-limit=80 --vhdl Test/Trace.hs &
    [1] 19000
    GHC: Setting up GHC took: 0.780s
    GHC: Compiling and loading modules took: 25m46s
    Clash: Parsing and compiling primitives took 2.185s
    GHC+Clash: Loading modules cumulatively took 2h13m39s
    Clash: Compiling Test.Trace.kpu_test32
    Clash: Normalization took 1d3h23m26s
    [WARNING] Dubious primitive instantiation for Clash.Signal.Internal.clockGen: Clash.Signal.Internal.clockGen is not synthesizable! (disable with -fclash-no-prim-warn)
    [WARNING] Dubious primitive instantiation for GHC.Integer.Type.integerToInt: GHC.Integer.Type.integerToInt: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
    [WARNING] Dubious primitive instantiation for GHC.Integer.Type.eqInteger#: GHC.Integer.Type.eqInteger#: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
    [WARNING] Dubious primitive instantiation for GHC.Integer.Type.integerToWord: GHC.Integer.Type.integerToWord: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
    [WARNING] Dubious primitive instantiation for GHC.Integer.Type.wordToInteger: GHC.Integer.Type.wordToInteger: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
    [WARNING] Dubious primitive instantiation for GHC.Integer.Type.smallInteger: GHC.Integer.Type.smallInteger: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)

    : error:
    Clash error call:
    [(Literal (Just (Unsigned 64,64)) (NumLit 32),Unsigned 64,True),(BlackBoxE "GHC.Integer.Type.wordToInteger" [] [] [] BBTemplate [Text "signed(std_logic_vector(",Arg 0,Text "))"] (Context {bbName = "GHC.Integer.Type.wordToInteger", bbResults = [(Identifier (RawIdentifier "result_6" Nothing []) Nothing,Signed 64)], bbInputs = [(Identifier (RawIdentifier "\x#\" Nothing []) Nothing,Unsigned 64,False)], bbFunctions = fromList [], bbQsysIncName = [], bbLevel = 4, bbCompName = UniqueIdentifier {i_baseName = "Test_Trace_kpu_test32_crypto_decryptA", i_baseNameCaseFold = "test_trace_kpu_test32_crypto_decrypta", i_extensionsRev = [], i_idType = Basic, i_hdl = VHDL, i_provenance = []}, bbCtxName = Just "$s$fFiniteBitsSigned_$s$fBitsSigned_$s$fBitsSigned6_result_6"}) True,Signed 64,False)]
    CallStack (from HasCallStack):
    error, called at src/Clash/Primitives/Sized/Signed.hs:29:13 in clash-lib-1.6.2-9DlpiycIEEQav411QKO6Y:Clash.Primitives.Sized.Signed

    [1] Exit 1 clash -XCPP -fconstraint-solver-iterations=0 -package silently -fclash-spec-limit=80 -fclash-inline-limit=80 --vhdl Test/Trace.hs

That is from this line in the patched source (I'll fix the indenting later):

    fromIntegerTF :: TemplateFunction
    fromIntegerTF = TemplateFunction used valid fromIntegerTFTemplate
    where
    used = [0,1]
    valid bbCtx = case bbInputs bbCtx of
    [(Literal _ (NumLit ),,True),v] -> case v of
    (Identifier {}, Signed {}, False) -> True
    (Literal {}, _, True) -> True
    _ -> error (show (bbInputs bbCtx)) <--------- HERE
    _ -> error (show (bbInputs bbCtx))

    fromIntegerTFTemplate
    :: Backend s
    => BlackBoxContext
    -> State s Doc
    fromIntegerTFTemplate bbCtx = getAp $ do
    let [(Literal _ (NumLit sz),,), (i@(Identifier iV m), Signed szI, _)] = bbIn
    puts bbCtx
    case compare sz (toInteger szI) of
    LT -> let sl = Sliced (Signed szI,fromInteger sz-1,0)
    m1 = Just (maybe sl (Nested sl) m)
    in expr False (Identifier iV m1)
    EQ -> expr False i
    GT -> "resize" <> tupled (sequenceA [expr False i
    ,expr False (Literal Nothing (NumLit sz)
    )])

Regards

PTB

pbreuer, 2022-03-28:
None of the errors occur in 1.5.0pre. I'll check that your example above causes no error there.

But it doesn't look likely to me as everything in sight is a stream to stream function, including topEntity. 

{-# ANN kpu_test32
  (Synthesize { t_name   = "KPU_Test"
              , t_inputs = [ ]
              , t_output = PortName "trace"
              }) #-}

One thing I do notice is that the top level is a foo = bar ... where bar :: .... bar = ... . I think there's a reason for that but I forget what. Something to do with specialization.  It looks like a silly thing to do, but it is done, so presumably there is a reason.

Thanks for the suggestion.

PTB

  

I could try locating it by module, but it'll be a lot of work.

pbreuer, 2022-03-29:
I would suggest varying the patch to not produce error but instead just carry on leaving some mark that I can look for in the eventual output, and hence locate where the cause of this is, hopefully.

martijnbastiaan, 2022-03-29:
> I think this is about what's happening:

@leonschoorl Does this happen pre 1.5 too? If it doesn't, that's a pretty good indication this is @pbreuer's issue.

pbreuer, 2022-03-29:
The only earlier version I have compiled up is 1.1.0 on a 32-bit machine.  the 1.5.0pre I have is  built July 7 2021. I can locate it more exactly if you point at  a definitive changelog to look at. There is a CHANGELOG.md.save0 but that goes up to 1.4.2 only (May 18 2021).

I really don't recognize anything like that Word->Signed function at top level. It's all stream to stream. Of course there must be tons like that at a lower level, but I would expect it to be normalized out (my impression is that normalization is failing). Perhaps I should look for inadvertent NOINLINEs that magically were disregarded earlier ...

I can also try bisecting between versions to locate the problem. Meanwhile I'm begging for a patch that carries on regardless where before it errored, but leaves some recognizable mark in the final output. 

Any suggestions?

On a strategic level, the vhdl produced by 1.5.0pre runs fine under simulation in ghdl, for thousands of cycles requiring about 100GB of RAM so there appears to be nothing _wrong_ with what 1.5.0pre does. 

Can you perhaps give me an idea of the abstract problem hinted at here? I may get an idea from that.

Otherwise .. please send patch that leaves a mark in the vhdl output but carries on. I don't care what it does after that point. It could die, as far as I care for this purpose, so long as there is output to look at.

Regards

PTB

martijnbastiaan, 2022-03-29:
> PS. I have just tried synthesizing that Word->Signed function under 1.1.0 and it works fine:

Thanks, that's very helpful! That's a pretty strong indicator that that's the root of the issue. I'm not sure how Clash could carry on after encountering this error, from the looks of it is simpler just fixing this bug than try to figure out how to carry on :-).

pbreuer, 2022-03-29:
A patch that manages to carry on would treat the error case as though the case statement had been provided with the most trivial instance of a non-error case imaginable maybe involving some number like 0x12345678 that I could later search for. Or 0xdeaddead.

Regards

PTB

pbreuer, 2022-03-29:
Actually we shouldn't have believed my success with 1.1.0 but further testing shows it was right, though I was wrong to say so. The issue is that was on a 32-bit machine so Word -> Signed 32 does not involve a size change whereas on a 64-bit machine it does. But I retried with Word -> Signed 16 and it still fine so the result should be believed.

On the 64-bit machine running 1.6.2 I have tried replacing fromIntegral for Word -> Signed 32 with unpack . resize . pack and it synthesizes fine.

I have also tried replacing Word -> Signed 32 with Word -> Signed 64 with fromIntegral still the implementation in 1.6.2 and that still errors.

So the problem seems (superficially) to be with fromIntegral in 1.6.2 .

There is precisely ONE use of fromIntegral in my 50,000 lines of code. I'll try replacing it and report.

leonschoorl, 2022-03-29:
It's caused by this VHDL specific primitive introduce in 2e4096a9fd94192d4f50ba1f27c0279430f0901c
Which was added somewhere between clash 1.4 and 1.6.

It requires its argument to be a [`Identifier`](https://hackage.haskell.org/package/clash-lib-1.6.2/docs/Clash-Netlist-Types.html#t:Expr), and fail for any other type of `Expr`.
Except for the special code path that handles `Literal`: https://github.com/clash-lang/clash-compiler/pull/2131#discussion_r836595929
But all other kinds of [`Expr`'s](https://hackage.haskell.org/package/clash-lib-1.6.2/docs/Clash-Netlist-Types.html#t:Expr) fail.

If only he had an equivalent of `~VAR[n]` for use in `TemplateFunction`, then we could easily force the argument to be an `Identifier`.

pbreuer, 2022-03-29:
Can you do that by assigning a new identifier for the Expr argument of fromInteger, during the translation?

martijnbastiaan, 2022-03-29:
@leonschoorl I think you can use https://hackage.haskell.org/package/clash-lib-1.6.2/docs/Clash-Primitives-DSL.html#v:toIdentifier

pbreuer, 2022-03-29:
I've eliminated explicit uses of fromInteger  in the code but something must remain because I'm still getting a (different) error at that same point. This time by sheer luck a NOINLINE indicates which module this is in and I should be able to narrow it down. It still seems to be unhappy about precisely wordToInteger :

> <no location info>: error:
>    Clash error call:
>    [(Literal (Just (Unsigned 64,64)) (NumLit 32),Unsigned 64,True),(BlackBoxE "
> GHC.Integer.Type.wordToInteger" [] [] [] BBTemplate [Text "signed(std_logic_vector(",Arg 0,Text "))"] (Context {bbName = "GHC.Integer.Type.wordToInteger", bbResults = [(Identifier (RawIdentifier "result_6" Nothing []) Nothing,Signed 64)], bbInputs = [(Identifier (RawIdentifier "\\x#\\" Nothing []) Nothing,Unsigned 64,False)], bbFunctions = fromList [], bbQsysIncName = [], bbLevel = 4, bbCompName = UniqueIdentifier {i_baseName = "Test_Trace_kpu_test32_crypto_decryptA", i_baseNameCaseFold = "test_trace_kpu_test32_crypto_decrypta", i_extensionsRev = [], i_idType = Basic, i_hdl = VHDL, i_provenance = []}, bbCtxName = Just "$s$fFiniteBitsSigned_$s$fBitsSigned_$s$fBitsSigned6_result_6"}) True,Signed 64,False)]
>   CallStack (from HasCallStack):
>     error, called at src/Clash/Primitives/Sized/Signed.hs:29:13 in clash-lib-1.6.2-9DlpiycIEEQav411QKO6Y:Clash.Primitives.Sized.Signed

pbreuer, 2022-03-31:
(partially) reversing patch https://github.com/clash-lang/clash-compiler/commit/2e4096a9fd94192d4f50ba1f27c0279430f0901c allowed vhdl synthesis to complete with 1.6.2.

> GHC: Compiling and loading modules took: 9m0s
> Clash: Parsing and compiling primitives took 0.967s
> GHC+Clash: Loading modules cumulatively took 1h0m49s
> Clash: Compiling Test.Trace.kpu_test32
> Clash: Normalization took 2h0m14s
> Clash: Netlist generation took 23.570s
> Clash: Compiling Test.Trace.kpu_test32 took 2h3m25s
> Clash: Total compilation took 3h4m15s

(I used a bigger computer than before! That is why it is about 10x faster).

The patch was introduced July 18 2021 and the working 1.5.0pre I have been using until 1.6.2 was built July 7, so by luck I happened to have more or less the last build without this particular bug, which explains some things.

The patch introduces Primitives/Sized/Signed.hs as backing for a change in prims/vhdl/Clash_Sized_Internal_Signed.primitives that modified the behaviour of  Clash.Sized.Internal.Signed.fromInteger# . I don't understand that change (it's commented of course but the comment needs more to ping my understanding) but I reversed it. I was not able to drop Signed.hs as something else seems to reference it somewhere else in 1.6.2. I couldn't find what.

Yes, I made the reversal in the .primitives.yaml file instead of the .primitives file as those are no longer what are used nowadays, but that's minor.

(the semantics was "VHDL generated forSigned.fromInteger now truncates, like the Clash simulation, when the result is smaller than the argument" ..... that must be evoked in Word64 -> Signed32 or similar, and how Integer gets into it I don't know, but smaller probably means shorter in bits,  not smaller in value, and it is saying to just truncate instead of anything else, but WHAT else is it that was done and no longer should be done but that I have resuscitated? Was/is the sign bit retained?)

pbreuer, 2022-04-01:
Of technical interest only - but I was not able to pull Primitives/Sized/Signed.hs even though it had been newly introduced in the patch that I reversed because there is a purely formal reference to its fromIntegerTF introduced later in Clash/Driver.hs, as part of the knownTemplateFunctions table.
Remove the reference from there too and one can completely reverse https://github.com/clash-lang/clash-compiler/commit/2e4096a9fd94192d4f50ba1f27c0279430f0901c . It doesn't fix the synthesis any better than it was fixed, but now we know.
That  is not the _correct_ fix, just a working hack. The correct fix is either to use the toIdentifier function  to wrap expressions passed to the template function (above), as identifier expressions are already validated as argument, or work out if anybody really cares when one gets a more generic expression and if not just drop the validation in the template, or just figure out what other expression forms should be validated by the template and do something about each of them.
I know nothing about the templating thing so I am afraid cannot help efficiently.
Regards
PTB

leonschoorl, 2022-04-01:
I'll try to explain what's going on.

Integers in haskell are unbounded in sized.
That's a problem for clash because it wants to pick a fixed size for every type it translates to a HDL type.
It picks 64 bits[^1] to represent Integers in HDL.

Because Integers are unbounded in Haskell and bounded in HDL this can create differences in  behavior between the generated HDL and simulations run in Haskell, when these Integer values under/overflow in HDL.
That's why it's we'd recommend against using Integers in your designs, and why clash generates all these `Integers are dynamically sized in simulation, but fixed-length after synthesis.`-warnings.

The [`Signed.fromInteger#` change](https://github.com/clash-lang/clash-compiler/commit/2e4096a9fd94192d4f50ba1f27c0279430f0901c) was to make the behavior of `Signed.fromInterger#` in VHDL match what it does in Haskell.
What it does is:
- when expanding: it does sign-extension
- when shrinking it: it just truncates higher bits, causing the value to wraparound, so the resulting sign may change

```
clashi> pack (-127 :: Signed 8)
0b1000_0001
clashi> toInteger (-127 :: Signed 8)
-127
clashi> i = toInteger (-127 :: Signed 8)

clashi> -- Shrinking:
clashi> fromInteger i :: Signed 7
1
clashi> pack it
0b000_0001

clashi> -- Expanding:
clashi> fromInteger i :: Signed 9
-127
clashi> pack it
0b1_1000_0001
```

Previously the VHDL implementation incorrectly preserved the sign when shrinking too.

[^1]: actually it picks the native machine word width, which is mostly 64 bits nowadays, but could be 32bits, and this can also be modified with [`-fclash-intwidth`](https://clash-lang.readthedocs.io/en/latest/developing-hardware/flags.html)

pbreuer, 2022-04-01:
Thanks for the explanation Leon!
Integers appear nowhere in my design (of course I don't want unbounded length things) and fromIntegral no longer appeared anywhere (I had used it once as a convenience  to implement a toEnum as I recall, and I replaced it with unpack . resize . pack), yet the error was still appearing. I assume it is an implicit use somewhere. 
Didn't you see that  via your (great!) example of Word 64 -> Signed 32? No use of Integer in sight, yet fromInteger# being used.
Why is  Integer introduced? Surely  the introduction must be temporary and be on the way to being  eliminated? If the error interrupts that sequence, then isn't that a normalisation failure?
Your explanation that "the sign bit is retained" sounds right for what goes wrong when using vhdl resize to do truncation having introduced (signed) Integer as an unexpected intermediate on the way from an Unsigned 64 to a Signed 32 type. The change of Unsigned 64 (Word) to Integer could (wrongly) take large positive to negative number and from then on it is all bad as the vhdl resize flushes the sign bit down with it while doing the shrinking.
But the code doesn't mention Integer so why is it introduced? Moving to Integer loest information on whether the original source type was signed or unsigned so there is always going to be guesswork going on here  and it is hard to imagine it not being  wrong guesswork in at least one case, whatever you do! The information is not there.

Tx

PTB

martijnbastiaan, 2022-04-08:
We've released [v1.6.3](https://github.com/clash-lang/clash-compiler/releases/tag/v1.6.3), which includes a fix for this issue.

pbreuer, 2022-04-08:
Thank you Martijn
I did try it last night and can confirm it "works for me". Synthesis goes all the way.

The colleague who told me about it got a tycon error, however. It sounds to me like a residue of a previous compilation. 

>  KPU/RFC7693/Blake2b.hs:500:1: error:

>   Clash.Netlist.BlackBox(680): Can't translate non-tycon type:
> Clash.Sized.Internal.Unsigned.Unsigned[8214565720323789636]
>      64
>    -> Clash.Sized.Internal.Unsigned.Unsigned[8214565720323789636]  
>         64

That;s impossible. So we'll figure what's up later.

I might be interested in making all the extra warnings go away. My guess is that Int, not Integer, is involved as the variable sized thing it does not like.

> [WARNING] Dubious primitive instantiation for GHC.Integer.Type.smallInteger: GHC.Integer.Type.smallInteger: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)

I wrote the code on a 32 bit machine originally, and later moved it to a 64 bit machine, and I would have done the port by making things like Word over to a fixed length Word everywhere. One can be pretty sure I wouldn't have left anything that can change size by context hanging around ... with the excepton of Int, which is used as an indexing type in so many functions and intermediates (e.g. fromEnum) that I decided to leave it be.

I remember consciously deciding that moving to fixed size Int  wouldn't help as I'd still need it to engage with all the library functions like shiftL.

Could that be the source of those warnings? They don't issue in 1.5.0pre.

Regards

PTB

alex-mckenna, 2022-04-08:
Those warnings appear in designs when you have / use functions which go via `Integer` / `Natural` to get to their result. Most of the time this is benign, and if Clash provides a black box for such a function the warning is suppressed. Before 1.6.0 was released (i.e. some 1.5.0 builds will have it, others won't) we added these warnings to the primitives which were missing them which made it a lot noisier to warn in some cases. It's really an instance of _most of the time this is fine, but in the rare instance it isn't fine you don't want the warning to just be suppressed_. 

So in terms of what's happening during compilation, nothing has changed - Clash is now just more vocal about it. We've already made some more black boxes to reduce these warnings (see https://github.com/clash-lang/clash-compiler/pull/2066).

pbreuer, 2022-04-08:
There is no instance of "Integer" anywhere in the code proper, but there is in the extra TestBench code base that forms part of the test rig, although it should not be in the compiled TestBench executable because it is used in the Elf loader that builds the insides of a PROM (from an Elf executable file) for use at simulation time, and only the PROM should appear in the synthesized code.
I suppose I could demonstrate that with a little better separation of the files ... but logically, I know it so I am not enthused. Perhaps there is some purely syntactic inclusion that I overlooked. I will see if I can make it go away.

Is there something I could look for in the VHDL?

Thanks

PTB

alex-mckenna, 2022-04-08:
You can mark test benches with

```haskell
{-# ANN myTestBench (TestBench 'entityBeingTested) #-}
```

and it should suppress the primitive warnings

pbreuer, 2022-04-08:
I removed all instances of Integer everywhere (it was only used in the Elf loader as a union type of Word 32 and Word 64, so I aliased it to Word and goodbye and good luck to anyone who wants to cross-compile - now it'll always be expecting 64 bit Elf on a 64 bit compile platform, etc). Still the same warnings.
I would use that ANN above but it needs a bit more explanation for me. Is myTestBench  what is also in my ANN  myTestBench (Synthesize ...) statement?
Then what is entityBeingTested if not that too?

Thanks again

PTB

alex-mckenna, 2022-04-08:
Ah, my mistake on that annotation usage: I got the order of the names mixed up. A better example [from our documentation](https://hackage.haskell.org/package/clash-prelude-1.6.3/docs/Clash-Annotations-TopEntity.html) is this:

```haskell
f :: Bool -> Bool
f = -- some entity
{-# ANN f (defSyn "f") #-}
{-# ANN f (TestBench 'g) #-}

g :: Signal Bool
g = -- some test bench
```

The annotation tells Clash _`g` is the test bench for the entity `f`_, similar to how a `Synthesize` annotation tells Clash _this is the HDL interface I want you to generate_. The test bench annotation (as far as I know) is just there so you can suppress things like warnings about non-synthesizable code, since they don't matter for test benches.

With regard to removing `Integer` / `Natural` entirely: this is actually somewhat difficult to do for non-trivial Haskell. There are a lot of functions (such as those which do type conversions) which go via `Integer`, which is why you still see things like the `smallInteger` primitive appear after normalization. When I have time I'll have a look through `clash-prelude` and see if there are more functions we define which use `Integer` which may now be producing these pointless warnings

pbreuer, 2022-04-08:
OK ... am I right in that  you are saying g is the thing I am synthesizing, comprising a test bench composed of a test rig connected with f, the thing notionally being tested?
And the effect will be to suppress warnings in g until the synthesis hits f. Surely I have to NOINLINE f and g  to make sure they are seen? And what do I do about the current annotation saying that g is to be synthesized? Leave it? Remove it? it is:

> {-# ANN kpu_test32
>  (Synthesize { t_name   = "KPU_Test"
>               , t_inputs = [ ]
>               , t_output = PortName "trace"
>               }) #-}

That would be "g".

leonschoorl, 2022-04-13:
> Ah, my mistake on that annotation usage: I got the order of the names mixed up. A better example [from our documentation](https://hackage.haskell.org/package/clash-prelude-1.6.3/docs/Clash-Annotations-TopEntity.html) is this:
> [...]

@alex-mckenna You did get it right the first time. The example in the documentation is wrong.
(Or at least it doesn't match clash's behaviour.)
See #1750 

So the example should be:
```haskell
f = -- some entity
{-# NOINLINE f #-}
{-# ANN f (defSyn "f") #-}

{-# ANN g (TestBench 'f) #-}
g :: Signal Bool
g = -- some test bench testing f
```

And you don't need an extra Synthesize[^1] annotation on `g`. The [TestBench](https://hackage.haskell.org/package/clash-prelude-1.6.3/docs/Clash-Annotations-TopEntity.html#v:TestBench) annotation will cause `g` to be synthesized by itself.
That is because both are constructors of the same type [TopEntity](https://hackage.haskell.org/package/clash-prelude-1.6.3/docs/Clash-Annotations-TopEntity.html#t:TopEntity). And clash will synthesize any top-level binder with an annotation of that type on it.

[^1]: Either directly using [`Synthesize`](https://hackage.haskell.org/package/clash-prelude-1.6.3/docs/Clash-Annotations-TopEntity.html#v:Synthesize), via [`defSyn`](https://hackage.haskell.org/package/clash-prelude-1.6.3/docs/Clash-Annotations-TopEntity.html#v:defSyn) or even via [`makeTopEntity`](https://hackage.haskell.org/package/clash-prelude-1.6.3/docs/Clash-Annotations-TH.html)

DigitalBrains1, 2022-04-13:
I quite like the explicitness of not naming it `f` and `g` but picking Alex's names:
```haskell
{-# ANN myTestBench (TestBench 'entityBeingTested) #-}
```

I'll soon fix this. (I could have sworn we already had. I actually did swear when I noticed we had not.)

commit 13fab3e6 (2022-04-01): Make sure the VHDL BB for Signed.fromInteger can handle any Netlist Expr

Previously it could only handle Identifier.
(And Literal, which is handled seperately in Clash.Backend.VHDL.expr_)

I've also renamed it to make it clear this blackbox is VHDL only.

Fixes #2149
PR #2157 (2022-04-04): Make sure the VHDL BB for Signed.fromInteger can handle any Netlist Expr
Previously it could only handle Identifier.
(And Literal, which is handled seperately in [Clash.Backend.VHDL.expr_](https://github.com/clash-lang/clash-compiler/blob/cb93b418865e244da50e1d2bc85fbc01bf761f3f/clash-lib/src/Clash/Backend/VHDL.hs#L1682-L1685))

I've also renamed it to make it clear this blackbox is VHDL only.

Fixes #2149

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files
commit 5a86ccff (2022-04-07): Make sure the VHDL BB for Signed.fromInteger can handle any Netlist Expr (#2157)

Previously it could only handle Identifier.
(And Literal, which is handled seperately in Clash.Backend.VHDL.expr_)

I've also renamed it to make it clear this blackbox is VHDL only.

Fixes #2149