clash-lang/clash-compiler #2150: Wrong port name annotations for nested tuples

gergoerdi, 2022-03-27:
```
module Bug (topEntity) where

import Clash.Prelude
import Clash.Annotations.TH

topEntity
    :: "clk" ::: Clock System
    -> Signal System
       ( "a" ::: Bit
       , ( "x" ::: ("u" ::: Bit, "v" ::: Bit)
         , "y" ::: ("q" ::: Bit, "p" ::: Bit)
         )
       )
topEntity clk = pure ((0, ((1, 1), (0, 0))))

makeTopEntity 'topEntity
```

With Clash 1.6.2, I get the following Verilog:

```
module topEntity
    ( // Inputs
      input  clk // clock

      // Outputs
    , output wire  a
    , output wire [1:0] x_u
    , output wire [1:0] x_v
    );
```

gergoerdi, 2022-03-27:
Interestingly, the following is an effective workaround:

```
topEntity
    :: "clk" ::: Clock System
    -> Signal System
       ( "a" ::: Bit
       , "" ::: ( "x" ::: ("u" ::: Bit, "v" ::: Bit)
                , "y" ::: ("q" ::: Bit, "p" ::: Bit)
                )
       )
```

leonschoorl, 2022-03-28:
`makeTopEntity` just silently skips over lots of things that aren't named with `:::`

```haskell
baz :: ( "a" ::: BitVector 1
       ,         BitVector 2
       , "c" ::: BitVector 3)
baz = (0,0,1)
makeTopEntity 'baz
```
Results in:
```verilog
module baz
    ( // No inputs

      // Outputs
      output wire [0:0] a
    , output wire [1:0] c
    , output wire [2:0] result_2
    );
```