clash-lang/clash-compiler #2178: Clash generates nested procedural blocks

alex-mckenna, 2022-04-25:
Currently, clash freely generates nested procedural blocks, e.g. (Verilog)

```verilog
always @* begin
  // some verilog code

  always @* begin
    // some more verilog
  end
end
```

This is technically invalid verilog, if we consult the formal grammar (IEEE 1364-2005, §§ A.6.2, A.6.4) we see that this is not allowed, as the `statement` production does not provide a way to derive `initial_construct` or `always_construct`. Despite this, the Verilog tooling we use appears to be somewhat lenient, and will happily simulate / synthesize this anyway.

The same unfortunately cannot be said about VHDL, where nested `process` are also forbidden _and_ the tooling we use will reject designs at elaboration time. Since this needs fixing in VHDL, we should also fix it in (System)Verilog. This means some shuffling around in the `Netlist` types, namely `Declaration` and `Seq`.

christiaanb, 2022-04-25:
Is there a reproducer for this issue?

alex-mckenna, 2022-04-25:
Ah... this is my mistake. On `master` this behaves as expected, but on my current branch it doesn't. Somewhat annoyingly this was caused by me changing the netlist type to remove one footgun, and in the process making a new one to shoot the other foot. Closing in favour of a source note in my branch.

closed 2022-04-25 by alex-mckenna