clash-lang/clash-compiler #2183: How to get code coverage

numero-744, 2022-05-01:
Hello,

I am learning Haskell to be able to use Clash. I am discovering `hpc` for code coverage, but I can't use it with Clash. I cannot tell Clash to generate an instrumented executable running my test bench with `-fhpc` because Clash cannot produce an executable (can it?). I have a file describing the component to test and another one containing its test bench (as you would do in VHDL or SystemVerilog). I have added a `main` function in the test bench, running it to completion:

```haskell
import qualified Data.List as L

main :: Maybe Bool
main = L.find id testBench

testBench :: [Bool]
testBench = sample done
  where
    -- same as tutorial here
```

I'd like to get an hpc `.tix` file after running the test bench, containing coverage data for the design file only. It there a way to do this? Maybe using unit tests? I could not find documentation for it (yet).

Else, how do you do to get code coverage of your designs?

Best regards,
Côme

rowanG077, 2022-05-01:
You can use normal GHC to compile your Clash code to an executable, you just need to have clash-prelude and dependencies in scope. The most straightforward to get coverage of a testbench is to just call it in the main function. You can also use any other unit testing framework if you want to.

See for example the clash-starter project where normal unit tests are run using GHC: https://github.com/clash-lang/clash-starters/tree/main/simple-nix

numero-744, 2022-05-02:
Thanks for the answer. With `ghc` (alias `stack ghc -- $*`) I get errors that I do not get with `clash` (alias `stack run clash -- $*`):

```
% ghc -fhpc src/Example/MAC_tb.hs
[1 of 1] Compiling Example.MAC_tb   ( src/Example/MAC_tb.hs, src/Example/MAC_tb.o )

src/Example/MAC_tb.hs:28:24: error:
    Illegal type: ‘8’ Perhaps you intended to use DataKinds
   |
28 | sim :: Top -> [(Signed 8, Signed 8)] -> [Signed 8]
   |                        ^

src/Example/MAC_tb.hs:28:34: error:
    Illegal type: ‘8’ Perhaps you intended to use DataKinds
   |
28 | sim :: Top -> [(Signed 8, Signed 8)] -> [Signed 8]
   |                                  ^

src/Example/MAC_tb.hs:28:49: error:
    Illegal type: ‘8’ Perhaps you intended to use DataKinds
   |
28 | sim :: Top -> [(Signed 8, Signed 8)] -> [Signed 8]
   |                                                 ^

```

The `MAC_tb.hs` file starts with these lines:

```haskell
module Example.MAC_tb where

import Prelude hiding (not) -- Added to remove an error with ghc but not with clash

import Clash.Prelude
import Clash.Explicit.Testbench

import qualified Data.List as L

import Example.MAC (topEntity, Top)
```

I've started from a Clash starter project (while following the tutorial) so I have a unit test that I can run. I did not manage to integrate my test bench into the unit tests yet, i'll try again. Do you advise to let the test bench in `src/` and then to call it from the unit tests or to add the test bench directly in the unit tests?

rowanG077, 2022-05-02:
So the reason you are getting the error is that Clash enables quite a few [Haskell Language Extensions](https://wiki.haskell.org/Language_extensions). The error you are getting indicates that using number literals requires the `DataKinds` extension. In the [`simple-nix.cabal`](https://github.com/clash-lang/clash-starters/blob/main/simple-nix/simple-nix.cabal#L9) file in the example project I linked earlier you can see that it enables these languages extensions.

Regarding the test bench. Please notice that test benches are synthesizeable. That means they can be ran in a specialized HDL simulator like GHDL or Verilator. Test benches are implemented so that they can be synthesized and ran using a specialized HDL simulator like GHDL or Verilator. You can use them to unit test but I personally think it's nicer to write them in Haskell that might not necessarily be able to synthesized. But that is completely up to you.

christiaanb, 2022-05-02:
Clash can generate binaries just like GHC can. i.e. `clash --make` is the same as `ghc --make`.

numero-744, 2022-05-02:
According to the readme, my project is based on `clash-lang/simple` (it mentions `stack new my-clash-project clash-lang/simple` in the "Getting this project" section). The extension is enabled in my `.cabal` file the same way it is in the `simple-nix.cabal` you mentioned. But the following command does not seem to take that information into account:

```
stack ghc -- -fhpc src/Example/MAC_tb.hs
```

As it does not work, I've added `{-# LANGUAGE DataKinds #-}` on the top of my `MAC_tb.hs` file. Now there are no errors anymore, but I can't find an executable file for `MAC_tb`.

```
% find . -executable -type f
./.stack-work/install/x86_64-linux-tinfo6/6d4836d6feaf180747cf3f03ba6f55b01faf46876da7f5fd489b46cdf7061aeb/8.10.7/lib/x86_64-linux-ghc-8.10.7/libHSbar-0.1-jTGXXgwZyyLWSNTLOvDud-ghc8.10.7.so
./.stack-work/install/x86_64-linux-tinfo6/6d4836d6feaf180747cf3f03ba6f55b01faf46876da7f5fd489b46cdf7061aeb/8.10.7/lib/x86_64-linux-ghc-8.10.7/libHSbar-0.1-JsFN727LafZ6m9nn5URxPP-ghc8.10.7.so
./.stack-work/install/x86_64-linux-tinfo6/6d4836d6feaf180747cf3f03ba6f55b01faf46876da7f5fd489b46cdf7061aeb/8.10.7/bin/clash
./.stack-work/install/x86_64-linux-tinfo6/6d4836d6feaf180747cf3f03ba6f55b01faf46876da7f5fd489b46cdf7061aeb/8.10.7/bin/clashi
./.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/setup/setup
./.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/libHSbar-0.1-jTGXXgwZyyLWSNTLOvDud-ghc8.10.7.so
./.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/clash/clash
./.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/clashi/clashi
```

For the test bench I think it would be comforting to elaborate the test bench to SystemVerilog along with the design to be sure that the generated design is okay.

UPDATE: `clash --make` takes into accounts the `.cabal` file but I still cannot find the generated executable file.

leonschoorl, 2022-05-09:
For GHC/Clash to generate an executable it needs a function `main :: IO ()` in a module called `Main`.
(Files without a module declaration are implicitly called Main)

To make the types work out in your example you could `print` the result of the `find`.

Clash does support the `-fhpc` in combination with `--make` to generate executables that produce `.tix` files when run.
But depending on how clash is installed the resulting executable might not run. (The snap installation has this issue)

It's probably more portable to use the starter project setup and have stack/cabal manage the building and running of the executables.
## Example

On top of the `stack new test-code-cov clash-lang/simple`, I added:
<details>
<summary> click for full details</summary>

`bin/test-code-cov.hs`:
```haskell
import Clash.Prelude
import Mac
import qualified Data.List as L

main :: IO ()
main = print $ L.find id (sample testBench)
```

`src/Mac.hs`:
```haskell
module Mac where

import Clash.Prelude
import Clash.Explicit.Testbench

ma acc (x,y) = acc + x * y

macT acc (x,y) = (acc',o)
  where
    acc' | x > 99 = 42   -- not reached in test?
         | otherwise = ma acc (x,y)
    o    = acc

mac inp = mealy macT 0 inp

topEntity
  :: Clock System
  -> Reset System
  -> Enable System
  -> Signal System (Signed 9, Signed 9)
  -> Signal System (Signed 9)
topEntity = exposeClockResetEnable mac

testBench :: Signal System Bool
testBench = done
  where
    testInput    = stimuliGenerator clk rst $(listToVecTH [(0,0), (1,1) :: (Signed 9,Signed 9),(2,2),(3,3),(4,4)])
    expectOutput = outputVerifier' clk rst $(listToVecTH [0 :: Signed 9,1,5,14,14,14,14])
    done         = expectOutput (topEntity clk rst en testInput)
    en           = enableGen
    clk          = tbSystemClockGen (not <$> done)
    rst          = systemResetGen
```

`test-code-cov.cabal` changes:
```patch
diff --git a/test-code-cov.cabal b/test-code-cov.cabal
index 2ef2271..3a55c75 100644
--- a/test-code-cov.cabal
+++ b/test-code-cov.cabal
@@ -77,8 +77,9 @@ custom-setup
 library
   import: common-options
   hs-source-dirs: src
+  ghc-options: -fhpc
   exposed-modules:
-    Example.Project
+    Mac
   default-language: Haskell2010
 
 -- Builds the executable 'clash', with test-code-cov project in scope
@@ -97,6 +98,14 @@ executable clashi
     ghc-options: -dynamic
   build-depends: base, clash-ghc, test-code-cov
 
+executable test-code-cov
+  main-is: bin/test-code-cov.hs
+  default-language: Haskell2010
+  Build-Depends: base, clash-prelude, test-code-cov
+  ghc-options: -fhpc
+  if !os(Windows)
+    ghc-options: -dynamic
+
 test-suite doctests
   import:           common-options
   type:             exitcode-stdio-1.0
```

</details>

```
stack run test-code-cov
hpc markup test-code-cov.tix
```

[image]

DigitalBrains1, 2022-05-09:
Note that we have a convenience function for the construct
```haskell
main :: IO ()
main = print $ L.find id (sample testBench)
```
like this:
```haskell
main :: IO ()
main = runUntil id testBench
```
which will output something like
```
Signal sampled for N cycles until value True
```

It's documented [here](https://hackage.haskell.org/package/clash-prelude-1.6.3/docs/Clash-Explicit-Signal.html#v:runUntil).

numero-744, 2022-05-09:
Wow thanks a lot for these explanations! Very clear, I could make it work on my example.

I think these excellent explanations are worth to be added into the documentation.

closed 2022-05-11 by numero-744