clash-lang/clash-compiler #2193: VHDL's `bool` hides X-propagation

martijnbastiaan, 2022-05-08:
Consider:

```haskell
bool :: Bool
bool = errorX "help"
```

This produces:

```vhdl
architecture structural of bool is

begin
  result <= true;
end;
```

Even though we don't assign any value to it in the Clash design (well, bottom), the HDL ends up getting a definitive value. As a consequence, it is indistinguishable from `bool = True`.

I therefore believe we should use `std_logic` for `Bool` in HDL. This would give us:

```vhdl
architecture structural of bool is

begin
  result <= '-';
end;
```

which would be much closer to the designer's intent.

christiaanb, 2022-05-09:
You still get bad `X` propagation regardless of this change. Currently we would translate:
```
if e then p else q
```
to:
```
with e select
  q when False,
  p when others;
```
With the proposed change, we would instead translate it to:
```
with e select
  q when '0'
  p when others;
```
i.e. we would still select `p` when `e` is `X`.

martijnbastiaan, 2022-05-09:
Yes, that's definitely related. See the other issue I wrote up on X-propagation: https://github.com/clash-lang/clash-compiler/issues/2194.

christiaanb, 2022-05-09:
But this is inherent to synthesisable VHDL: unless we match on `x` we can't propagate it through a choice construct.

martijnbastiaan, 2022-05-09:
I'd be in favor of using `X` over `-` for choice constructs. This is a conservative approach, but miles better than what we do now. And it wouldn't involve matching on `X`. Please see the other issue and the paper attached.