clash-lang/clash-compiler #2194: Clash case generation hides X-propagation (sometimes)

martijnbastiaan, 2022-05-08:
Consider:

```haskell
ab :: AB -> Bit
ab A = 1
ab B = 0
```

This gets compiled to:

```vhdl
begin
  a_or_b_0 <= Test_ab_types.AB_0'(Test_ab_types.fromSLV(a_or_b));

  with (a_or_b_0) select
    result <= '1' when AB_0'(A),
              '0' when others;
end;
```

This is slightly wrong as it hides X-propagation: if `a_or_b` ends up being `X` (or `-` etc, you get the gist), `result` will happily be assigned to `0`. This is relatively normal in hand-written VHDL designs because it's tedious to correctly, which would look something like:

```vhdl
begin
  a_or_b_0 <= Test_ab_types.AB_0'(Test_ab_types.fromSLV(a_or_b));

  with (a_or_b_0) select
    result <= '1' when AB_0'(A),
              '0' when AB_0'(B),
              'X' when others;
end;
```

however Clash could generate this. This would get us inline with the recommendation's made by Mike Turpin in his [The Dangers of Living with an X (bugs hidden in your Verilog)](https://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.86.5387&rep=rep1&type=pdf):

[image]

Interestingly, Clash doesn't always do this incorrectly. Given:

```haskell
inv :: Bit -> Bit
inv 0 = high
inv 1 = low
```

Clash produces an X-propagating variant:

```
architecture structural of inv is
  signal \c$case_alt\               : std_logic;
  signal \c$case_alt_selection_res\ : boolean;
  signal result_selection_res       : boolean;

begin
  \c$case_alt_selection_res\ <= the_bit = '1';

  \c$case_alt\ <= '0' when \c$case_alt_selection_res\ else
                  '-';

  result_selection_res <= the_bit = '0';

  result <= '1' when result_selection_res else
            \c$case_alt\;

end;
```

christiaanb, 2022-05-09:
That last case is not so surprising. That's because the Haskell is translate to the following Core:

```
\ds -> 
  case fromInteger ds == 0 of 
    { _ -> case fromInteger ds == 1 of 
        { _ -> patError
        ; True -> low}
    ; True -> high}
```

christiaanb, 2022-05-09:
Also, any code that has VHDL enums will hide X-propagation, whether it's the built-in one (`bool`), or the ones we create. i.e. even if we were to translate Haskell's `Bool` to `std_logic`.

Then the following Haskell:
```
data T = A | B

foo x = if x then A else B
```
can only be translated to:
```
with x select
  res <= B when '0',
  res <= A when others;
```
There's no point in adding a third alternative, because there's no more constructors.

The only other alternative would be to add an "X-like" constructor to every enum:
```
type T is A, B, T_X

with x select
  res <= B when '0',
  res <= A when '1'
  res <= T_X when others;
```
and hope logic synthesis will drop branches where `T_X` is created, and drop branches where `T_X` is matched; because if it doesn't we just made circuits larger for the sake of X-prop in simulation.

martijnbastiaan, 2022-05-09:
My suggestion would be to ditch VHDL enums: if they fundamentally hide X-propagation they're incredibly dangerous and should be avoided. We can keep the generated code readable (if you can call it that today..) by using constants.

martijnbastiaan, 2022-05-09:
A more subtle point @lmbollen (indirectly) brought up: how would we deal with catch all patterns in Haskell? Say:

```haskell
f :: Signed 16 -> Signed 16
f 0 = 0
f _ = 1
```

Clash simulation is fine with relation to X-propagation, given that:

```haskell
f (errorX "bla")
```

yields the X. I'm not sure how you would write that in VHDL.

```vhdl
with s select
  res <= 0 when 0,
  res <= 1 when ??????,
  res <= X when others;
```

what you really want to ask at `??????` is: are all the bits I queried in the previous branches clearly defined? If so, then take this branch, else, fall through to the X case.

christiaanb, 2022-05-09:
Somewhat ironically, rendering enums was only introduced recently https://github.com/clash-lang/clash-compiler/pull/1879 We can easily disable rendering enums by making `-fclash-no-render-enums` the default. Though we should add a `-fclash-render-enums` flag then.

DigitalBrains1, 2022-05-09:
> what you really want to ask at ?????? is: are all the bits I queried in the previous branches clearly defined?

http://www.pldworld.com/_hdl/4/_ref/greg_baker/std_logic/1164/is_x.html perhaps?

martijnbastiaan, 2022-05-09:
Nice find! The only thing that would then slip past would be `L` and `H`. Or do those magically get converted to `0` and `1` when comparing?

christiaanb, 2022-05-09:
But synthesis for `is_x` (where it should synthesize to `const false`) is not guaranteed I think.

DigitalBrains1, 2022-05-09:
I think `L` should map to `0` and `H` to `1` for comparisons? I don't know if that is what currently happens, but that is what should happen.

We could make our own `is_x` variant that uses `pragma translate`, and invoke that function wherever we need it.

martijnbastiaan, 2022-05-09:
In IEEE Std 1076™-2008 I found:

[image]

So something like:

```
with s select
  res <= 0 when 0,
  res <= 1 when not(is_x(s)),
  res <= X when others;
```

should work with synthesis.

alex-mckenna, 2022-05-09:
Worth noting that this isn't in the '93 spec

christiaanb, 2022-05-09:
@martijnbastiaan I don't see how the section you linked relates at all to the example you gave: `s` is not a static metalogical value, nor is `is_x(s)`, nor is `not(is_x(s)))`.

christiaanb, 2022-05-09:
Also, that example won't work, `with .... select ... ` is like pattern matching. `not (is_x(s))` is of type `bool`, while `s` isn't. At best you can do:
```
with s select
  res <= 0 when 0,
  res_d <= when others;

res_d <= 1 when not(is_x(s)) else X;
```

DigitalBrains1, 2022-05-09:
> I don't see how the section you linked relates at all 

It is also exclusively about relational operators. I don't know if `with` ... `select` is somehow syntactic sugar for relational expressions, though?

DigitalBrains1, 2022-05-09:
I guess another interesting question is: does
```
with s select
  res <= 0 when 0,
[...]
```
also select that branch if `s` is `L`? Because that's what hardware would do. Section 16.8.2.4.2 might be about this, or it might not be:

[image]

martijnbastiaan, 2022-05-09:
I'm assuming `is_x` is implemented in terms of these operations. 

> Also, that example won't work, with .... select ... is like pattern matching. 

I generally write pseudo-code VHDL on GitHub issues; I think you got the gist!

christiaanb, 2022-05-09:
It is not (https://www.csee.umbc.edu/portal/help/VHDL/packages/std_logic_1164.vhd):
```
FUNCTION Is_X ( s : std_ulogic        ) RETURN  BOOLEAN IS
    -- pragma subpgm_id 235
    BEGIN
        CASE s IS
            WHEN 'U' | 'X' | 'Z' | 'W' | '-' => RETURN TRUE;
            WHEN OTHERS => NULL;
        END CASE;
        RETURN FALSE;
    END;
```

martijnbastiaan, 2022-05-09:
Well, we'll have to pull it through some simulators/synthesizers. I think this discussion is -while ontopic- getting a bit much though. These are all implementation details I'm sure we'll figure out along the way.