clash-lang/clash-compiler #2200: Haddock before GHC 8.10.5 doesn't support dashes in anchors

DigitalBrains1, 2022-05-10:
We have this in our documentation:
https://github.com/clash-lang/clash-compiler/blob/6cf8edbdaf2a1865f449b788a6f1221064231fb2/clash-prelude/src/Clash/Explicit/Testbench.hs#L79
https://github.com/clash-lang/clash-compiler/blob/6cf8edbdaf2a1865f449b788a6f1221064231fb2/clash-prelude/src/Clash/Prelude/Testbench.hs#L59

With GHC before 8.10.5 (Haddock reports version 2.24.2 in 8.10.5), linking to that with a module-link doesn't render the link:
https://github.com/clash-lang/clash-compiler/blob/6cf8edbdaf2a1865f449b788a6f1221064231fb2/clash-prelude/src/Clash/Explicit/Signal.hs#L876-L877
renders [like this](https://hackage.haskell.org/package/clash-prelude-1.7.0/candidate/docs/Clash-Explicit-Signal.html#v:runUntil) (or if you're looking at this issue once the issue has been fixed, I suppose 1.6.2 will still show the [offending behaviour](https://hackage.haskell.org/package/clash-prelude-1.6.2/docs/Clash-Explicit-Signal.html#v:runUntil)).

Interestingly, page-relative links like
https://github.com/clash-lang/clash-compiler/blob/6cf8edbdaf2a1865f449b788a6f1221064231fb2/clash-prelude/src/Clash/Explicit/Testbench.hs#L227-L228
[render fine](https://hackage.haskell.org/package/clash-prelude-1.7.0/candidate/docs/Clash-Explicit-Testbench.html#v:outputVerifier). Probably because that is just a plain HTML link, not a Haddock module link.

The problem is the dash in the anchor, if you remove that it all works. However, updating Haddock to the version bundled with GHC 8.10.5 or newer _also_ fixes the issue, so I think we should update the version of GHC we use in `build_sdist.sh` in CI. If this is problematic, I guess we can also get rid of the dash.

DigitalBrains1, 2022-05-10:
So you're saying get rid of the dash? No problem, I just don't like to paper over bugs instead of fixing them, but sometimes you just have to suck it up.

alex-mckenna, 2022-05-10:
Actually, you could upgrade to building the docs with `ghc-9.0.2` since it's both supported by Clash _and_ has the documentation [uploaded to hackage](https://hackage.haskell.org/package/ghc-9.0.2)

alex-mckenna, 2022-05-10:
But Clash doesn't support GHC 9.2.x so any other versions are verboden

alex-mckenna, 2022-05-10:
Oh, apparently 8.10.7 also has the docs up, that wasn't the case for a _very_ long time... I would probably still switch to using 9.0.2 though

commit 943065df (2022-05-10): CI: Upgrade 8.10 to 8.10.7, Haddock to 9.0.2

Now that the ghc package documentation is on Hackage for 8.10.7, we can
upgrade our build version.

Furthermore, we now build Haddock documentation for Hackage with 9.0.2
now.

This fixes a Haddock bug we encountered:

Fixes #2200
commit 426cf3b7 (2022-05-10): CI: Upgrade 8.10 to 8.10.7, Haddock to 9.0.2

Now that the ghc package documentation is on Hackage for 8.10.7, we can
upgrade our build version.

Furthermore, we now build Haddock documentation for Hackage with 9.0.2
now.

This fixes a Haddock bug we encountered:

Fixes #2200
commit eb2d053d (2022-05-12): CI: Upgrade 8.10 to 8.10.7, Haddock to 9.0.2

Now that the ghc package documentation is on Hackage for 8.10.7, we can
upgrade our build version.

Furthermore, we now build Haddock documentation for Hackage with 9.0.2
now.

This fixes a Haddock bug we encountered:

Fixes #2200
commit de2e0cf9 (2022-05-30): CI: Upgrade 8.10 to 8.10.7, Haddock to 9.0.2

Now that the ghc package documentation is on Hackage for 8.10.7, we can
upgrade our build version.

Furthermore, we now build Haddock documentation for Hackage with 9.0.2
now.

This fixes a Haddock bug we encountered:

Fixes #2200
closed 2022-05-13 by martijnbastiaan