clash-lang/clash-compiler #2223: GitHub Actions jobs for external contibuters fail

alex-mckenna, 2022-05-27:
#2221 shows GitHub actions workflows that pull the `clash-ci` image from `ghcr.io` all fail to pull the image. This has apparently slipped under the radar since these jobs are less frequently executed.

martijnbastiaan, 2022-05-29:
Why does those jobs run under my username..?

```
  /usr/bin/docker --config /home/runner/work/_temp/.docker_0a94d0bf-b928-48c6-be26-24f6c6307282 login ghcr.io -u martijnbastiaan --password-stdin
```

alex-mckenna, 2022-05-29:
They use `${{ secrets.GITHUB_TOKEN }}` which then apparently uses the creds of whoever started the job. See the previous failing runs with

```
/usr/bin/docker --config /home/runner/work/_temp/.docker_7fbf77e1-4c24-4129-85c1-d1d3c716b775 login ghcr.io -u alex-mckenna --password-stdin
```

and

```
/usr/bin/docker --config /home/runner/work/_temp/.docker_2ec5be1c-ca4f-43c0-b14a-eedc0dc1187e login ghcr.io -u felixonmars --password-stdin
```

commit e056c75c (2023-01-04): Correct GHC versions on GHA

Fixes #2223