clash-lang/clash-compiler #2233: Netlist generation fails with error: `TyCon has no DataCons ... "GHC.Natural.Natural"`

rowanG077, 2022-06-01:
In a design I'm working on I hit the following error when compiling:

```
<no location info>: error:
    Clash error call:
    Clash.Core.Util(718): Can't create selector ("Clash.Normalize.Transformations.ANF(384): doPatBndr",1,0) for: (x[LocalId] :: Natural)
    Additional info: TyCon has no DataCons: Name {nameSort = User, nameOcc = "GHC.Natural.Natural", nameUniq = 3674937295934324786, nameLoc = UnhelpfulSpan "<no location info>"} Natural
    CallStack (from HasCallStack):
      error, called at src/Clash/Core/Util.hs:735:35 in clash-lib-1.7.0-inplace:Clash.Core.Util
      mkSelectorCase, called at src/Clash/Normalize/Transformations/ANF.hs:384:24 in clash-lib-1.7.0-inplace:Clash.Normalize.Transformations.ANF
      collectANF, called at src/Clash/Normalize/Transformations/ANF.hs:158:36 in clash-lib-1.7.0-inplace:Clash.Normalize.Transformations.ANF
      makeANF, called at src/Clash/Normalize/Transformations/ANF.hs:140:9 in clash-lib-1.7.0-inplace:Clash.Normalize.Transformations.ANF
      makeANF, called at src/Clash/Normalize/Transformations/ANF.hs:140:9 in clash-lib-1.7.0-inplace:Clash.Normalize.Transformations.ANF
      makeANF, called at src/Clash/Normalize/Strategy.hs:43:73 in clash-lib-1.7.0-inplace:Clash.Normalize.Strategy
```

After some debugging I was able to create a standalone reproducer:

```
module Bla where

import Clash.Prelude
import Data.Constraint
import Unsafe.Coerce (unsafeCoerce)

newtype CounterOverflow (n :: Nat) (o :: Nat)
  = CounterOverflow (Unsigned (CLog 2 (Max n o)))
  deriving (Eq, Ord, Generic, NFDataX)

type Counter (n :: Nat) = CounterOverflow n n

maxAxiomL
  :: forall (n :: Nat) (p :: Nat) (q :: Nat)
   . (n <= p) :- (n <= Max p q)
maxAxiomL = Sub (unsafeCoerce (Dict :: Dict (a ~ a)))

tick
  :: forall (n :: Nat)
            (o :: Nat)
   . KnownNat n
  => KnownNat o
  => 1 <= n
  => CounterOverflow n o
  -> CounterOverflow n o
tick (CounterOverflow i) = case maxAxiomL @1 @n @o of
                             Sub Dict -> CounterOverflow (i + 1)

data Spec = Spec Nat Nat

type family GetParam1 (s :: Spec) :: Nat where
  GetParam1 ('Spec v _) = v

type family GetParam2 (s :: Spec) :: Nat where
  GetParam2 ('Spec _ v) = v

type KnownSpecBase (s :: Spec)
  = ( KnownNat (GetParam1 s)
    , KnownNat (GetParam2 s)
    )

type KnownSpec (s :: Spec)
  = ( KnownSpecBase s
    , 1 <= GetParam1 s
    , 1 <= GetParam2 s
    )

data Err s
   = Err1 (Counter (GetParam1 s))
   | Err2 (Counter (GetParam2 s))

errT
  :: forall (dom :: Domain)
            (spec :: Spec)
   . KnownSpec spec
  => Err spec
  -> Err spec
errT (Err1 t) = Err1 (tick t)
errT (Err2 t) = Err2 (tick t)
```

with a `topEntity` like this:

```
topEntity
  :: Signal System (Err ('Spec 2 2))
  -> Signal System (Err ('Spec 2 2))
topEntity s = errT @System @('Spec 2 2) <$> s
```

It's relatively finicky though. For example a few things that "fix" it:
- Removing the `Err1` or `Err2` cases in `errT`
- Not using both Params in the `Err` data type
- Not using two constraints and merging `KnownSpec` and `KnownSpecBase`
- Removing the `CLog` inside the `CounterOverflow`
- Removing the `Max` inside the `CounterOverflow`

There are probably more but I encountered those while trying to further simplify it.

rowanG077, 2022-06-01:
See this gist for the normalization output with `DebugApplied` output: https://gist.github.com/rowanG077/37f3e18ff4398acb45a92d4cc600a051

It seems to error out when it tries to normalize `$fOrdNatural_$cmax` which is introduced by the `tick` function. I would think `$fOrdNatural_$cmax` is something like the `max` function from the `Ord` instance of `Natural`. Which probably means there should be an evaluator rule for that?

alex-mckenna, 2022-06-01:
> I'm using GHC 8.10.7

Out of interest, do you see the same behaviour on 9.0.2? If you don't have it installed then I can try it tomorrow on my machine since I use 9.0.2

> I would think $fOrdNatural_$cmax is something like the max function from the Ord instance of Natural

It is

> Which probably means there should be an evaluator rule for that?

I'd be hesitant to patch it up like this, since `max` in many `Ord` instances (including the one for `Natural` from `ghc-bignum`) simply has a default implementation like so

```haskell
max x y = if x <= y then y else x
```

which Clash should already be able to handle for every type with no special-casing

alex-mckenna, 2022-06-01:
I wonder if the lack of a branch for `GHC.TypeLits.Extra.Max` [here](https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/Clash/Core/Type.hs#L500) might have something to do with it :thinking: Either way it seems like an oversight that `Max` and `Min` aren't included here

christiaanb, 2022-06-01:
> I wonder if the lack of a branch for `GHC.TypeLits.Extra.Max` [here](https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/Clash/Core/Type.hs#L500) might have something to do with it 🤔 Either way it seems like an oversight that `Max` and `Min` aren't included here

Not an oversight, unlike the other type functions they aren’t magically defined, but build out of other type functions: https://github.com/clash-lang/ghc-typelits-extra/blob/fe80881503c8ba80c86f17fbdaf481dcd1304d44/src/GHC/TypeLits/Extra.hs#L115

alex-mckenna, 2022-06-01:
Ah, of course they are. This is what I get for late night compiler debugging :sweat_smile: 

I tried the reproducer on 9.0.2 as well, I still can't reproduce locally

christiaanb, 2022-06-01:
The problem is:
```
DEC {43}
Changes when applying rewrite to:
case eta[LocalId] of
  Err1
    (t :: CounterOverflow (GetParam1 (Spec 2 2)) (GetParam1 (Spec 2 2))) ->
    Err1 @(Spec 2 2)
      (tick[GlobalId] 2 2
         (Eq# @Bool @(<=? 1 (GetParam1 (Spec 2 2))) @True
            (_CO_ @(~# Bool Bool (<=? 1 (GetParam1 (Spec 2 2))) True)))
         t[LocalId])
  Err2
    (t :: CounterOverflow (GetParam2 (Spec 2 2)) (GetParam2 (Spec 2 2))) ->
    Err2 @(Spec 2 2)
      (tick[GlobalId] 2 2
         (Eq# @Bool @(<=? 1 (GetParam2 (Spec 2 2))) @True
            (_CO_ @(~# Bool Bool (<=? 1 (GetParam2 (Spec 2 2))) True)))
         t[LocalId])
Result:
letrec
  c$tickOut :: CounterOverflow 2 2
  = letrec
      c$tupIn :: (,,,)
                   Natural
                   Natural
                   (~ Bool (<=? 1 (GetParam1 (Spec 2 2))) True)
                   (CounterOverflow (GetParam1 (Spec 2 2)) (GetParam1 (Spec 2 2)))
      = case eta[LocalId] of
          Err1
            (t :: CounterOverflow (GetParam1 (Spec 2 2)) (GetParam1 (Spec 2 2))) ->
            (,,,) @Natural @Natural @(~ Bool (<=? 1 (GetParam1 (Spec 2 2))) True)
              @(CounterOverflow (GetParam1 (Spec 2 2)) (GetParam1 (Spec 2 2)))
              2
              2
              (Eq# @Bool @(<=? 1 (GetParam1 (Spec 2 2))) @True
                 (_CO_ @(~# Bool Bool (<=? 1 (GetParam1 (Spec 2 2))) True)))
              t[LocalId]
          Err2
            (t :: CounterOverflow (GetParam2 (Spec 2 2)) (GetParam2 (Spec 2 2))) ->
            (,,,) @Natural @Natural @(~ Bool (<=? 1 (GetParam1 (Spec 2 2))) True)
              @(CounterOverflow (GetParam1 (Spec 2 2)) (GetParam1 (Spec 2 2)))
              2
              2
              (Eq# @Bool @(<=? 1 (GetParam2 (Spec 2 2))) @True
                 (_CO_ @(~# Bool Bool (<=? 1 (GetParam2 (Spec 2 2))) True)))
              t[LocalId]
  in tick[GlobalId]
       (case c$tupIn[LocalId] of
          (,,,)
            (c$sel :: Natural)
              (c$wild :: Natural)
              (c$wild :: ~ Bool (<=? 1 (GetParam1 (Spec 2 2))) True)
              (c$wild :: CounterOverflow
                           (GetParam1 (Spec 2 2))
                           (GetParam1 (Spec 2 2))) ->
            c$sel[LocalId])
       (case c$tupIn[LocalId] of
          (,,,)
            (c$wild :: Natural)
              (c$sel :: Natural)
              (c$wild :: ~ Bool (<=? 1 (GetParam1 (Spec 2 2))) True)
              (c$wild :: CounterOverflow
                           (GetParam1 (Spec 2 2))
                           (GetParam1 (Spec 2 2))) ->
            c$sel[LocalId])
       (case c$tupIn[LocalId] of
          (,,,)
            (c$wild :: Natural)
              (c$wild :: Natural)
              (c$sel :: ~ Bool (<=? 1 (GetParam1 (Spec 2 2))) True)
              (c$wild :: CounterOverflow
                           (GetParam1 (Spec 2 2))
                           (GetParam1 (Spec 2 2))) ->
            c$sel[LocalId])
       (case c$tupIn[LocalId] of
          (,,,)
            (c$wild :: Natural)
              (c$wild :: Natural)
              (c$wild :: ~ Bool (<=? 1 (GetParam1 (Spec 2 2))) True)
              (c$sel :: CounterOverflow
                          (GetParam1 (Spec 2 2))
                          (GetParam1 (Spec 2 2))) ->
            c$sel[LocalId])
in case eta[LocalId] of
     Err1
       (t :: CounterOverflow (GetParam1 (Spec 2 2)) (GetParam1 (Spec 2 2))) ->
       Err1 @(Spec 2 2) c$tickOut[LocalId]
     Err2
       (t :: CounterOverflow (GetParam2 (Spec 2 2)) (GetParam2 (Spec 2 2))) ->
       Err2 @(Spec 2 2) c$tickOut[LocalId]
```
which is preventing specialization of `tick` on those `2` values.

christiaanb, 2022-06-01:
Perhaps something is going wrong inside https://github.com/clash-lang/clash-compiler/blob/61c13961d50ddb5c4e0da5daae447d491e42659e/clash-lib/src/Clash/Normalize/Transformations/DEC.hs#L552 

Since DEC has codepaths to ensure that shared arguments (such as those `2`s) don’t become part of the selection/projection tuple.

rowanG077, 2022-06-01:
Tiny bit smaller reproducer:

```
module Bla where

import Clash.Prelude
import Data.Constraint
import Unsafe.Coerce (unsafeCoerce)

newtype CounterOverflow (n :: Nat) (o :: Nat)
  = CounterOverflow (Unsigned (CLog 2 (Max n o)))
  deriving (Eq, Ord, Generic, NFDataX)

type Counter (n :: Nat) = CounterOverflow n n

maxAxiomL
  :: forall (n :: Nat) (p :: Nat) (q :: Nat)
   . (n <= p) :- (n <= Max p q)
maxAxiomL = Sub (unsafeCoerce (Dict :: Dict (a ~ a)))

tick
  :: forall (n :: Nat)
            (o :: Nat)
   . KnownNat n
  => KnownNat o
  => 1 <= n
  => CounterOverflow n o
  -> CounterOverflow n o
tick (CounterOverflow i) = case maxAxiomL @1 @n @o of
                             Sub Dict -> CounterOverflow (i + 1)

type KnownSpecBase n p
  = ( KnownNat n
    , KnownNat p
    )

type KnownSpec n p
  = ( KnownSpecBase n p
    , 1 <= n
    , 1 <= p
    )

data Err n p
   = Err1 (Counter n)
   | Err2 (Counter p)

errT
  :: forall (dom :: Domain)
            (n :: Nat)
            (p :: Nat)
   . KnownSpec n p
  => Err n p
  -> Err n p
errT (Err1 t) = Err1 (tick t)
errT (Err2 t) = Err2 (tick t)
```

martijnbastiaan, 2022-08-30:
We've released [v1.6.4](https://github.com/clash-lang/clash-compiler/releases/tag/v1.6.4), which includes a fix for this issue.