clash-lang/clash-compiler #2242: Vivado does not accept clock names with $ in Verilog

lmbollen, 2022-06-16:
Even though ```$``` is an accepted character in Verilog's basic identifiers. Vivado does not accept it in clock names. 
``` [Vivado 12-2270] Clock names may not contain tcl special characters: '"{};$# - Skipping 'c$arg' [/filepath]``

Possible fixes

- [ ]  Adjust -fclash-hdlsyn Vivado to not generate ```$``` in basic identifiers.
- [ ] Generate more basic identifiers by default (excluding tcl special characters?).
- [ ] Never generate tcl special characters in clock names.
- [ ] Change the ```c$``` prefix.

DigitalBrains1, 2022-06-16:
Note that changing the `c$` prefix doesn't help; if you use an implicit clock, the argument is named like `$d(%,%)` which maps to the basic Verilog identifier `_$d_0` exhibiting the same issue.

alex-mckenna, 2023-06-10:
We're going to work on fixing this at ZuriHac

alex-mckenna, 2023-06-10:
Changing the ID generation is indeed a little hairy here. The identifiers which cause problems ultimately come from `unsafeFromCoreId`, meaning we don't do the sanitising that we get from `makeBasic` etc. We probably don't want to just change this function to strip out characters like `$` because some names might be specified by users directly in things like synthesize annotations.

Perhaps a solution could be to add a flag to actually do this mangling, i.e. `-fclash-sanitise-netlist-ids`, which does this stripping of potentially troublesome characters with the caveat that user-specified names that end up going through `unsafeFromCoreId` may no longer stay as exactly what the user specified.