clash-lang/clash-compiler #2260: `traceVecSignal1` with `dumpVCD` never finishes

rowanG077, 2022-06-29:
Test with GHC 8.10.7 and current master

Reproducer:

```
module Infinite where

import Clash.Explicit.Prelude hiding (writeFile)
import Data.Text.IO (writeFile)

infinite :: IO ()
infinite = do
  let s = register systemClockGen resetGen enableGen (True :> Nil) s
  let traced = traceVecSignal1 "infinite" s
  vcdE <- dumpVCD (0, 100) traced ["infinite"]
  vcd <- case vcdE of
          Left e -> fail e
          Right x -> pure x

  writeFile "infinite.vcd" vcd
```

Using `traceSignal1` works.

rowanG077, 2022-06-29:
Actually you have to specify `"infinite_0"` to get the first signal etc. So this is not a bug.

closed 2022-06-29 by rowanG077

martijnbastiaan, 2022-06-29:
Documentation bug perhaps?