clash-lang/clash-compiler #2263: Vivado ignores `ifdefs`s when indented

martijnbastiaan, 2022-07-02:
As part of `SimIO.Test00` Clash emits:

```verilog
        `ifdef VERILATOR
              $c("std::exit(0);");
              `else
              // NOTE: $finish_and_return is an iverilog extension!
              $finish_and_return(0);
              `endif
```

Vivado stumbles on this, mentioning:

> ERROR: [XSIM 43-4287] "/tmp/clash-test-4d43b61f310df194/Test00.topEntity/topEntity.v" Line 95. Undefined system task '$finish_and_return'

This is weird, because it doesn't get passed `VERILATOR` as a define. A recent client had similar issues where Vivado ignored ``` `define```s indented with a single space, so I'm guessing this is the same issue.

This issue has been discovered as part of https://github.com/clash-lang/clash-compiler/pull/2257.

DigitalBrains1, 2022-07-02:
So this is
```verilog
`ifdef VERILATOR
    a
`else
    b
`endif
```
and Vivado is saying that `b` does not exist. This is correct, right? It is simply stumbling because `finish_and_return` is `iverilog` specific. There is no path in `Clash.Explicit.SimIO.finish` that executes correctly on anything else than Verilator or Icarus Verilog, and especially not Vivado. We don't know how to return an exit code from Vivado at all.

I'd already discussed this with @vmchale on #tech-general, last Wednesday. I suggested this test be skipped on Vivado.

martijnbastiaan, 2022-07-02:
Oh :facepalm:. I completely skipped the `else`. Thanks!

closed 2022-07-02 by martijnbastiaan