clash-lang/clash-compiler #2289: Netlist generation fails on `Unsigned 1`

rowanG077, 2022-07-24:
After some refactoring I hit a bug where Clash failed during Normalization. When trying reducing to reduce it to a smaller case I no-longer hit a Normalization but hit a netlist generation issue. So I probably will create another issue once this one is fixed.

Anyway I can reproduce the issue with GHC 8.10.7 and with GHC 9.0.2 with both Clash 1.6.3 and the current master.

When I run synthesis I get this error:

```
<no location info>: error:
    Clash error call:
    Clash.Netlist(1117): mkDcApplication undefined for: (Unsigned 1,Name {nameSort = User, nameOcc = "GHC.Tuple.(,)", nameUniq = 3963167672086036486, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo},[Tick (NameMod PrefixName (LitTy (SymTy "$p2(%,%)"))) (Literal (NaturalLiteral 1)),Tick (NameMod PrefixName (LitTy (SymTy "$p2(%,%)"))) (Literal (NaturalLiteral 1))],[Just (Unsigned 64),Just (Unsigned 64)])
    CallStack (from HasCallStack):
      error, called at src/Clash/Netlist.hs:1117:9 in clash-lib-1.7.0-inplace:Clash.Netlist
      mkDcApplication, called at src/Clash/Netlist.hs:842:16 in clash-lib-1.7.0-inplace:Clash.Netlist
      mkExpr, called at src/Clash/Netlist.hs:505:31 in clash-lib-1.7.0-inplace:Clash.Netlist
```

And the complet compilation output with DebugApplied enabled available here(GHC 9.0.2 + Clash master):  https://gist.github.com/rowanG077/3c330bd65a2c7f66907189eef96443c4

I have created a reproducer which contains 3 modules so to make it easy to debug it I have created a repo for it: https://github.com/rowanG077/clash-netlist-gen-bug

It's based on clash-starter so to run it just clone it and run `cabal run clash -- Example.TopEntity --verilog` It assumes clash sources are available on directory up.

alex-mckenna, 2022-07-25:
It goes wrong here:

```
DEC {72}
Changes when applying rewrite to:
case eta[8214565720323789618][LocalId] of
  Example.Core.SDRAMInitRefresh[8214565720323786373]
    (ds[8214565720323789723] :: Example.Counter.CounterOverflow[8214565720323786416]
                                  1
                                  1) ->
    case Example.Counter.hasOverflown[2017][GlobalId]
           1
           1
           <prefixName>"$p1(%,%)"
           (GHC.Types.Eq#[3891110078048108571]
              @GHC.Types.Bool[3674937295934324744]
              @GHC.Types.True[3891110078048108586]
              @GHC.Types.True[3891110078048108586]
              (_CO_
                 @(GHC.Prim.~#[3674937295934324842]
                     GHC.Types.Bool[3674937295934324744]
                     GHC.Types.Bool[3674937295934324744]
                     GHC.Types.True[3891110078048108586]
                     GHC.Types.True[3891110078048108586])))
           ds[8214565720323789723][LocalId] of
      GHC.Types.False[3891110078048108556]  ->
        Example.Core.SDRAMInitRefresh[8214565720323786373]
          @(Example.Core.SDRAMSpec[8214565720323786354] 1)
          (Example.Counter.tick[2019][GlobalId] 1 1
             <prefixName>"$p1(%,%)"
             (GHC.Types.Eq#[3891110078048108571]
                @GHC.Types.Bool[3674937295934324744]
                @GHC.Types.True[3891110078048108586]
                @GHC.Types.True[3891110078048108586]
                (_CO_
                   @(GHC.Prim.~#[3674937295934324842]
                       GHC.Types.Bool[3674937295934324744]
                       GHC.Types.Bool[3674937295934324744]
                       GHC.Types.True[3891110078048108586]
                       GHC.Types.True[3891110078048108586])))
             ds[8214565720323789723][LocalId])
      GHC.Types.True[3891110078048108586]  ->
        Example.Core.SDRAMInitLoadModeReg[8214565720323786372]
          @(Example.Core.SDRAMSpec[8214565720323786354] 1)
          <prefixName>"initCount"
          (Clash.Sized.Internal.Unsigned.fromInteger#
             @(GHC.TypeNats.+[3674937295934325068]
                 1
                 (GHC.TypeLits.Extra.CLog[8214565720323785221]
                    2
                    (GHC.TypeLits.Extra.Max[8214565720323785219]
                       (Example.Core.SDRAMtBDL[8214565720323786364]
                          (Example.Core.SDRAMSpec[8214565720323786354] 1))
                       (Example.Core.SDRAMtBDL[8214565720323786364]
                          (Example.Core.SDRAMSpec[8214565720323786354]
                             1)))))
             (GHC.Num.Natural.naturalAdd 1
                <prefixName>"$w$cnatSing2"
                0)
             (GHC.Num.Integer.integerFromNatural
                (GHC.Num.Natural.naturalAdd
                   (GHC.Num.Natural.naturalAdd
                      (GHC.Num.Natural.naturalSubThrow
                         (GHC.Num.Natural.naturalShiftL# 1
                            (GHC.Prim.int2Word#
                               (GHC.Num.Integer.integerToInt#
                                  (GHC.Num.Integer.integerFromNatural
                                     <prefixName>"$w$cnatSing2"
                                     0))))
                         <prefixName>"$p2(%,%)"
[... 160 lines cut ...]
                   (GHC.Num.Natural.naturalAdd
                      (GHC.Num.Natural.naturalAdd
                         (GHC.Num.Natural.naturalSubThrow
                            (GHC.Num.Natural.naturalShiftL# 1
                               (GHC.Prim.int2Word#
                                  (GHC.Num.Integer.integerToInt#
                                     (GHC.Num.Integer.integerFromNatural
                                        <prefixName>"$w$cnatSing2"
                                        0))))
                            <prefixName>"$p2(%,%)"
                            1)
                         1)
                      0)))
     Example.Core.SDRAMInitLoadModeReg[8214565720323786372]
       (ds[8214565720323790191] :: Example.Counter.CounterOverflow[8214565720323786416]
                                     (Example.Core.SDRAMtBDL[8214565720323786364]
                                        (Example.Core.SDRAMSpec[8214565720323786354]
                                           1))
                                     (Example.Core.SDRAMtBDL[8214565720323786364]
                                        (Example.Core.SDRAMSpec[8214565720323786354]
                                           1))) ->
       Example.Core.SDRAMInitLoadModeReg[8214565720323786372]
         @(Example.Core.SDRAMSpec[8214565720323786354] 1)
         c$tickOut[3047][LocalId]
```

christiaanb, 2022-07-26:
Ugh... DEC, the bane of my existence... as a clash developer.

Anyhow, it seems to mess up selection of the evidence for the coercion. The short term "fix" would be to disable DEC for GADTs. The better fix is thinking of how to deal with GADTs inside DEC.

rowanG077, 2022-08-11:
How can I detect whether it's GADT? I rather not disable the entire DEC, since I then also have to use a custom Clash version. I'd be willing to implement this check and make a PR.

closed 2022-08-12 by christiaanb

martijnbastiaan, 2022-08-30:
We've released [v1.6.4](https://github.com/clash-lang/clash-compiler/releases/tag/v1.6.4), which includes a fix for this issue.

PR #2296 (2022-08-12): DEC shouldn't care about the shape of proofs
Whether we have `Refl : True ~ True` or `SomeAxiom : (1 <=? 2) ~ True` it doesn't matter, since when we normalize both sides we always end up with a proof of `True ~ True`.

Since DEC only fires for applications where all the type arguments are equal, we can deduce that all equality arguments witness the same equality, hence we don't have to care about the shape of the proof.

Fixes #2289

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files
commit d585d3c8 (2022-08-12): DEC shouldn't care about the shape of proofs

Whether we have `Refl : True ~ True` or `SomeAxiom : (1 <=? 2) ~ True`
it doesn't matter, since when we normalize both sides we always end
up with a proof of `True ~ True`.

Since DEC only fires for applications where all the type arguments
are equal, we can deduce that all equality arguments witness the same
equality, hence we don't have to care about the shape of the proof.

Fixes #2289
commit 4b60d5d7 (2022-08-16): DEC shouldn't care about the shape of proofs (#2297)

Whether we have `Refl : True ~ True` or `SomeAxiom : (1 <=? 2) ~ True`
it doesn't matter, since when we normalize both sides we always end
up with a proof of `True ~ True`.

Since DEC only fires for applications where all the type arguments
are equal, we can deduce that all equality arguments witness the same
equality, hence we don't have to care about the shape of the proof.

Fixes #2289

(cherry picked from commit d585d3c8b9d182c60d1ffabd696f18adab5c6bec)