clash-lang/clash-compiler #2306: Installation Problem - Mac Monterey 12.5.1 - `gcc' failed in phase `C pre-processor'

AmgadA, 2022-08-24:
Not able to install Clash on my MacBook pro with Monterey 12.5.1. I've reinstalled Xcode CL yet the same problem is still there.
The error message is as follows
---------------------------------------------------
src/Clash/Driver.hs:813:14: error:
     fatal error: 'cstdlib' file not found
        #include <cstdlib>
                 ^~~~~~~~~
    |
813 |     #include <cstdlib>
    |              ^
1 error generated.

src/Clash/Driver.hs:1:1: error:
    `gcc' failed in phase `C pre-processor'. (Exit code: 1)
  |
1 | {-|
  | ^
cabal: Failed to build clash-lib-1.6.1 (which is required by exe:clash from
clash-ghc-1.6.1 and exe:clashi from clash-ghc-1.6.1). See the build log above
for details.

martijnbastiaan, 2022-08-24:
https://github.com/clash-lang/clash-compiler/issues/2098 seems to suggest a fix for this was released in 1.6. Could you try the latest dot release of 1.6 (1.6.3)?

AmgadA, 2022-08-24:
I'm following the procedure from [http://clash-lang.github.io/](url).
So, I'm using _cabal install clash-ghc --enable-documentation_.
What should I change?

martijnbastiaan, 2022-08-24:
Oh boy, that website should be taken offline. We've got everything on https://clash-lang.org/ now. Could you try from there?

martijnbastiaan, 2022-08-24:
To save yourself a lot of trouble down the road, use "Option B. Setup a project". It's a bit more work to do at first, but it allows you to use dependencies published on Hackage. If you don't like Stack, note that you can download the starter projects as a `.zip` too. It should all be linked on that website.

Good luck!

martijnbastiaan, 2022-08-24:
No worries, let us know how it went, so we can close the issue.

The old web page now redirects to the new one. How did you get a hold of the old one btw?

AmgadA, 2022-08-25:
Thanks a lot.
Everything is ok with me now. The installation procedure is very easy.
For the old site, I'm trying to reach it now however, it's not working. 
Thanks again.

closed 2022-08-26 by martijnbastiaan