clash-lang/clash-compiler #2309: Clash doesn't properly reset terminal boldness after emitting a warning

martijnbastiaan, 2022-08-26:
See:

[image]

rowanG077, 2022-08-29:
It does for me. When I look at the raw output it actually does the correct thing:

```
Clash: Normalization took 0.001s
[1m[WARNING] Dubious primitive instantiation for GHC.Num.Integer.integerAdd: GHC.Num.Integer.integerAdd: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
[0mClash: Netlist generation took 0.000s
Clash: Compiling TopEntity.topEntity took 0.047s
Clash: Total compilation took 0.584s
```

`[1m[` starts bold `[0m` ends bold. If you look at the raw output do you also so these?

rowanG077, 2022-08-29:
@martijnbastiaan You could try this in your terminal to see if your terminal does the correct thing: `echo "echo "notbold\n\033[1mbold\n\033[0mnotbold"`

martijnbastiaan, 2022-08-30:
Hmm, good point. So this is probably Shake interfering. Thanks for checking!

closed 2022-08-30 by martijnbastiaan