clash-lang/clash-compiler #2310: clash 1.6.3 verilog bit operator precedence (shift vs and)

pbreuer, 2022-08-27:
Clash is generating verilog

> (32'sd1 <<< (ww1 >>> 64'sd1) & 64'sd2147483647) <<< 64'sd10

i.e., bearing in mind that those are signed operands
       
>       (1 <<< (ww1 >>> 1) & 0x7fffffff) << 10

The problem is that expression is intrinsically ambiguous but is disambiguated by bit AND having greater precedence than (arithmetic) bit SHIFT. The intention is (I suppose) to mask off the sign bit of ww1 which has been possibly filled by the arithmetic shift right on a signed operand because a logical shift was prescribed in the clash code. It was ...

(1 `shiftL` (ww1 `shiftR` 1)) `shiftL` 10

(the constants actually come as snatToNum on d1, d10) and logical shifts are specified there as those are Bits operations.

The problem is that the **2001** verilog standard says bit SHIFT has greater precedence than bit AND, not vice versa. See https://inst.eecs.berkeley.edu/~cs150/fa06/Labs/verilog-ieee.pdf table 12 of section 4.1.2 (about page 42). Also https://sutherland-hdl.com/pdfs/verilog_2001_ref_guide.pdf and https://technobyte.org/verilog-operators/ .

Clash is producing **2001** verilog:

> AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
>  ** GENERATED BY CLASH 1.6.3. DO NOT MODIFY.

It seems that the 2005 standard has the precedence the other way round. See http://staff.ustc.edu.cn/~songch/download/IEEE.1364-2005.pdf about page 74 table 5-4.
That is the same as in the 2009 standard. See http://www.ece.uah.edu/~gaede/cpe526/2012%20System%20Verilog%20Language%20Reference%20Manual.pdf about page 221 table 11-2.

I have been hunting some other verilog code generation problem, but having seen this, or imagined I have seen this (you will tell me), I'm reporting it. Apologies if I'm nuts. I am seeing problems that are proving difficult to isolate that are NOT due to the above, if that is a problem.

But the lack of one explicit parenthesis in the verilog expression makes me nervous anyway, so i would be glad if you could check that you are putting enough into the primitives that bindings end up as YOU want them to be, never mind  what the/a standard says. Murphy's law applies.

Regards

PTB

rowanG077, 2022-08-27:
this should already be fixed. Since https://github.com/clash-lang/clash-compiler/issues/2213 was fixed a while back.

Are you using the current Clash master or 1.6.3? 1.6.3 indeed doesn't include that fix yet

pbreuer, 2022-08-27:
Using 1.6.3 from a while back plus some vhdl patches. Thanks for the info
and quick reply and I'll check out the patch you've pointed to.

Regards

PTB

martijnbastiaan, 2022-08-27:
1.6 has accumulated [quite a number of patches](https://github.com/clash-lang/clash-compiler/tree/1.6/changelog), so it's probably wise to release 1.6.4.

pbreuer, 2022-08-27:
#2213 is about vhdl not verilog. I believe I have the patch for that in - since the clash executable is generating correct vhdl (not verlog).

Clash, version 1.6.3 (using clash-lib, version: 1.6.3)
compiled April 7 this year. It contains the patch "[PATCH] Render ~ARG[n] enclosed in parentheses" from Leon Schoorl, with comment "Fixes #2213", which is a one liner:

-  renderOneLine <$> getAp (expr False e)
+  renderOneLine <$> getAp (expr True e)

in clash-lib/src/Clash/Netlist/BlackBox/Util.hs.

Are you saying that you think that would have put more parentheses into the verilog I quoted too? It hasn't. 

I'm not going to speculate on how come. I could, but I would be wrong!

Regards

PTB

"

rowanG077, 2022-08-27:
Since you didn't gave an explicit reproducer I tried it myself. This TopEntity:

```
topEntity :: Signal System Int -> Signal System Int
topEntity inp = let go ww1 = (1 `shiftL` ((1 `shiftL` (ww1 `shiftR` 1)) .&. 0x7fffffff)) `shiftL` 10
                in fmap go inp
```

Produces this verilog:

```
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.7.0. DO NOT MODIFY.
*/
`default_nettype none
`timescale 100fs/100fs
module topEntity
    ( // Inputs
      input wire signed [63:0] inp

      // Outputs
    , output reg signed [63:0] c$case_alt
    );
  wire signed [63:0] \x# ;
  wire signed [63:0] \i# ;
  wire signed [63:0] \i#1 ;
  reg signed [63:0] c$case_alt_1;
  wire signed [63:0] c$case_alt_selection_res;
  wire signed [63:0] c$case_alt_selection_res_0;

  assign \x#  = inp;

  assign \i#  = \x#  >>> 64'sd1;

  assign c$case_alt_selection_res = (\i#  >= 64'sd64) ? 64'sd1 : 64'sd0;

  always @(*) begin
    case(c$case_alt_selection_res)
      64'sd1 : c$case_alt = 64'sd1024;
      default : c$case_alt = c$case_alt_1;
    endcase
  end

  assign \i#1  = (64'sd1 <<< \i# ) & 64'sd2147483647;

  assign c$case_alt_selection_res_0 = (\i#1  >= 64'sd64) ? 64'sd1 : 64'sd0;

  always @(*) begin
    case(c$case_alt_selection_res_0)
      64'sd1 : c$case_alt_1 = 64'sd0;
      default : c$case_alt_1 = (64'sd1 <<< \i#1 ) <<< 64'sd10;
    endcase
  end

endmodule
```

From a quick glance it seems everything is parenthesized.

pbreuer, 2022-08-28:
Sorry - I didn't know how to generate a small example, but your code teaches me.
Can you try this?

`module Test.Example(topEntity)
  where

import Clash.Prelude

topEntity :: Signal System Int -> Signal System (Signed 32)
topEntity = fmap sizB

sizB :: Int -> Signed 32
sizB dat =  (1 `myShiftL` (dat `shiftR` 1)) `shiftL` 10

-- 31-bit mask on shift value
myShiftL           :: (Bits a) => a -> Int -> a
myShiftL x y        = shiftL x (y .&. (bit 31 - 1))
`

rowanG077, 2022-08-28:
Produces this verilog:

```
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.7.0. DO NOT MODIFY.
*/
`default_nettype none
`timescale 100fs/100fs
module topEntity
    ( // Inputs
      input wire signed [63:0] x

      // Outputs
    , output wire signed [31:0] result
    );
  wire signed [63:0] \x# ;

  assign \x#  = x;

  assign result = (32'sd1 <<< ((\x#  >>> 64'sd1) & 64'sd2147483647)) <<< 64'sd10;

endmodule
```

Which I believe is what you want.

pbreuer, 2022-08-28:
Yes, that looks good. It is not what I am getting from 1.6.3 nominally plus #2213 patch, however. I'm missing parens as first reported. It may be that my 1.6.3 plus patch compile  was not clean enough. I'll recompile it very cleanly tomorrow.
1.5.0 is producing enough parens. More than enough.

pbreuer, 2022-08-28:
I am worried however why arithmetic shift right is performed on a signed integer by shiftR. Isn't that a Bits operation? So it should work on the bits representation, which is the same between signed and unsigned integer, and hence shiftR should do the same on both (principle of no surprises).

pbreuer, 2022-08-28:
Yes, recompile from a very clean state produces the same verilog as you got from 1.6.3 plus patch #2213.

`/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.6.3. DO NOT MODIFY.
*/
`timescale 100fs/100fs
module topEntity
    ( // Inputs
      input signed [63:0] x

      // Outputs
    , output wire signed [31:0] result
    );
  wire signed [63:0] ww1;

  assign ww1 = x;

  assign result = (32'sd1 <<< ((ww1 >>> 64'sd1) & 64'sd2147483647)) <<< 64'sd10;

endmodule
`

rowanG077, 2022-08-28:
I will close this issue now since it's resolved in the current master.

closed 2022-08-28 by rowanG077