clash-lang/clash-compiler #2322: Clash 1.6.4 build against ghc 9.0.2 MAY fail

pbreuer, 2022-09-03:
Well, it fails here, but there are a lot of variables and you probably compile against 9.0 so it's likely some other infrastructure and you might be able to identify which from the error message. Clas 1.6.4 compiles and runs FINE against ghc 8.8, by the way. I just tried updating the haskell installation in debian unstable to 9.0 and recompiling, whence ....

```
src/Clash/Driver.hs:544:11: error:
    * Could not deduce (exceptions-0.10.4:Control.Monad.Catch.MonadMask
                          m)
        arising from a use of `Hint.unsafeRunInterpreterWithArgsLibdir'
      from the context: (MonadIO m, MonadMask m)
        bound by the type signature for:
                   loadImportAndInterpret :: forall (m :: Type -> Type) a.
                                             (MonadIO m, MonadMask m, MonadMask m) =>
                                             [String]
                                             -> [String]
                                             -> String
                                             -> Hint.ModuleName
                                             -> String
                                             -> String
                                             -> m (Either Hint.InterpreterError a)
        at src/Clash/Driver.hs:(523,1)-(538,39)
      Possible fix:
        add (exceptions-0.10.4:Control.Monad.Catch.MonadMask
               m) to the context of
          the type signature for:
            loadImportAndInterpret :: forall (m :: Type -> Type) a.
                                      (MonadIO m, MonadMask m, MonadMask m) =>
                                      [String]
                                      -> [String]
                                      -> String
                                      -> Hint.ModuleName
                                      -> String
                                      -> String
                                      -> m (Either Hint.InterpreterError a)
```
That looks to me that whatever supplies "exceptions" has a local version of MonadMask.

Any insight very welcome! It was compiling source file 110 of 111.

Regards

PTB

rowanG077, 2022-09-03:
If you use cabal. Could you run the `cabal freeze`? This will output `cabal.project.freeze` file which contains all the exact versions used by your build. With a bit if luck we only need to tighten some version bounds.

pbreuer, 2022-09-03:
No I don't use cabal (directly), i run cabal-debian to build a debian/ subdir in which the magic is found. But I can certainly find the exact versions of everything .. if only by running ghc with verbose options, AFAIR, and if not that, some other way. But this will have to wait till tonight.

Do you get a sense of what import is related with "exceptions". I see ...
Control.Exception                (throw)
Clash.Util       (ClashException(..),...)

later ... Hint.GhcException

and that's all I see in Drivers.hs. It was complaining about bbfE <- Hint.unsafeRunInterpreterWithArgsLibdir  so it looks on the face of it like Hint has a more local version of MonadMask but it can't be exactly that because there are plenty of other uses of Hint.stuff and it would be strange if that were the only one to need whatever it is and get it locally. Perhaps it is that sort of thing but at one remove.

PTB

pbreuer, 2022-09-03:
Verbose flags on the compile (ghc -v) show up the probable issue:
```
Warning:
    This package indirectly depends on multiple versions of the same package. This is very likely to cause a compile failure.
      package temporary (temporary-1.3-FW3rcq7XCDL8cgO9MswHTe) requires exceptions-0.10.4
      package resourcet (resourcet-1.2.6-6XIzfiYtIwKIJcdzx5v9j4) requires exceptions-0.10.4
      package lens (lens-5.0.1-HDGp0ETtBrVJmoIxpPbogg) requires exceptions-0.10.4
      package hint (hint-0.9.0.6-HgKYJTrorOJIGDC8f1M0HL) requires exceptions-0.10.4
      package ghc (ghc-9.0.2) requires exceptions-0.10.4
      package free (free-5.1.9-5ocKcmGnukTGZylLDfZkD) requires exceptions-0.10.4
      package conduit (conduit-1.3.4.2-KcZzDdHyntUJ6sJMJQrJQa) requires exceptions-0.10.4
      package clash-lib (clash-lib-1.6.4) requires exceptions-0.10.5-6vU3qh6cQ1dADxrb7ixe7t

```

pbreuer, 2022-09-03:
This looks like a debian snafu (temporary, in debian unstable). ghc is compiled against exceptions 0.10.4 but 0.10.5 is installed (debian did that, I didn't).

ii  libghc-exceptions-dev                        0.10.5-1                              amd64        extensible optionally-pure exceptions

With luck, I can downgrade that and try again.

pbreuer, 2022-09-03:
Yes, I removed the external standalone exceptions package (0.10.5) leaving only ghc 9.0's internal 0.10.4 package and now Clash.Driver.hs compiles.

i imagine the package was external for ghc 8.8 and is internal in 9.0 and somebody updated the standalone external. 

So essentially my build problem and nothing to do with you. The cabal file gives a wide constraint for the exceptions package   '>= 0.8.3    && < 0.11.0'. I suppose that is right, and there is no way to say "use ghc's internal version if it is there'.

Apologies.

closed 2022-09-03 by rowanG077