clash-lang/clash-compiler #2327: Heisenbug? Inlining makes synthesis fail with "Can't translate non-tycon (function) type'

pbreuer, 2022-09-10:
How can more inlining make a function appear where before there was none? Inlining should substitute the arguments of a function evoking partial evaluation to a result. If someone can explain what may be going on I would be very grateful. I am only reportng it (I can get rid of it in practice here) as it does not match my mental model.

Clash 1.6.4, ghc 9.0.2, 64-bit AMD.

KPU/RFC7693/Blake2b.hs:500:1: error:
    Clash.Netlist.BlackBox(680): Can't translate non-tycon type: Clash.Sized.Internal.Unsigned.Unsigned ... 64
    -> Clash.Sized.Internal.Unsigned.Unsigned ...  64

Synthesising verilog with -fclash-no-render-enums. On seeing that I NOINLINEd every subfunction in that file, intending to localize the problem, but it then all synthesized. The error reported was in the top level function, the only one exported from that file, which was itself noinlined.

That is a hash ... from 512 bits to 384 bits, so the 64b -> 64b function it was complaining about is some subfunction. The hash is done in 12 rounds (24 half rounds), each of them stateful. It's  a blake2b hash as described in https://tools.ietf.org/html/rfc7693 if interested.

The state consists of a length 16 vector of 64b words so a transform on a vector element is likely responsible... but how can that go away with more inlining?

I would have guessed that I was incautious enough to create a vector of the rounds/half-rounds, each being state to state transforms, and then compose the elements of that vector of functions to get the complete hash function. But I have looked at the code and no. It works around that with synthesis clearly in mind.

Does anyone want me to try to chase this down further?  Right now it looks (to me) like it will a heisenbug - one that appears and disappears randomly. But if there is  a possible explanation, that may help locate it.

Regards

PTB

leonschoorl, 2022-09-15:
Going on the [error location](https://github.com/clash-lang/clash-compiler/blob/v1.6.4/clash-lib/src/Clash/Netlist/BlackBox.hs#L680), it seems clash encounters a primitive/blackbox which is somehow under-applied, because its "result type" is a function `Unsigned 64 -> Unsigned 64`.
And it fails to compute a HDL type for the result of the primitive.

I'd guess there is a bug where some transformation is somehow dropping an argument in some special situation, causing this under-applied primitive.
And this special situation may only occur when certain things are inlined.
Thus causing the problem to dissappear when you force them to be not inlined.

it would be great if you have an simple way to reproduce this that you can share, then we could take a look.
Selectively adding/removing NOINLINEs until the error (dis)appears should help locating which part your code is triggering the bug.

leonschoorl, 2022-09-21:
You can also try passing arguments `-fclash-debug DebugSilent` to clash, that'll enable some extra internal checks during compilation that might also help tracing things down.