clash-lang/clash-compiler #2361: Can't use strict `Signed` in register's initial content

martijnbastiaan, 2022-11-18:
```haskell
{-# LANGUAGE DataKinds #-}
{-# LANGUAGE DeriveAnyClass #-}
{-# LANGUAGE DeriveGeneric #-}

module TXXXX where

import Clash.Explicit.Prelude

data T = T !(Signed 32) deriving (Generic, NFDataX)

topEntity ::
  Clock System ->
  Enable System ->
  Signal System T ->
  Signal System T
topEntity clk ena t = delay clk ena (T 1) t
```

Produces:

```
<no location info>: error:
    
    Clash.Netlist.BlackBox(216): Couldn't instantiate blackbox for Clash.Signal.Internal.delay#. Verification procedure reported:
    
    Argument 4 should be literal, as blackbox used ~CONST[4], but was:
    
    DataCon (Signed 64) (DC (Void Nothing,-1)) [Identifier (RawIdentifier "dt1" Nothing [("unsafeMake",SrcLoc {srcLocPackage = "clash-lib-1.7.0-inplace", srcLocModule = "Clash.Netlist.Id", srcLocFile = "src/Clash/Netlist/Id.hs", srcLocStartLine = 246, srcLocStartCol = 20, srcLocEndLine = 246, srcLocEndCol = 30}),("unsafeFromCoreId",SrcLoc {srcLocPackage = "clash-lib-1.7.0-inplace", srcLocModule = "Clash.Netlist", srcLocFile = "src/Clash/Netlist.hs", srcLocStartLine = 849, srcLocStartCol = 33, srcLocEndLine = 849, srcLocEndCol = 54}),("mkExpr",SrcLoc {srcLocPackage = "clash-lib-1.7.0-inplace", srcLocModule = "Clash.Netlist", srcLocFile = "src/Clash/Netlist.hs", srcLocStartLine = 993, srcLocStartCol = 21, srcLocEndLine = 993, srcLocEndCol = 64}),("mkDcApplication",SrcLoc {srcLocPackage = "clash-lib-1.7.0-inplace", srcLocModule = "Clash.Netlist.BlackBox", srcLocFile = "src/Clash/Netlist/BlackBox.hs", srcLocStartLine = 313, srcLocStartCol = 30, srcLocEndLine = 313, srcLocEndCol = 99})]) Nothing]
```

`DebugApplied` output:

```
topEntity before normalization:

λclk ->
λena ->
λt ->
delay# $fKnownDomain"System"[GlobalId] $fNFDataXT[GlobalId] clk[LocalId] ena[LocalId]
  topEntity1[GlobalId]
  t[LocalId]

removeUnusedExpr {0}
Changes when applying rewrite to:
delay# $fKnownDomain"System"[GlobalId] $fNFDataXT[GlobalId] clk[LocalId] ena[LocalId]
  topEntity1[GlobalId]
  t[LocalId]
Result:
delay# removedArg removedArg clk[LocalId] ena[LocalId] topEntity1[GlobalId] t[LocalId]

inlineWorkFree {1}
Changes when applying rewrite to:
topEntity1[GlobalId]
Result:
case topEntity2[GlobalId] of
  S dt1 ->
    T dt1[LocalId]

inlineWorkFree {2}
Changes when applying rewrite to:
topEntity2[GlobalId]
Result:
topEntity4[GlobalId] topEntity3[GlobalId]

inlineSmall {3}
Changes when applying rewrite to:
topEntity4[GlobalId] topEntity3[GlobalId]
Result:
fromInteger# topEntity5[GlobalId] topEntity3[GlobalId]

inlineWorkFree {4}
Changes when applying rewrite to:
topEntity5[GlobalId]
Result:
32

inlineWorkFree {5}
Changes when applying rewrite to:
topEntity3[GlobalId]
Result:
1

Unmatchable constant as case subject: fromInteger# 32 1
WHNF is: fromInteger# 32 1
ANF {6}
Changes when applying rewrite to:
λclk ->
λena ->
λt ->
delay# removedArg removedArg clk[LocalId] ena[LocalId]
  (case (fromInteger# 32 1) of
     S dt1 ->
       T dt1[LocalId])
  t[LocalId]
Result:
λclk ->
λena ->
λt ->
letrec
  dt1
  = case fromInteger# 32 1 of
      S c$sel ->
        c$sel[LocalId]
  c$case_alt
  = T dt1[LocalId]
in delay# removedArg removedArg clk[LocalId] ena[LocalId] c$case_alt[LocalId] t[LocalId]

Unmatchable constant as case subject: fromInteger# 32 1
WHNF is: fromInteger# 32 1
topLet {7}
Changes when applying rewrite to:
letrec
  dt1
  = case fromInteger# 32 1 of
      S c$sel ->
        c$sel[LocalId]
  c$case_alt
  = T dt1[LocalId]
in delay# removedArg removedArg clk[LocalId] ena[LocalId] c$case_alt[LocalId] t[LocalId]
Result:
letrec
  dt1
  = case fromInteger# 32 1 of
      S c$sel ->
        c$sel[LocalId]
  c$case_alt
  = T dt1[LocalId]
  result
  = delay# removedArg removedArg clk[LocalId] ena[LocalId] c$case_alt[LocalId] t[LocalId]
in result[LocalId]

inlineCleanup {8}
Changes when applying rewrite to:
letrec
  dt1
  = case fromInteger# 32 1 of
      S c$sel ->
        c$sel[LocalId]
  c$case_alt
  = T dt1[LocalId]
  result
  = delay# removedArg removedArg clk[LocalId] ena[LocalId] c$case_alt[LocalId] t[LocalId]
in result[LocalId]
Result:
letrec
  dt1
  = case fromInteger# 32 1 of
      S c$sel ->
        c$sel[LocalId]
  result
  = delay# removedArg removedArg clk[LocalId] ena[LocalId] (T dt1[LocalId]) t[LocalId]
in result[LocalId]

Unmatchable constant as case subject: fromInteger# 32 1
WHNF is: fromInteger# 32 1
normalization {9}
Changes when applying rewrite to:
λclk ->
λena ->
λt ->
delay# $fKnownDomain"System"[GlobalId] $fNFDataXT[GlobalId] clk[LocalId] ena[LocalId]
  topEntity1[GlobalId]
  t[LocalId]
Result:
λclk ->
λena ->
λt ->
letrec
  dt1
  = case fromInteger# 32 1 of
      S c$sel ->
        c$sel[LocalId]
  result
  = delay# removedArg removedArg clk[LocalId] ena[LocalId] (T dt1[LocalId]) t[LocalId]
in result[LocalId]

topEntity after normalization:

λclk ->
λena ->
λt ->
letrec
  dt1
  = case fromInteger# 32 1 of
      S c$sel ->
        c$sel[LocalId]
  result
  = delay# removedArg removedArg clk[LocalId] ena[LocalId] (T dt1[LocalId]) t[LocalId]
in result[LocalId]

Clash: Transformations:
normalization: 1
inlineCleanup: 1
inlineSmall: 1
topLet: 1
ANF: 1
removeUnusedExpr: 1
inlineWorkFree: 4

Clash: Applied 10 transformations
Clash: Normalization took 0.003s
```

I'm assuming this key part is:

```
Unmatchable constant as case subject: fromInteger# 32 1
```

Would that be fair @alex-mckenna @christiaanb?

christiaanb, 2022-11-18:
That’s just the red herring, the issue is that the transformed version of `T` wants to store an Integer instead of a Signed.

what happens when you pass `-fno-unbox-small-strict-fields -fno-unbox-strict-fields`?

martijnbastiaan, 2022-11-18:
Right, that "fixes" it. Do you think there's a way to either:

* Detect this and emit a more sensible warning
* De-unbox it

?