clash-lang/clash-compiler #2365: Non-sensical error message when failing to parse InlineYaml from precompiled module

martijnbastiaan, 2022-11-22:
If your (starter) project contains a primitive such as:

```
...
{-# NOINLINE foo #-}
{-# ANN foo hasBlackBox #-}
{-# ANN foo (InlineYamlPrimitive [minBound..] [__i|
 BlackBox:
   kind: Declaration
   name: Some.Incorrect.Name.bar
   template: |-
     // begin foo
     foo
     // end foo
|]) #-}
```

Clash will fail with a very misleading error message:

```
<no location info>: error:
    module ‘Some.Correct.Name’ cannot be found locally
```

Ultimately, this is caused by functions in `LoadInterfaceFiles` not propagating up their errors cleanly:

https://github.com/clash-lang/clash-compiler/blob/a8b213ab86df143c9e259dd58ced8ec81f4eee5b/clash-ghc/src-ghc/Clash/GHC/LoadInterfaceFiles.hs#L405-L461

leonschoorl, 2022-11-25:
There is another instance of this same sort of behaviour in [loadImportAndInterpret](https://github.com/clash-lang/clash-compiler/blob/f2077410f9f8e2eded675b3972e0fbc2709a0776/clash-lib/src/Clash/Driver.hs#L542-L566), it's like
```haskell
res <- loadModule FromPackageDB ...
case res of
  Success _ -> return res
  Failed _ -> loadModule FromLocalSourceFile ...
```
Where clash tries to load something from the (global) package database, and if anything goes wrong it assumes that it doesn't exist there, throws away the error message and tries again to load that module from a local source file, possible failing with the confusing error that the module can't be found locally.

We should probably fix those two instances in one go.

closed 2024-02-13 by martijnbastiaan

christiaanb, 2025-01-04:
We've released [v1.8.2](https://github.com/clash-lang/clash-compiler/releases/tag/v1.8.2), which includes a fix for this issue.

commit 6cd63675 (2024-02-08): Print previous errors if local module fails to load

Fixes #2365
commit d80e44c6 (2024-02-08): Print previous errors if local module fails to load

Fixes #2365
commit 57644866 (2024-02-08): Print previous errors if local module fails to load

Fixes #2365
commit 7abc72e3 (2024-02-13): Print previous errors if local module fails to load

Fixes #2365