clash-lang/clash-compiler #2371: Simpler AutoReg instance for Vec?

jonfowler, 2022-12-13:
This is more of a question. The current `AutoReg` instance of `Vec` looks like this:
```haskell
instance (KnownNat n, AutoReg a) => AutoReg (Vec n a) where
  autoReg
    :: forall dom. (HasCallStack, KnownDomain dom)
    => Clock dom -> Reset dom -> Enable dom
    -> Vec n a -- ^ Reset value
    -> Signal dom (Vec n a)
    -> Signal dom (Vec n a)
  autoReg clk rst en initVal xs =
    bundle $ smap go (lazyV initVal) <*> unbundle xs
   where
    go :: forall (i :: Nat). SNat i -> a  -> Signal dom a -> Signal dom a
    go SNat = suffixNameFromNatP @i . autoReg clk rst en
  {-# INLINE autoReg #-}
```
The definition of `go` here seems to force the Vec to be unrolled. This is bad as it doesn't work for large vectors (performance wall), but also leads to large blocks of repeated RTL.

Is there a reason it can't be something like:
```haskell
instance (KnownNat n, AutoReg a) => AutoReg (Vec n a) where
  autoReg
    :: forall dom. (HasCallStack, KnownDomain dom)
    => Clock dom -> Reset dom -> Enable dom
    -> Vec n a -- ^ Reset value
    -> Signal dom (Vec n a)
    -> Signal dom (Vec n a)
  autoReg clk rst en initVal xs =
    bundle $ zipWith (autoReg clk rst en) initVal (unbundle xs)
  {-# INLINE autoReg #-}
```
Which should then not be unrolled?

christiaanb, 2022-12-13:
It could be. It depends on how much we care about register naming. With the current code, for an `AutoReg` of type:
```
data T = MkT { a :: Int, b :: Int}

Vec 2 T
```
We would get registers called:
```
XYZ_0_A
XYZ_0_B
XYZ_1_A
XYZ_1_B
```
With your suggestion, Clash would generate:
```
XYZ_A
XYZ_B
```
inside a generate-for loop. And then the RTL synthesis tools would create the following register names:
```
XYZ_A
XYZ_B
XYZ_A_0
XYZ_B_0
```

jonfowler, 2022-12-13:
Not unrolling also seems to be prone to these kind of errors:
```
    Clash.Netlist.BlackBox(201): Couldn't instantiate blackbox for Clash.Signal.Internal.register#. Verification procedure reported:

    Argument 5 should be literal, as blackbox used ~CONST[5], but was:

    Identifier (RawIdentifier "~ARGN[1][0]" Nothing []) Nothing
```
Presumable because the register initial value is not being evaluated to a constant. So it would need more than a simple change.

closed 2022-12-13 by jonfowler

christiaanb, 2022-12-14:
As a work-around you could do:
```haskell
newtype LargeVector n a = LV {unLV :: Vec n a}

-- | WARNING: all reset values must be the same
instance (KnownNat n, AutoReg a, 1 <= n) => AutoReg (LargeVector n a) where
  autoReg
    :: forall dom. (HasCallStack, KnownDomain dom)
    => Clock dom -> Reset dom -> Enable dom
    -> Vec n a -- ^ Reset value
    -> Signal dom (LargeVector n a)
    -> Signal dom (LargeVector n a)
  autoReg clk rst en initVal xs =
    fmap LV $ bundle $ map (autoReg clk rst en (head initVal)) (unbundle $ fmap unLV xs)
  {-# INLINE autoReg #-}
```