clash-lang/clash-compiler #2373: Implement Xilinx CDC primitives

martijnbastiaan, 2022-12-15:
We currently offer clock domain crossing through dual flip flop synchronizers and dual clock FIFOs. Depending on the use case, this is either insufficient or overkill. Ideally, users would be able to simply follow the following flow chart:

[image]

--------

To implement:

- [ ] `XPM_FIFO_ASYNC`
- [ ] `XPM_CDC_GRAY`
- [ ] `XPM_CDC_HANDSHAKE`
- [ ] `XPM_CDC_ARRAY_SINGLE`

DigitalBrains1, 2022-12-17:
"Is the data a counter?"
That is underspecified. Is the data a counter, and does it only increment or decrement by a maximum of one count each cycle? The Gray counter CDC will still corrupt if the counter skips values. Maybe to some people this is obvious, but I think many people will not realise it.

martijnbastiaan, 2024-04-08:
I'm closing this issue as we have all of the (properly implementable) CDC primitives now. We should probably open an issue detailing the full list of Xilinx primitives.