clash-lang/clash-compiler #2386: Derived BitPack instance introduces Dubious primitive instantiation warning

lmbollen, 2023-01-02:
The implementation for derived `BitPack` instances produces the following warning when producing HDL:
```
GHC: Setting up GHC took: 0.167s
GHC: Compiling and loading modules took: 0.088s
Clash: Parsing and compiling primitives took 0.143s
GHC+Clash: Loading modules cumulatively took 0.534s
Clash: Compiling Example.Project.topEntity
Clash: Normalization took 0.069s
[WARNING] Dubious primitive instantiation for GHC.Num.Integer.integerToInt#: GHC.Num.Integer.integerToInt#: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
Clash: Netlist generation took 0.001s
Clash: Compiling Example.Project.topEntity took 0.071s
Clash: Total compilation took 0.606s
```

Reproducer: 
```haskell

module Example.Project where

import Clash.Prelude

data AorB = A | B
  deriving (Generic, BitPack)

topEntity :: Bool -> AorB
topEntity = bitCoerce

```
GHC 9.0.2
Clash 1.6.4

leonschoorl, 2023-01-02:
In the [definition of unpack](https://github.com/clash-lang/clash-compiler/blob/9ae2c195a7be7b3e49eff693ade63ac17f9dfea0/clash-prelude/src/Clash/Class/BitPack/Internal.hs#L144) we effectively have:
```haskell
sc = unpack @Int (x :: BitVector 64)
```
That generates the error seen here.

DigitalBrains1, 2023-01-02:
I think we should just add a prim for `unpack @Int` (but don't forget about `-fclash-intwidth`)

DigitalBrains1, 2026-05-03:
We've released [v1.10.0](https://github.com/clash-lang/clash-compiler/releases/tag/v1.10.0), which includes a fix for this issue.