clash-lang/clash-compiler #2424: Weird documentation artifacts on Hackage

martijnbastiaan, 2023-02-17:
[image]

DigitalBrains1, 2023-02-19:
Yeah, I've noticed. It's because of `LANGUAGE CPP` and Haddock not seeing eye to eye. There are more instances of this; `Clash.Signal` is riddled with it with no multiple-hidden. `delayedI` goes well merely because the blocks are so small. `tdfold` exhibits the problem, as do `head`, `tail`, `init`, `last` and `take`. `izipWith` and `stencil2d` escape punishment because the blocks are short. The `populationCount` _example usage_ exhibits the issue.

I think that is actually a complete list for `clash-prelude`!

I think to fix it you need some text duplication:
```diff
diff --git a/clash-prelude/src/Clash/Sized/Vector.hs b/clash-prelude/src/Clash/Sized/Vector.hs
index e6565e768..33f143056 100644
--- a/clash-prelude/src/Clash/Sized/Vector.hs
+++ b/clash-prelude/src/Clash/Sized/Vector.hs
@@ -409,12 +409,13 @@ head (x `Cons` _) = x
 
 {-# NOINLINE tail #-}
 {-# ANN tail hasBlackBox #-}
+
+#if __GLASGOW_HASKELL__ >= 900
 {- | Extract the elements after the head of a vector
 
 >>> tail (1:>2:>3:>Nil)
 2 :> 3 :> Nil
 
-#if __GLASGOW_HASKELL__ >= 900
 >>> tail Nil
 <BLANKLINE>
 <interactive>:...
@@ -424,8 +425,13 @@ head (x `Cons` _) = x
     • In the first argument of ‘tail’, namely ‘Nil’
       In the expression: tail Nil
       In an equation for ‘it’: it = tail Nil
-
+-}
 #else
+{- | Extract the elements after the head of a vector
+
+>>> tail (1:>2:>3:>Nil)
+2 :> 3 :> Nil
+
 >>> tail Nil
 <BLANKLINE>
 <interactive>:...
@@ -435,9 +441,9 @@ head (x `Cons` _) = x
     • In the first argument of ‘tail’, namely ‘Nil’
       In the expression: tail Nil
       In an equation for ‘it’: it = tail Nil
-
-#endif
 -}
+#endif
+
 tail :: Vec (n + 1) a -> Vec n a
 tail (_ `Cons` xs) = xs
```

and so on and so forth.

DigitalBrains1, 2023-02-19:
I worked on this in #1957 but apparently that didn't fix it. Before that PR, swaths of documentation were omitted altogether. Now they are included but contain weird artifacts.

leonschoorl, 2023-02-28:
Duplicate of #2278

closed 2023-02-28 by leonschoorl