clash-lang/clash-compiler #2446: Sandwich of `repeat` crashes Vivado, even with `-fhdl-syn Vivado`

martijnbastiaan, 2023-03-31:
Clash: 

```haskell
testBench :: Vec 5 (Bit, Bit)
testBench = (repeat @2 x) ++ (x :> repeat @2 x)
 where
  x = (0, 0)
```

VHDL:

```vhdl
-- Automatically generated VHDL-93
[..]
architecture structural of testBench is
  signal result   : TdpBlockRam_testBench_types.array_of_Tup2(0 to 4);
  [..]
begin
  result <= TdpBlockRam_testBench_types.array_of_Tup2'(TdpBlockRam_testBench_types.array_of_Tup2'((TdpBlockRam_testBench_types.array_of_Tup2'(0 to 2-1 =>  TdpBlockRam_testBench_types.toSLV(TdpBlockRam_testBench_types.Tup2'((Tup2_sel0_std_logic_0 => '0', Tup2_sel1_std_logic_1 => '0'))) ))) & TdpBlockRam_testBench_types.array_of_Tup2'(TdpBlockRam_testBench_types.array_of_Tup2'(std_logic_vector'((TdpBlockRam_testBench_types.toSLV('0')
           & TdpBlockRam_testBench_types.toSLV('0'))) & TdpBlockRam_testBench_types.array_of_Tup2'(0 to 2-1 =>  TdpBlockRam_testBench_types.toSLV(TdpBlockRam_testBench_types.Tup2'((Tup2_sel0_std_logic_0 => '0', Tup2_sel1_std_logic_1 => '0'))) ))));
  [..]
end;
```

Gives:

```
                  FATAL_ERROR: Vivado Simulator kernel has discovered an exceptional condition from which it cannot recover. Process will terminate. For technical support on this issue, please open a WebCase with this project attached at http://www.xilinx.com/support.
                  Time: 0 ps  Iteration: 0  Process: /testBench/line__32
                    File: /tmp/clash-test-a28f230dc2798bd6/vivado-testBench/TdpBlockRam.testBench/testBench.vhdl
                  
                  HDL Line: /tmp/clash-test-a28f230dc2798bd6/vivado-testBench/TdpBlockRam.testBench/TdpBlockRam_testBench_types.vhdl:23
```

---------------------------

Not strictly the same issue, but the following _without_ `-fclash-hdlsyn Vivado` also crashes Vivado in the same way:

```vhdl
testBench :: Vec 2 (BitVector 2, BitVector 2)
testBench = ((0, 0) :> Nil) ++ ((0, 0) :> Nil)
```

christiaanb, 2023-03-31:
Is this `xsim`? Or actual Vivado Synthesis and P&R?

martijnbastiaan, 2023-03-31:
Simulation with xsim. I'm assuming this is an upstream issue with Vivado but I ran out of time to properly debug it. Posting it here in case:

* Someone knows a workaround 
* It turns out to be an issue with Clash after all