clash-lang/clash-compiler #2453: Missing `Hashable` instance for `clash-lib`

kleinreact, 2023-04-19:
I get the following error when I try to compile `clash-lib` with ghc-9.0.2:
```
src/Clash/Core/Literal.hs:68:40: error:
    • No instance for (Hashable ByteArray)
        arising from the 'deriving' clause of a data type declaration
      Possible fix:
        use a standalone 'deriving instance' declaration,
          so you can specify the instance context yourself
      There are instances for similar types:
        instance Hashable data-array-byte-0.1.0.1:Data.Array.Byte.ByteArray
          -- Defined in ‘hashable-1.4.2.0:Data.Hashable.Class’
    • When deriving the instance for (Hashable Literal)
   |
68 |   deriving (Eq,Ord,Show,Generic,NFData,Hashable,Binary)
   |                                        ^^^^^^^^
```

martijnbastiaan, 2023-04-19:
Are you using `master`? Otherwise we might need to backport:

 * https://github.com/clash-lang/clash-compiler/pull/2448
 * https://github.com/clash-lang/clash-compiler/pull/2449

kleinreact, 2023-04-19:
Yes. `master`. Fresh checkout.

martijnbastiaan, 2023-04-19:
Right. I think this might be an interaction between the newer versions of `hashable` and `primitive`:

Primitive:

> When building with base-4.17 and newer, re-export the ByteArray and MutableByteArray types from base instead of defining them in this library. This does not change the user-facing interface of Data.Primitive.ByteArray.

Hashable:

> Add Hashable ByteArray instance using data-array-byte compat package 
> Add instance for Data.Array.Byte.ByteArray.

Plus Cabal somehow picking a cached version of either package over a new one. As a workaround could you:

* `cabal update` and try again?
* ... remove your `.cabal` and try again? This is how we "fixed" it on CI.

DigitalBrains1, 2023-04-19:
That's odd. The `cabal update` fixed it for me on GHC 9.0.2... After `cabal update`, it picked up a new version of `primitive`.

Maybe we need to implement the full logic then, support all combinations of `hashable` and `primitive`. This is like a perfect storm of in themselves benign changes in our dependencies that together completely go to shit.

kleinreact, 2023-04-20:
Running
```
rm -Rf ~/.cabal
git clone https://github.com/clash-lang/clash-compiler.git && cd clash-compiler
cabal update
cabal build clash-lib
```
still results in the same error, so the GHC environment file should not be the problem here.

I am using
```
ghc 9.0.2
cabal-install version 3.6.2.0 (compiled using version 3.6.3.0.1 of the Cabal library)
```
The index state picked by `cabal update` is `2023-04-19T21:29:05Z`.

If go back to 117e2bac, then everything compiles, so it must be #2448/#2449 that introduce the issue.

martijnbastiaan, 2023-04-20:
I'd be really surprised if this works, but does it work if you remove your `.ghc.environment*` file from the repository root? I agree with @DigitalBrains1 that we should probably fix this for the full cross product of relevant primitive/hashable versions.

kleinreact, 2023-04-20:
I do a fresh `clone`, which ensures that there cannot be any old `.ghc.environment*` files.

martijnbastiaan, 2023-04-20:
Okay thanks. I'll investigate.

DigitalBrains1, 2023-04-26:
Can't you just add a `Hashable` instance for `Data.Primitive.ByteArray` when `primitive == 0.7.4.0`? I mean, I can't immediately come up with the proper combination of the `ifdef`s, but it seems pretty doable? Since the problem is that `hashable == 1.4.2.0` defines a `Hashable` instance for `Data.Array.Byte.ByteArray` but **not** for `Data.Primitive.ByteArray`.

kleinreact, 2023-04-26:
The workaround works for me. Thanks.

DigitalBrains1, 2023-04-27:
I have created a little repository to do the testing with Martijns script outside `clash-compiler`. This means we can quickly test the solution, and another big plus: we can actually test this `base >= 4.17` stuff. The results are:

```
FAIL: ghc-8.10.7, hashable-1.4.2.0, primitive-0.7.4.0
FAIL: ghc-8.6.5,  hashable-1.4.2.0, primitive-0.7.4.0
FAIL: ghc-8.8.4,  hashable-1.4.2.0, primitive-0.7.4.0
FAIL: ghc-9.0.2,  hashable-1.4.2.0, primitive-0.7.4.0
FAIL: ghc-9.2.7,  hashable-1.4.2.0, primitive-0.7.4.0
FAIL: ghc-9.4.5,  hashable-1.4.1.0, primitive-0.7.4.0
FAIL: ghc-9.4.5,  hashable-1.4.2.0, primitive-0.7.4.0

OK:   ghc-8.10.7, hashable-1.4.0.2, primitive-0.7.4.0
OK:   ghc-8.10.7, hashable-1.4.0.2, primitive-0.8.0.0
OK:   ghc-8.10.7, hashable-1.4.1.0, primitive-0.7.4.0
OK:   ghc-8.10.7, hashable-1.4.1.0, primitive-0.8.0.0
OK:   ghc-8.10.7, hashable-1.4.2.0, primitive-0.8.0.0
OK:   ghc-8.6.5,  hashable-1.4.0.2, primitive-0.7.4.0
OK:   ghc-8.6.5,  hashable-1.4.0.2, primitive-0.8.0.0
OK:   ghc-8.6.5,  hashable-1.4.1.0, primitive-0.7.4.0
OK:   ghc-8.6.5,  hashable-1.4.1.0, primitive-0.8.0.0
OK:   ghc-8.6.5,  hashable-1.4.2.0, primitive-0.8.0.0
OK:   ghc-8.8.4,  hashable-1.4.0.2, primitive-0.7.4.0
OK:   ghc-8.8.4,  hashable-1.4.0.2, primitive-0.8.0.0
OK:   ghc-8.8.4,  hashable-1.4.1.0, primitive-0.7.4.0
OK:   ghc-8.8.4,  hashable-1.4.1.0, primitive-0.8.0.0
OK:   ghc-8.8.4,  hashable-1.4.2.0, primitive-0.8.0.0
OK:   ghc-9.0.2,  hashable-1.4.0.2, primitive-0.7.4.0
OK:   ghc-9.0.2,  hashable-1.4.0.2, primitive-0.8.0.0
OK:   ghc-9.0.2,  hashable-1.4.1.0, primitive-0.7.4.0
OK:   ghc-9.0.2,  hashable-1.4.1.0, primitive-0.8.0.0
OK:   ghc-9.0.2,  hashable-1.4.2.0, primitive-0.8.0.0
OK:   ghc-9.2.7,  hashable-1.4.0.2, primitive-0.7.4.0
OK:   ghc-9.2.7,  hashable-1.4.0.2, primitive-0.8.0.0
OK:   ghc-9.2.7,  hashable-1.4.1.0, primitive-0.7.4.0
OK:   ghc-9.2.7,  hashable-1.4.1.0, primitive-0.8.0.0
OK:   ghc-9.2.7,  hashable-1.4.2.0, primitive-0.8.0.0
OK:   ghc-9.4.5,  hashable-1.4.1.0, primitive-0.8.0.0
OK:   ghc-9.4.5,  hashable-1.4.2.0, primitive-0.8.0.0
```

Note that `hashable-1.4.0.2` does not support GHC 9.4, so no results for that.

The script is [here](https://gist.github.com/DigitalBrains1/adda2182bb0dd9dc86af3ce2f148ceb4). It's accompanying repo (which it will clone automatically) is [here](https://github.com/DigitalBrains1/test-hashable-primitive). It's current branch [`current-situation`](https://github.com/DigitalBrains1/test-hashable-primitive/tree/current-situation) is with the code from `clash-compiler`.

DigitalBrains1, 2023-04-27:
And the version on branch [`fixed`](https://github.com/DigitalBrains1/test-hashable-primitive/tree/fixed) works for the full test matrix. I'd like to suggest we use that and be done with it :-).

commit c56131d0 (2023-04-28): Correct logic when to orphan Hashable ByteArray (#2460)

PR #2448 and #2449 were incomplete, missing some cases where we still
needed to create an orphan instance for `Hashable ByteArray` ourselves.
This should get it right, including for when we will support GHC 9.4
(which is guarded by the expression `MIN_VERSION_base(4,17,0)`).

Fixes #2453
closed 2023-04-28 by DigitalBrains1