clash-lang/clash-compiler #2455: Use of `~LONGESTPERIOD` prevents absolute time calculation in Clash simulation

martijnbastiaan, 2023-04-23:
Xilinx's primitive models use a global set/reset (GSR) that is used to implement initial values. This reset signal is (by default) defined as asserted for 100 ns. Therefore, to faithfully replicate the VHDL/Verilog model, Clash simulation will have to know the absolute simulation time. Unfortunately, Clash's clock generation primitives wait for `~LONGESTPERIOD` before it makes clock tick, a value only known at RTL compile time.

------

Encountered in: https://github.com/clash-lang/clash-compiler/pull/2452. Waveform:

[image]

Note that the first three cycles (0000, 0001, 0002) are ignored by the pipeline because GSR is asserted.

-------------

@christiaanb Do you remember the rationale behind `~LONGESTPERIOD`? Could we simply deprecate it / set it to a static value?

christiaanb, 2023-04-23:
It was done to fix issue https://github.com/clash-lang/clash-compiler/issues/2001 so a static value will always be wrong if we have clock period that is longer than said static value.

So perhaps we can add a static value on top of `~LONGESTPERIOD`.

christiaanb, 2023-04-23:
Why do we even model GSR in #2452? According to https://docs.xilinx.com/r/en-US/ug900-vivado-logic-simulation/Global-Set-and-Reset-Net you only want to deal with explicit GSR in post-synt/post-impl simulation.

christiaanb, 2023-04-23:
Perhaps we want to change: 

https://github.com/clash-lang/clash-compiler/blob/053669c0c115be7af48f290a92923a4e0898e4a0/tests/src/Test/Tasty/Vivado.hs#L92

to: `lauch_simulation -mode behavioral`

See also: https://docs.xilinx.com/r/2021.2-English/ug900-vivado-logic-simulation/Behavioral-Simulation

DigitalBrains1, 2023-04-23:
Huh, I never realised we were _not_ running a behavioural simulation in Xilinx 😅

I agree, let's switch to behavioural simulation.

martijnbastiaan, 2023-04-23:
> Why do we even model GSR in #2452? 

It's what the Xilinx's `XPM_CDC_GRAY` model does (unless `#ifdef ONESPIN`). The GSR is defined by `verunilibs`'s `glbl.v`. So if we want a faithful correspondence between Verilog/VHDL simulation and Clash simulation we'll need to replicate what the GSR does. (It's not a hard requirement for me for that PR, but not be able to query the absolute time during Clash simulation _does_ seem like a fundamental shortcoming to me.)

> Perhaps we want to change:

\+

> Huh, I never realised we were not running a behavioural simulation in Xilinx

We do :). `behavioral` is the default.

martijnbastiaan, 2023-04-23:
> So perhaps we can add a static value on top of `~LONGESTPERIOD`.

Adding 100 ns would entirely sidestep my issues.

christiaanb, 2023-04-24:
@martijnbastiaan where in the docs does it say that behavioral is the default? https://docs.xilinx.com/r/2021.2-English/ug900-vivado-logic-simulation/Behavioral-Simulation seems to suggest otherwise.

martijnbastiaan, 2023-04-24:
UG835 on `launch_simulation`:

[image]

Running with `launch_simulation -mode behavioral` gives me:

[image]

`GSR` would be the global reset, `glblGSR_xpmcdc` is the reset line in `XPM_CDC_GRAY` (copied from `GSR`).

closed 2023-05-04 by martijnbastiaan

christiaanb, 2023-11-10:
We've released [v1.8.0](https://github.com/clash-lang/clash-compiler/releases/tag/v1.8.0), which includes a fix for this issue.

commit 9953049e (2023-04-23): Make ~LONGESTPERIOD return 100 ns at minimum

Fixes #2455