clash-lang/clash-compiler #2488: hackage.haskell.org doesn't have the latest version 1.6.4

yurivict, 2023-06-05:
[here](https://hackage.haskell.org/package/clash-systemverilog)

alex-mckenna, 2023-06-06:
The packages `clash`, `clash-vhdl`, `clash-verilog`, and `clash-systemverilog` on Hackage are not the currently maintained / released packages. At some point the `clash-HDL` packages were folded into `clash-lib`, and `clash` was deprecated in favour of `clash-ghc`.

Maybe we should consider removing these old packages, or at least make sure the `clash-HDL` ones mentioned they are deprecated in favour of `clash-lib` (a similar warning is in the contents of `clash` on Hackage)

martijnbastiaan, 2023-06-06:
The only thing you should need for Clash to work on FreeBSD would be for GHC/Cabal to work. It does not look like [GHC officially supports FreeBSD](https://github.com/haskell/ghcup-hs/issues/834#issuecomment-1571336642) though. Apart from that I don't think Clash does any OS-specific things.

alex-mckenna, 2023-06-06:
It shouldn't be the most difficult thing to get FreeBSD support though, since `ghc` used to support it but dropped it [a few versions ago](https://github.com/haskell/ghcup-hs/issues/834#issuecomment-1571336642) for some reason

yurivict, 2023-06-06:
> It does not look like https://github.com/haskell/ghcup-hs/issues/834#issuecomment-1571336642 though.

ghc works fine on FreeBSD though. We have many Haskell-based ports.

alex-mckenna, 2023-06-08:
I think we should mark `clash-vhdl`, `clash-verilog` and `clash-systemverilog` as deprecated on Hackage (as we have with `clash`) so this confusion doesn't happen again. Other than that there isn't really anything to do for this issue.

@martijnbastiaan Would you mind, since you have permission

martijnbastiaan, 2023-06-13:
Done!

closed 2023-06-13 by martijnbastiaan