clash-lang/clash-compiler #2493: Maintain Core ID -> Netlist ID mapping in `NetlistEnv`

alex-mckenna, 2023-06-11:
When converting from core identifiers to netlist identifiers, we often end up using `unsafeFromCoreId`, which pulls out a `Text` from an identifier and turns this into a `RawIndentifier` in netlist. This is, as the name suggests, unsafe, as we do nothing to ensure the identifier is a valid identifier for our target HDL. Using the safe `makeBasicId` (or similar) as a default would be more sensible, but currently switching all our calls to `unsafeFromCoreId` to `makeBasicId` would mean instead of

```verilog
wire foo

// much later
assign foo = bar;
```

we would end up creating

```verilog
wire foo

// much later
assign foo_1 = bar;
```

We can prevent this problem by taking advantage of the fact that core is well-scoped lexically, and should have no name shadowing. If we include a mapping from core identifiers to netlist identifiers, we can store this in a local environment (i.e. inside the `NetlistEnv`, __not__ the `NetlistState`) and update our mapping as we walk through the syntax tree. Checking this mapping before making a basic identifier lets us avoid re-generating new basic identifiers when we can.

Something subtle: we would by default still like places where a name is explicitly provided by a user to be unsafe and not alter this name. We should make sure that these identifiers still use `unsafeFromCoreId` instead of moving to `makeBasicId`.

alex-mckenna, 2023-06-11:
Actually, moving to the `NetlistEnv` while nicer seems to be more work than I'd like. The advantage would be that we could generate nicer IDs in some cases as we would generate the same basic identifier in contexts where the same identifier appears in distinct scopes, so names would be closer to the ones we request. It's a bit too fiddly because with the reader style of doing this you tend to need to nest functions / use a more CPS style when you want to run an action then return the modified local environment.

Let's reduce scope here and keep using it in the state. We can still get the real benefit of being able to use `makeBasicId` in more places and reduce use of `unsafeFromCoreId`.

martijnbastiaan, 2023-06-13:
This feels like it could be a preprocess step that replaces every Core ID with a Netlist ID. That basically guarantees we don't mess it up. Would that be an option here?

alex-mckenna, 2023-06-13:
I think we might be able to do this without that, by changing the `is_store` in `IdentifierSet` from a `HashSet Identifier` to a `VarEnv Identifier`. I'm pretty sure netlist generation shouldn't make new identifiers _completely_ from scratch, they should all come from some binder in core. If I'm wrong about this, these extra names are in some way special and it's perhaps not a bad thing to make this more obvious (and then add a `NOTE` in source)

christiaanb, 2023-06-13:
Netlist generation makes up new identifiers completely from scratch all the time:
* `~GENSYM[new][n]`
* `~VAR[n]`
* Blackbox declarations (vs expressions)

martijnbastiaan, 2023-06-14:
Same goes for Haskell black boxes, they call `Id.makeBasic "intermediate_var"` a lot.

With regards to the one-pass approach: we already do this in `mkUniqueNormalized`. We just throw away the evidence that we passed through the `Netlist.Id` logic. So instead of tacking another mapping on top, we should preserve the evidence (maybe through `Term Id` -> `Term Identifier`) right there.

alex-mckenna, 2023-06-14:
Right, of course 🤦 Sorry, I haven't been thinking about netlist for a while.

I think maintaining this mapping as a separate item in the identifier set is still how I'd do it. We don't need to care about identifiers that didn't come from core for this mapping because it's just a first check to see if we even need to generate (instead of a safety mechanism to make sure we don't generate the same name twice).

This actually makes the implementation even easier in my mind. I'll try and throw a quick PR together at some point