clash-lang/clash-compiler #2502: Clash doesn't respect `NOINLINE` on polyvariadic functions with "extra" arguments

martijnbastiaan, 2023-06-15:
Just going by the title I'm tempted to mark this as `easy` ;-).

# Reproducer
```haskell
import Data.String.Interpolate (__i)
import Clash.Explicit.Prelude
import Clash.Annotations.Primitive

class X a where
  x :: a

instance X (Signal System Int) where
  x = pure 3

instance X a => X (Signal System Int -> a) where
  x !_i = x

bb :: X a => Int -> a
bb !_ = x
{-# NOINLINE bb #-}
{-# ANN bb hasBlackBox #-}
{-# ANN bb (
  let bbName = show 'bb
  in InlineYamlPrimitive [minBound..] [__i|
    BlackBox:
      name: "#{bbName}"
      kind: Expression
      template: "~ARG[2]"
|]) #-}

bbWrapper :: X a => Int -> a
bbWrapper i = bb i
{-# NOINLINE bbWrapper #-}

topEntity :: Int -> Signal System Int -> Signal System Int
topEntity i0 i1 = bbWrapper i0 i1
```

Gives:

```
$ ls vhdl/Main.topEntity/
clash-manifest.json  Main_topEntity_types.vhdl  topEntity.vhdl
```

`bbWrapper` did **not** get its own file. [`DebugApplied` log](https://github.com/clash-lang/clash-compiler/files/11756817/reproducer.log).

# Removing the "extra" argument

```haskell
import Data.String.Interpolate (__i)
import Clash.Explicit.Prelude
import Clash.Annotations.Primitive

class X a where
  x :: a

instance X (Signal System Int) where
  x = pure 3

instance X a => X (Signal System Int -> a) where
  x !_i = x

bb :: X a => a
bb = x
{-# NOINLINE bb #-}
{-# ANN bb hasBlackBox #-}
{-# ANN bb (
  let bbName = show 'bb
  in InlineYamlPrimitive [minBound..] [__i|
    BlackBox:
      name: "#{bbName}"
      kind: Expression
      template: "~ARG[1]"
|]) #-}

bbWrapper :: X a => a
bbWrapper = bb
{-# NOINLINE bbWrapper #-}

topEntity :: Signal System Int -> Signal System Int
topEntity i1 = bbWrapper i1
```

Gives:

```
$ ls vhdl/Main.topEntity/
bb1.vhdl  clash-manifest.json  Main_topEntity_types.vhdl  topEntity.vhdl
```

`bbWrapper` **did** get its own file. [`DebugApplied` log](https://github.com/clash-lang/clash-compiler/files/11756818/reproducer_no_extra_arg.log).

# Removing the polyvariadic aspect
```haskell
import Data.String.Interpolate (__i)
import Clash.Explicit.Prelude
import Clash.Annotations.Primitive

bb :: Int -> Signal System Int -> Signal System Int
bb !_ !_ = pure 3
{-# NOINLINE bb #-}
{-# ANN bb hasBlackBox #-}
{-# ANN bb (
  let bbName = show 'bb
  in InlineYamlPrimitive [minBound..] [__i|
    BlackBox:
      name: "#{bbName}"
      kind: Expression
      template: "~ARG[1]"
|]) #-}

bbWrapper :: Int -> Signal System Int -> Signal System Int
bbWrapper i = bb i
{-# NOINLINE bbWrapper #-}

topEntity :: Int -> Signal System Int -> Signal System Int
topEntity i0 i1 = bbWrapper i0 i1
```

Gives:

```
$ ls vhdl/Main.topEntity/
bbWrapper.vhdl  clash-manifest.json  Main_topEntity_types.vhdl  topEntity.vhdl
```

`bbWrapper` **did** get its own file. [`DebugApplied` log](https://github.com/clash-lang/clash-compiler/files/11756819/reproducer_no_polyvariadic.log).

christiaanb, 2023-06-15:
Easy when you know where to look:
https://github.com/clash-lang/clash-compiler/blob/26a4bc2d78eb8d544cf0a922257fddaf4a1080ab/clash-lib/src/Clash/Normalize/Transformations/Specialize.hs#L443

This doesn’t deal well with the case where the argument we specialize on _lacks_ a NOINLINE spec; in two ways:
1. It doesn’t preserve the NOINLINE spec of the function being specialized
2. It doesn’t pick the name of the function being specialized when it has a NOINLINE, while the argument being specialized on does _not_

closed 2023-06-27 by christiaanb

christiaanb, 2023-11-10:
We've released [v1.8.0](https://github.com/clash-lang/clash-compiler/releases/tag/v1.8.0), which includes a fix for this issue.

commit 0bc4932a (2023-06-17): Respect function's `NOINLINE` pragma when specializing

Fixes #2502
commit c2f9cee3 (2023-06-17): Respect function's `NOINLINE` pragma when specializing

Fixes #2502
commit 334b6841 (2023-06-17): Respect function's `NOINLINE` pragma when specializing

Fixes #2502
commit 2547386b (2023-06-17): Respect function's `NOINLINE` pragma when specializing

Fixes #2502