clash-lang/clash-compiler #2516: `clash-testsuite`: dependencies in `sequentialTestGroup` wrong

DigitalBrains1, 2023-06-21:
It seems that the dependency resolution of the new Tasty test structure is somehow bugged. For instance, it would appear that Vivado VHDL tests pick GHDL as their dependency instead of `clash (gen)`:

```
$ cabal run clash-testsuite -- -j8 -p Calculator.VHDL.Vivado -l
Up to date
..examples.Calculator.VHDL.GHDL.ghdl (import testBench)
..examples.Calculator.VHDL.GHDL.ghdl (make testBench)
..examples.Calculator.VHDL.GHDL.ghdl (sim testBench)
..examples.Calculator.VHDL.Vivado.vivado (sim testBench)
```

and indeed:
```
Up to date
.
  examples
    Calculator
      VHDL
        GHDL
          ghdl (import testBench): FAIL
            Exception: /tmp/clash-test-8f7246e8d30587e8/hdl: getDirectoryContents:openDirStream: does not exist (No such file or directory)
            Use -p '/Calculator.VHDL.Vivado/&&/ghdl (import testBench)/' to rerun this test only.
          ghdl (make testBench):   SKIP
            
            Use -p '/Calculator.VHDL.Vivado/&&/ghdl (make testBench)/' to rerun this test only.
          ghdl (sim testBench):    SKIP
            
            Use -p '/Calculator.VHDL.Vivado/&&/ghdl (sim testBench)/' to rerun this test only.
        Vivado
          vivado (sim testBench):  SKIP
            
            Use -p '/Calculator.VHDL.Vivado/&&/vivado (sim testBench)/' to rerun this test only.

4 out of 4 tests failed (0.03s)
```

It runs GHDL without `clash (gen)` and then `SKIP`s also Vivado because `ghdl (import testBench)` was `FAIL`.

This works (but runs GHDL as well):
```
$ cabal run clash-testsuite -- -j8 -p '/Calculator.VHDL.Vivado/ || /Calculator.VHDL.clash (gen)/'
Up to date
.
  examples
    Calculator
      VHDL
        clash (gen):               OK (2.37s)
        GHDL
          ghdl (import testBench): OK
          ghdl (make testBench):   OK (0.48s)
          ghdl (sim testBench):    OK
        Vivado
          vivado (sim testBench):  OK (17.01s)

All 5 tests passed (19.91s)
```

I'm going to guess that Vivado doesn't start until GHDL has completed, even though they should have launched in parallel. I didn't monkey patch anything to verify that guess.

martijnbastiaan, 2023-06-22:
We suspect this PR introducing the issue: https://github.com/clash-lang/clash-compiler/pull/2284.

martijnbastiaan, 2023-07-20:
This issue turned out to be partly caused by Tasty, fixed over here: https://github.com/UnkindPartition/tasty/pull/376. The other half of the issue is that we really did have false dependencies in our testsuite.

commit c79466ec (2023-07-29): Remove false dependencies from `clash-testsuite`

Fixes #2516
closed 2023-07-29 by martijnbastiaan