clash-lang/clash-compiler #2525: Use newtypes for Clash number types on GHC 9.4+

christiaanb, 2023-07-03:
Once #2511 is merged, we should use newtype again for our number types on GHC 9.4:

https://github.com/clash-lang/clash-compiler/blob/48f7f52feab5aa1619b6945d6f1dbbb9a7ec445d/clash-prelude/src/Clash/Sized/Internal/Index.hs#L160-L170

https://github.com/clash-lang/clash-compiler/blob/48f7f52feab5aa1619b6945d6f1dbbb9a7ec445d/clash-prelude/src/Clash/Sized/Internal/Signed.hs#L187-L197

https://github.com/clash-lang/clash-compiler/blob/48f7f52feab5aa1619b6945d6f1dbbb9a7ec445d/clash-prelude/src/Clash/Sized/Internal/Unsigned.hs#L203-L213

They are the reason the `OPAGUE` came into being in the first place, as GHC 9 (base-4.15) introduced W/W transform on newtypes/casts even for `-O0`; which OPAGUE now stops.

gergoerdi, 2023-07-11:
For anyone grepping for issues: it's `OPAQUE`, not `OPAGUE`.

christiaanb, 2023-08-11:
Implemented by #2535

closed 2023-08-11 by christiaanb